pavel-demin/red-pitaya-notes #23: SDR receiver question

zevv, 2015-05-08:
Hi,

Thanks for the SDR receiver project; I'm trying to integrate this as an SDR frontent for some custom code, but I run into some issues with the downsampled data.

The pitaya runs sdr_receiver binary with proper bitstream loaded in the FPGA. I'm getting the samples from the TCP socket, but with some kind of periodic distortion goin on.

Attached are two screenshots of a 5 Mhz carrier received by the pitaya. The receiver is tuned to 5Mhz, and there's a slight offset of about 100Hz between sender and received. The problem is that there's a kind of periodicity in the received wave of 134 samples, as can be seen in the waveform.

Do you have any kind of explanation for what I'm seeing, and any idea how I can solve this?

Thanks,

[image]
[image]

pavel-demin, 2015-05-08:
Hi Ico,

Thanks for your interest in this project. 

The tuning frequency difference is due to the frequency deviation of the 125 MHz oscillator on the Red Pitaya board. For my board it's around 25 ppm at room temperature. 

Here is how it can be corrected:
```
  frequency = 5000000;
  correction = 25; // ppm
  frequency = (int)floor(frequency * (1.0 + correction * 1.0e-6) + 0.5);
```

As for the distortion, I've never looked at the raw I/Q data very closely. I'm not sure that I understand what is shown on the first two plots.

Would it be enough to reproduce these plots if I plug a 5 MHz sine wave from a generator to Red Pitaya, register the I/Q data from the sdr-receiver program and then plot log10(I) and log10(Q)?

Kind regards,

Pavel

zevv, 2015-05-08:
* On 2015-05-08 18:03:19 +0200, Pavel Demin wrote:
 
> The tuning frequency difference is due to the frequency deviation of
> the 125 MHz oscillator on the Red Pitaya board. For my board it's
> around 25 ppm at room temperature. 

I understand, the difference is not the problem; it was merely a
description to explain what's on the plot :)

> As for the distortion, I've never looked at the raw I/Q data very
> closely. I'm not sure that I understand what is shown on the first two
> plots.

Sorry for not explaining properly what's in there: these are simply
time-domain graphs of the Q and I data, with the second graph zoomed in on
a detail. The graph should show a sine wave of approx 100HZ (the
frequency differente between osciallator and pitya), but instead it
shows fragments of a sine wave, each 134 samples long, with jumps in
between there.

The 134 samples can not be measured from these graphs, but I was able to
measure this by inspecting the output sample-by-sample by hand.

I first suspected there to be buffer overruns somewhere in the signal
chain, but with this 134 sample interval the problem is very unlikely to
be in the buffer domain.

> Would it be enough to reproduce these plots if I plug a 5 MHz sine
> wave from a generator to Red Pitaya, register the I/Q data from the
> sdr-receiver program and then plot log10(I) and log10(Q)?

Apart from the log(), that's exactly what it is, a very basic setup:
simply plot linear Q and Y.

If you're not able to reproduce I can be sure the problem is on my
side :)

Thanks,

-- 
:wq
^X^Cy^K^X^C^C^C^C

pavel-demin, 2015-05-09:
In order to test the signal processing chain I replaced the ADC interface with a digital sine wave generator.

Here are the sources of the FPGA configuration together with the server and client programs:

https://github.com/pavel-demin/red-pitaya-notes/tree/develop/projects/sdr_receiver_test

and here are the pre-built FPGA configuration and the server program:

https://googledrive.com/host/0B-t5klOOymMNfmJ0bFQzTVNXQ3RtWm5SQ2NGTE1hRUlTd3V2emdSNzN6d0pYamNILW83Wmc/sdr_receiver_test/

Nothing changed on the client side. So, you can directly use your program.

When running all this together, I obtain the following plots for the raw I signal.

Could you check if you obtain something similar with this test FPGA configuration?

[image]
[image]

zevv, 2015-05-09:
Hi Pavel,

Thanks for the update; I've got my hardware at the office so I'm not
able to test right now, I'll get back to you somewhere next week.

Ico

pavel-demin, 2015-05-09:
Just realized that I did the plots for my previous comment at 621 kHz.

Here are the same plots at 5 MHz.

[image]
[image]

104012, 2015-05-10:
Hi Pavel,

Thanks for this awesome SDR-project! I tested your 5 MHz wave and this is what I get using your client:
[image]
This is zoomed in at the peak, and it looks quite similar to your graph. When I use Ico's script to pipe data to baudline, I get something similar,
[image]

In frequency domain:
[image]

It seems that this fluctuation gets worse as I increase the frequency, just like your signal at 621 KHz was quite smooth. Any idea what the problem may be?

pavel-demin, 2015-05-10:
Thanks for mentioning that you are using baudline. It helped me to better understand the problem.

Here is what I found in the baudline manual (http://www.baudline.com/manual/options.html):
"When one of the floating point -format modes (le32f, be32f) are selected, the stdin floating point samples will be multiplied by the -scaleby option and then quantized to fit into a 16 bit signed integer."

Looks like baudline internally uses only 16 bits for the samples.

In the I/Q data coming from the SDR receiver, the 22 most significant bits contain useful information.

I've added two client programs that output the I/Q data in the le16 and le32f format:

https://github.com/pavel-demin/red-pitaya-notes/blob/develop/projects/sdr_receiver_test/client/sdr-receiver-client-le16.c
https://github.com/pavel-demin/red-pitaya-notes/blob/develop/projects/sdr_receiver_test/client/sdr-receiver-client-le32f.c

Here are the commands to compile and run them together with baudline:

```
gcc sdr-receiver-client-le16.c -o sdr-receiver-client-le16 -lm
./sdr-receiver-client-le16 | ./baudline -stdin -quadrature -channels 2 -format le16 -samplerate 100000
```

```
gcc sdr-receiver-client-le32f.c -o sdr-receiver-client-le32f -lm
./sdr-receiver-client-le32f | ./baudline -stdin -quadrature -channels 2 -format le32f -scaleby 32767 -samplerate 100000
```

Here is a waveline display that I obtain with sdr-receiver-client-le16:

[image]

I'd say that it now looks very similar to the plots that I posted in my previous comments.

pavel-demin, 2015-05-10:
Concerning high noise at higher frequencies.

The sine wave generator is running at 125 MHz. At 5 MHz, we have 25 samples per period. As there is no filtering between the generator and the signal processing chain, I'd say that 25 samples are not enough to produce a smooth signal.

104012, 2015-05-15:
Hi Pavel,

Thank you for your efforts. Forgive me for not being an expert in this domain, but is this 'unsmoothness' at higher frequencies therefore a characteristic of the digital down-converter? Is this the price we pay in order to undersample a signal?

pavel-demin, 2015-05-15:
I'd say that this unsmoothness is specific to this test FPGA configuration where I replaced the ADC interface with a digital sine wave generator. I think that it's this digital sine wave generator that is the main source of the unsmoothness.

It would be interesting to make the same plots with the normal SDR receiver configuration, an external sine wave generator and `sdr-receiver-client-le16.c` as an interface to baudline.

At the moment, I don't have any suitable generator.

If it was not clear from my previous long comment, here is a shorter summary:
 - the distortions that Ico observed are most probably due to a wrong conversion from 32 bits to 16 bits;
 - the 20-22 (depends on the output sample rate) most significant bits contain useful information;
 - the conversion from 32 bits to 16 bits in `sdr-receiver-client-le16.c` is done as  `bufferOut[i] = (int16_t)(bufferIn[i] >> 10);` and seems to work;
 - it's possible that `bufferOut[i] = (int16_t)(bufferIn[i] >> 11);` or `bufferOut[i] = (int16_t)(bufferIn[i] >> 12);` would also work.

I'm very curious to see an update from Ico.

zevv, 2015-05-16:
* On 2015-05-16 00:14:11 +0200, Pavel Demin wrote:
 
> I'm very curious to see an update from Ico.

I'm very sorry, but I did not find a chance to test this yet. I do not
own the hardware myself, but use it at a customer's site where I only
spent my fridays. This friday I did not find the time to test
unfortunately.

-- 
:wq
^X^Cy^K^X^C^C^C^C

closed 2016-02-22 by pavel-demin