pavel-demin/red-pitaya-notes #52: debain.seed won't execute cleanly

ul5255, 2015-07-24:
I attempted an installation of Debian 8 (32bit) in VirtualBox under Windows 7.
Installation failed at https://github.com/pavel-demin/red-pitaya-notes/blob/gh-pages/etc/debian.seed#L86
because apt-get couldn't locate lib32ncurses5. I then changed that to libncurses5 and installation went fine.

pavel-demin, 2015-07-24:
Thanks for testing debian.seed with a 32-bit system.

However, the instructions at 

http://pavel-demin.github.io/red-pitaya-notes/development-machine/

are for 64-bit virtual machines.

So, it's normal that they don't work with a 32-bit virtual machine.

Is there any reason to install a 32-bit virtual machine in 2015?

ul5255, 2015-07-24:
> Is there any reason to install a 32-bit virtual machine in 2015?

I have VirtualBox 4.3.30 on Windows 7 64bit (Intel Core i5). When I set up the VM I got only offered 'Debian (32bit)' so I thought I play it save and install a 32bit Debian. Anyways, I am currently downloading Vivado 2015.2 and see how that goes ...

ul5255, 2015-07-24:
... and then xsetup failed:
ERROR: This installation is not supported on 32 bit platforms.

So back to installation of 64bit Debian VM.

closed 2015-07-24 by ul5255

ul5255, 2015-07-24:
Addendum: After I enabled VT-x in the BIOS I was able to do a proper installation of 64bit Debian based on your instructions.