pavel-demin/red-pitaya-notes #56: Building Deb Environment needs root

hmaarrfk, 2015-08-14:
Is there a way to make building the debian environment NOT require root?

Requiring scripts to run as root is scary.

It can also affect the development environment by starting or stopping scripts.

(Also awesome work :D)

pavel-demin, 2015-08-14:
Thanks for your suggestion.

The image.sh, debian.sh and ubuntu.sh scripts require administrative privileges for the following:
 - losetup
 - chroot
 - mount
 - setting correct owners and permissions inside the SD card image

At the moment, I don't see how it can be done without administrative privileges. Any ideas?

Personally, I'm using a dedicated and easily recoverable machine to run all these scripts.

I've found a similar question on stackoverflow:
http://stackoverflow.com/questions/10949169/creating-sd-image-without-root-privileges
However, the proposed answers don't look very convincing.

hmaarrfk, 2015-08-15:
Yea compiling for different archs seems to be rather complicated.

I thought losetup and mount could be saved until the end. You shouldn't have to create the SD card everytime, and you could make manually do it.

I thought fakeroot and fakechroot would do it, but it seems like it isn't that easy, especially with qemu-arm-static

Maybe just keep this as a feature request...

pavel-demin, 2015-08-15:
Just found a script that helps to run qemu-arm-static with fakechroot:

https://lists.debian.org/debian-embedded/2011/04/msg00001.html

hmaarrfk, 2015-08-15:
I have a fedora machine, and I've been trying to follow those instructions. It doesn't seem super trivial
1. The binary names are not exactly the same.
2. he had to do an strace to find the right directories. I've never used an strace. I feel like things have changed sicne 2011

hmaarrfk, 2015-08-15:
This might be useful to myself

https://rwmj.wordpress.com/2013/12/22/how-to-run-aarch64-binaries-on-an-x86-64-host-using-qemu-userspace-emulation/#content

Getting rather late here, I should probably go to sleep soon.

XYEFA, 2016-07-19:
What has been helpful to me is adding myself to the sudo'ers list and sudo when necessary ... It has worked.

pavel-demin, 2016-07-19:
Using sudo is not a solution.

We are looking for a way to build SD card images without administrative privileges.

pavel-demin, 2016-09-24:
This issue has been open for more than a year now. Looks like running image.sh, debian.sh and ubuntu.sh scripts in a disposable or easily recoverable virtual machine is still the only solution. I'm closing this issue since I don't think that we'll find other solutions.

closed 2016-09-24 by pavel-demin