pavel-demin/red-pitaya-notes #87: Why do you need gcc-arm-linux-gnueabihf

hmaarrfk, 2015-10-01:
Aren't the compilers shipped with the Xilinx SDK good enough?

pavel-demin, 2015-10-01:
Hi Mark,

The problem with the compilers shipped with the Xilinx SDK is that they only support the armel architecture.

I prefer to use the armhf architecture for the following reasons:
 - better performance in floating point intensive applications
 - supported by the recent releases of the Debian and Ubuntu distributions
 - supported by the latest version of the Red Pitaya ecosystem

The arm-linux-gnueabihf compilers support the armhf architecture and are readily available from the Debian and Ubuntu repositories. So, I don't see any reason to not use them.

Cheers,

Pavel

hmaarrfk, 2015-10-01:
Thanks for the explanation. I didn't know there was a difference in the instruction set for armel and armhf. 

I'll have to find the appropriate compiler for Fedora.

pavel-demin, 2015-10-01:
The latest version of the Red Pitaya system is built with Linaro 14.11. The Linaro binaries should run on Fedora. They can be downloaded from the following link:

https://releases.linaro.org/14.11/components/toolchain/binaries/arm-linux-gnueabihf/

closed 2015-10-01 by pavel-demin

hmaarrfk, 2015-10-01:
Thanks,

Seems that Fedora uses different naming conventions. Just want to document what I found for now:
```
sudo dnf install gcc-arm-linux-gnu
```

should do the trick. hf is not in the name of the compiler, but rather must be passed in as a parameter.
Compiling with
```
arm-linux-gnu-gcc -mfloat-abi=hard ...
```
should be the equivalent to what you have working for debian based installs.

I might run into other problems because it seems that the GCC version that is running is 5.2

```
$ arm-linux-gnu-gcc --version
arm-linux-gnu-gcc (GCC) 5.2.1 20150716 (Red Hat Cross 5.2.1-3)
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

hmaarrfk, 2015-10-01:
Ok, seems that compiling userspace programs with default Fedora compilers is not supported. Maybe I'll just transfer my development machine to Ubuntu.

Sorry for having so much spam in this question.