pavel-demin/red-pitaya-notes #92: rp.h behind?

Gulzt, 2015-10-12:
Hi, your build is awesome, thanks! Only one issue I encounter. I can't get i2c commands from the API to run. When checking the filesystem on the redpitaya I found out these commands don't exist in the file  /opt/include/rp.h

However, on the github of RedPitaya those API commands show up in the rp.h file. Is this headerfile (and related object file) behind?

pavel-demin, 2015-10-12:
It's indeed possible that rp.h is not in sync with the latest Red Pitaya release.

To build my Debian based SD card image with the Red Pitaya ecosystem I'm still using version 0.92 of the Red Pitaya ecosystem. The latest Red Pitaya ecosystem version is 0.94.

Since the latest Red Pitaya SD card image is based on my Debian based SD card image, I'm not sure if I still should keep my version updated. Is there any reason to prefer my version?

Gulzt, 2015-10-12:
Wow, thank you for your quick responds.Yes, there is much reason to use your version:

**trouble with 0.94**
 The guides for the the 0.94 ecosystem are behind to the level where I can't get the basic functionality working. I'm unable to build my own app with a html-GUI, which was working fine with your build.

- the Red pitaya run.sh script runs with errors. I left a post on the forums:
[link](http://forum.redpitaya.com/viewtopic.php?f=8&t=796)

In 0.94 the apps folder is now read-only. if I change the permissions the entire system stops working. It's costing me a lot of time I would rather spend on developing my application.

**yours**
Only think missing from your build seems to be using the I2C commands. If those work I can develop in piece ;-)

pavel-demin, 2015-10-12:
Thanks for all the reasons :-)

I didn't realize that the latest Red Pitaya release had all those problems.

However, there are some problems with adding the I2C code to my build:
 - the source codes for the 0.92 API are not open and I can't easily add the I2C functions to librp.so that I use to build my SD card image,
 - since I'm not sure if the 0.94 API is compatible with the 0.92 FPGA configuration, I can't just take the 0.94 API and rebuild it,
 - if I take the 0.94 FPGA configuration, I'll have to take the 0.94 web server and applications and we'll be back to the problems that you have with 0.94.

Looking at the new I2C code at
https://github.com/Gulzt/RedPitaya/blob/master/api/rpbase/src/i2c.h
https://github.com/Gulzt/RedPitaya/blob/master/api/rpbase/src/i2c.c
I'd say that the easiest solution would be just add these two files to your project.

There are some new return codes that are used in i2c.c and that are not in the 0.92 API:
```
/** Failed to open bus */
#define RP_EFOB   18
/** Failed to close bus */
#define RP_EFCB   19
/** Failed to acquire bus access */
#define RP_EABA   20
/** Failed to read from the bus */
#define RP_EFRB   21
/** Failed to write to the bus */
#define RP_EFWB   22
```
You can just add them to i2c.h.

What do you think about this? Any other ideas?

Gulzt, 2015-10-12:
Thank you, yes, your solution works perfectly. Really glad you helped me out, thanks a lot!

closed 2015-10-12 by pavel-demin