pavel-demin/red-pitaya-notes #114: TX mode of ExtIO_RedPitaya_TRX does not work

KlausLiebschner, 2015-11-09:
Hello Pavel and followers,
I am using Red-Pitaya and it's EXTIO together with HDSDR as a receiver successfully. Now I want to replay the recorded files with HDSDR again using the TX function of HDSDR version 2.70.I follow the explanations on "http://pavel-demin.github.io/red-pitaya-notes/sdr-transceiver/" exactly, but when I want to enable the "SDR TX Support" in the "TX menu" of HDSDR, then I get the error message "Actual DLL does not support TX!
Any idea what is going wrong ?

Klaus

pavel-demin, 2015-11-09:
Hello Klaus,

The error message is correct. This ExtIO does not support TX. I could not find any documentation or example of how to implement this functionality.

Currently, TX mode is only working with GNU Radio.

Alternatively,  you can use the SDR transceiver compatible with HPSDR and the software developed by the OpenHPSDR project.

If you could point me to any example showing how to implement TX mode compatible with HDSDR, I'll add it to the ExtIO plug-in.

Best regards,

Pavel

pavel-demin, 2015-11-09:
Just found a similar issue in the ExtIO_HackRF project:

https://github.com/jocover/ExtIO_HackRF/issues/6

Looks like HDSDR has some problems with the TX functionality and it could not be easily added to the ExtIO_HackRF plug-in.

Maybe it only works with the SDR transceivers connected to the sound card.

KlausLiebschner, 2015-11-10:
Hello Pavel,
yes I wanted to mention, that on the HDSDR page (http://www.hdsdr.de/faq.html), there is a kind of a documentation including an example. I assume, that the HDSDR GUI does not support to replay the recorded files, but that the TX function is rather a "voice transmission" function from a microphone input and not intended for replaying baseband IQ files. That's very unfortunate, but I will try work with the HPSDR.

Thanks, Klaus

pavel-demin, 2015-11-10:
Hello Klaus,

Actually, this plug-in is based on the HDSDR example.

It's explained in my [SDR receiver related notes] (http://pavel-demin.github.io/red-pitaya-notes/sdr-receiver/#user-interface). There are links to [HDSDR FAQ](http://www.hdsdr.de/faq.html) and [ExtIO_Demo_101.zip](http://hdsdr.de/download/ExtIO/ExtIO_Demo_101.zip) in the User interface part of these notes.

Unfortunately, looking at the HDSDR example it's not clear how to implement the TX functionality.

Best regards,

Pavel

KlausLiebschner, 2015-11-12:
I see.
After considering my options to have a Receiver/Transmitter HW/SW, I decided to have a look into the GNU Radio Companion. As per now, I tried to duplicate your SDR-tranceiver setup, but I get an error message:

Executing: "/home/klaus/red-pitaya-notes/projects/sdr_transceiver/gnuradio/trx_am.py"

Traceback (most recent call last):
  File "/home/klaus/red-pitaya-notes/projects/sdr_transceiver/gnuradio/trx_am.py", line 251, in <module>
    tb = trx_am()
  File "/home/klaus/red-pitaya-notes/projects/sdr_transceiver/gnuradio/trx_am.py", line 103, in __init__
    corr=0
  File "/home/klaus/red-pitaya-notes/projects/sdr_transceiver/gnuradio/red_pitaya.py", line 45, in __init__
    self.connect(blocks.file_descriptor_source(gr.sizeof_gr_complex, fd), self)
  File "/usr/lib/python2.7/dist-packages/gnuradio/blocks/blocks_swig0.py", line 766, in make
    return _blocks_swig0.file_descriptor_source_make(*args, **kwargs)
TypeError: Required argument 'repeat' (pos 3) not found

Any idea, what is wrong ? Any idea what to do fro having a receive and a transmit path ?

Klaus

pavel-demin, 2015-11-12:
What OS and GNU Radio version are you using?

I've just double checked that trx_am.grc works with Ubuntu 14.04 and GNU Radio 3.7.2.

Google finds the following bug report with an error message very similar to yours:
https://lists.gnu.org/archive/html/discuss-gnuradio/2013-08/msg00525.html

I'd suggest to install a more recent GNU Radio version or to apply the fix included in the above bug report.

pavel-demin, 2015-11-12:
> Any idea what to do for having a receive and a transmit path?

Have you tried [SDR transceiver compatible with HPSDR](http://pavel-demin.github.io/red-pitaya-notes/sdr-transceiver-hpsdr/)?

It works with several excellent SDR programs that support the HPSDR/Metis communication protocol.

There are currently two ways to setup this SDR transceiver:
 - download and install small customized SD card image following the instructions from the above link
 - or download the latest Red Pitaya SD card image from http://redpitaya.com/quick-start and install the "SDR transceiver compatible with HPSDR" application from the Red Pitaya application marketplace http://bazaar.redpitaya.com

Onnosr, 2015-12-02:
Dear Pavel,

I have the red pitaya now and installed the sdr transceiver compatible with HPSDR.
I installed Powersdr mrX (latest version von November 2015).
I connected a small homemade PA (input 100 mw output 30 watt)

And I have to say CONGRATULATIONS for your work.
It works "tadelos" on all (in powersdr unlocked tx) frequencies.
I made already a couple of nice QSO"s
And all the reports I get are more than ok especially the quality of the modulation (in ssb)
  
I have a couple of suggestions.
I know that I can trigger an external pa with the cat command- ddutil etc in powersdr.

But is it possible for you to implement the following.
Use in2 as a ptt switch  (low or high put the system to transmit)
use out2 as a ptt output (low or high activates via a transistor the ptt relais from the pa amp.

My name is 
Onno Bakker.
QTH Rotterdam
Callsign PA0ONO

Thanks in advance for thinking about it.

Onno

pavel-demin, 2015-12-02:
Dear Onno,

Thank you for the feedback and suggestions.

In the current version, there is a PTT-out signal connected to pin DIO0_P of the [extension connector E1](http://wiki.redpitaya.com/index.php?title=User_Manual#Extension_connectors).

In2 is already used as input for RX2. It can be used for diversity reception with PowerSDR mRX PS.

It's possible to use another pin of the extension connector E1 as PTT-in. Is it OK?

Best regards,

Pavel

Onnosr, 2015-12-02:
Dear Pavel,

Thank you for your fast reply.

Indeed pin DIO0_P  goes high (3 volt) if I put powersdr into transmit.
Now a simpel optocoupler to trigger the pa relais.
Do you know the maximum current allowed from that pin ?

And for ppt in any other pin from connector E1 can be used.
I leave it up to you to choose a pin (exept the 3.3 volt pin 1 and  pin DIO0_P  )

Thanks in advance for your reply.

Very best regards and 73

BTW are you a radioham ?

Onno

   

From: Pavel Demin 
Sent: Wednesday, December 02, 2015 22:18
To: pavel-demin/red-pitaya-notes 
Cc: Onnosr 
Subject: Re: [red-pitaya-notes] TX mode of ExtIO_RedPitaya_TRX does not work (#114)

Dear Onno,

Thank you for the feedback and suggestions.

In the current version, there is a PTT-out signal connected to pin DIO0_P of the extension connector E1.

In2 is already used as input for RX2. It can be used for diversity reception with PowerSDR mRX PS.

It's possible to use another pin of the extension connector E1 as PTT-in. Is it OK?

Best regards,

Pavel

pavel-demin, 2015-12-03:
> Do you know the maximum current allowed from that pin?

According to the [Zynq-7000 datasheet](http://www.xilinx.com/support/documentation/data_sheets/ds187-XC7Z010-XC7Z020-Data-Sheet.pdf), the maximum current is 10 mA.

Onnosr, 2015-12-03:
Dear Pavel,
Thanks for your reply.
I use now as optocoupler a PC817 iwith a resistor in serie ,1 kohm from  pin DIO0_P. to the led anode Cathode goes to pin 1 ground.. Running about 8 ma in tx
The transsistor part is driving a C 3866  (NPN silicium tor that is driving the tx/rx relais .
It is sitting on a very small doughterboard plugable into extension connector ex1.
The Rp is driving a small 40 watt amp that is driving a big 1kw amp.
No hf problems in the RP.
You have to know that all reports I get love the audio quality and also the signal quality.
I hope that the the people from powersdr mrX will unlock in the coming version the 60 meter band.
Now it is unlocked only on 6 channels (USA and GB)
In the Netherlands we have since December 2 (yepp yesterday) that band from 3535.0 to 3554.o MHz
free in al modes with a maximum of 100 watt pep and only for full licenced hams.

Many thanks again for your input.

Onno
pa0ono

pavel-demin, 2015-12-03:
> I hope that the the people from powersdr mrX will unlock in the coming version the 60 meter band.

I did not try it myself, but I think that it's possible to modify the band plan by editing the `BandText` tags in  `AppData\Roaming\FlexRadio Systems\PowerSDR mRX PS\database.xml`.

There is also a program called Band Text Editor that can be downloaded from the following link:
http://k9dur.rnaconsultingservices.com/powersdr_utilities.html

Onnosr, 2015-12-04:
I know that program but asfaik it is only editing the view of what you see on the screen corresspondending with the band .
It is not editing the powersdr.exe file.  
So you see on the screen that it is ok to transmit but still it is unlocked and no transmitting.

Onno
PA0ONO

Onnosr, 2015-12-06:
Dear Pavel,

I got the unlock file.
And  it is working perfect.
The second receiver for diversity is doing a great job too.

Best regards,

Onno
pa0ono

From: Pavel Demin 
Sent: 04 December, 2015 00:03
To: pavel-demin/red-pitaya-notes 
Cc: Onnosr 
Subject: Re: [red-pitaya-notes] TX mode of ExtIO_RedPitaya_TRX does not work (#114)

  I hope that the the people from powersdr mrX will unlock in the coming version the 60 meter band.

I did not try it myself, but I think that it's possible to modify the band plan by editing the BandText tags in AppData\Roaming\FlexRadio Systems\PowerSDR mRX PS\database.xml.

There is also a program called Band Text Editor that can be downloaded from the following link:
http://k9dur.rnaconsultingservices.com/powersdr_utilities.html

Onnosr, 2015-12-17:
Dear Pavel,

I do have a little automatic antenna tuner, the T1 from Elecraft.
The manual is attached.

To run it with powersdr I made a arduino nano board as suggested by Matthew Robinson.
see  Github   :  https://gist.github.com/mttrb/4712611
and  : http://blog.zensunni.org/ 

Do you have a possibility to implement that is  the sdr transceiver and use some pin
 from ex1 to connect direct (or via optocoupler)  to the T1 (or other antennatuner) ? 

Best regards,

Onno
PA0ONO

.

[T1 owners man rev A2.pdf](https://github.com/pavel-demin/red-pitaya-notes/files/65331/T1.owners.man.rev.A2.pdf)

pavel-demin, 2015-12-17:
Dear Onno,

If you have already Arduino, then it would be easier to connect it to Red Pitaya via UART. Then you can modify the code at lines 217-233 in [`sdr-transceiver-hpsdr.c`](https://github.com/pavel-demin/red-pitaya-notes/blob/master/projects/sdr_transceiver_hpsdr/server/sdr-transceiver-hpsdr.c#L217) to send the RX and/or TX frequencies to UART.

Here is an example showing how to use UART from a program written in C:
https://salilkapur.wordpress.com/2013/03/08/communicating-with-arduino-using-c/

The UART port on Red Pitaya should be accessible as `/dev/ttyPS1`.

If you're using the SD card image downloaded from http://redpitaya.com/quick-start/, then you can modify and compile `sdr-transceiver-hpsdr.c` directly on Red Pitaya.

Best regards,

Pavel

Onnosr, 2016-01-11:
Dear Pavel,

I tried the RP with GSM G4 connection.
After editing the card in linux with leafpad the RP has a static IP.
I can connect to the RP ok: the oscilloscope and the spectrum analyzer are working remotely over internet ok.
The hpsdr app not.  
The metis protocol has to see port 1024 open to go over internet.
Is it possible that 1024 is not set open or is there an other point that I not see.

Further I understand that on ext1 a cw paddle and or key can be connected. Can you tell which pins has to be used. 

HPSDR has a CWX feature,  using the keyboard shows ok but does not set the RP to transmit.

Last point,  HPSDR can work with ALEX, a low pass filter box. Is there a possibility to implement that using the 7 output pins on EXT1 ?

In The Netherlands I was the first running the RP as SDR tranceiver. This was and still is a great boost.
Many hams want to come now with the RP on the air and there are already a couple.
Everybody is hot for it.

Very best regards

Onno  PA0ONO

 

 

From: Pavel Demin 
Sent: Thursday, December 17, 2015 13:51
To: pavel-demin/red-pitaya-notes 
Cc: Onnosr 
Subject: Re: [red-pitaya-notes] TX mode of ExtIO_RedPitaya_TRX does not work (#114)

Dear Onno,

If you have already Arduino, then it would be easier to connect it to Red Pitaya via UART. Then you can modify the code at lines 217-233 in sdr-transceiver-hpsdr.c to send the RX and/or TX frequencies to UART.

Here is an example showing how to use UART from a program written in C:
https://salilkapur.wordpress.com/2013/03/08/communicating-with-arduino-using-c/

The UART port on Red Pitaya should be accessible as /dev/ttyPS1.

If you're using the SD card image downloaded from http://redpitaya.com/quick-start/, then you can modify and compile sdr-transceiver-hpsdr.c directly on Red Pitaya.

Best regards,

Pavel

pavel-demin, 2016-01-11:
Dear Onno,

What do you mean by "GSM G4 connection"? Do you have a G4 modem connected to your Red Pitaya?

Anyway, the discovery protocol most probably won't  work over an internet connection.  The discovery protocol uses UDP multicast and multicast packages aren't routed.

You can try to configure PowerSDR to use static IP address as described in:
http://pa3ang.nl/wp/archives/5245

CW keys and ALEX are not supported yet.

I've already implemented an IP core for the ALEX protocol:
https://github.com/pavel-demin/red-pitaya-notes/tree/master/cores/axis_alex_v1_0
but I can't find time to add it to the the sdr_transceiver_hpsdr project.

Best regards,

Pavel

Onnosr, 2016-01-12:
Dear Pavel,

I know pa0ang Johan. He was triggert by me to write his blog as you can read in his blog.

My RP (card)  has already a static IP ( know how to do that). and is on the segment from my router.
My router has (oc) an outside (what is my ip) address

I can connect to the RP via a “personal hotspot gsm iphone on G4 to a laptop or using a mifi G4 capable router. .    
And I can connect to my RP using a next door wifi network (I know the wpa Key)

In both situations I go to my own (outside ) IP address and see my RP startpage.

Everything works except the hpsdr.
Do you mean the discovery protocol from PowerSDR mrX PS is using UDP multicast ?

Best regards,

Onno

. en 
From: Pavel Demin 
Sent: Tuesday, January 12, 2016 00:08
To: pavel-demin/red-pitaya-notes 
Cc: Onnosr 
Subject: Re: [red-pitaya-notes] TX mode of ExtIO_RedPitaya_TRX does not work (#114)

Dear Onno,

What do you mean by "GSM G4 connection"? Do you have a G4 modem connected to your Red Pitaya?

Anyway, the discovery protocol most probably won't work over an internet connection. The discovery protocol uses UDP multicast and multicast packages aren't routed.

You can try to configure PowerSDR to use static IP address as described in:
http://pa3ang.nl/wp/archives/5245

CW keys and ALEX are not supported yet.

I've already implemented an IP core for the ALEX protocol:
https://github.com/pavel-demin/red-pitaya-notes/tree/master/cores/axis_alex_v1_0
but I can't find time to add it to the the sdr_transceiver_hpsdr project.

Best regards,

Pavel

pavel-demin, 2016-01-12:
> Do you mean the discovery protocol from PowerSDR mrX PS is using UDP multicast?
It's exactly what I mean.

pavel-demin, 2016-01-12:
Here is the screenshot from the [blog post](http://pa3ang.nl/wp/archives/5245) linked from my previous comment showing how to setup static IP address in PowerSDR mrX PS:
[image]

KlausLiebschner, 2016-01-18:
Hello Pavel,
I would like to follow up on this topic. We were talking about the TX function of your SDR Transceiver project. I assumed that I misunderstood TX in this case. Now I understand that using the TX with HDSDR means, that "pure audio data" (voice) from a microphone is sent to the Red Pitaya and then AM modulated and upconverted to the desired RF-frequency. I assumed, that TX sends pure RF IQ data similar as RX receives pure IQ data according the selected frequency&sampling rate .

My desire is to use the Red Pitaya as RF IQ recording and streaming device, so I do not need the AM modulation. I would like to use the Red Pitaya as indicated in the attached GNUradio project file. Would it be a lot of work for you to create another SDR Transceiver project which supports this function ?

Klaus
[trx_am_IQ.zip](https://github.com/pavel-demin/red-pitaya-notes/files/94215/trx_am_IQ.zip)

pavel-demin, 2016-01-18:
Hello Klaus,

Thanks for the information about HDSDR.

> to use the Red Pitaya as RF IQ recording and streaming device

Wouldn't it be as simple as connecting the Red Pitaya Source block directly to the File Sink block like it's shown on the following screenshot?
[image]

KlausLiebschner, 2016-01-19:
Hi Pavel,
indeed you are correct! It seems, as if HDSDR saves the file in a different format as GNUradio wants to sent directly to the Red Pitaya. As an experiment, I recorded with GNUradio a 1kHz AM tone on the desired frequency and then replayed it also with GNUradio. I tried 100kHz, 250kHz and 500kHz sampling rate. With 500kHz, recording does not work. I get the following error messages in GNUradio-companion:
Generating: "/home/red-pitaya-notes/projects/sdr_transceiver/gnuradio/trx_am.py"

Executing: "/home/red-pitaya-notes/projects/sdr_transceiver/gnuradio/trx_am.py"

Using Volk machine: avx_64_mmx_orc
gr_remez: insufficient extremals -- cannot continue
Traceback (most recent call last):
  File "/home/red-pitaya-notes/projects/sdr_transceiver/gnuradio/trx_am.py", line 174, in <module>
    tb = trx_am()
  File "/home/red-pitaya-notes/projects/sdr_transceiver/gnuradio/trx_am.py", line 111, in __init__
    audio_stop=5500,
  File "/usr/lib/python2.7/dist-packages/gnuradio/analog/am_demod.py", line 54, in __init__
    60)	          # Stopband attenuation
  File "/usr/lib/python2.7/dist-packages/gnuradio/filter/optfir.py", line 55, in low_pass
    taps = filter.pm_remez (n + nextra_taps, fo, ao, w, "bandpass")
  File "/usr/lib/python2.7/dist-packages/gnuradio/filter/filter_swig.py", line 449, in pm_remez
    return _filter_swig.pm_remez(*args, **kwargs)
RuntimeError: gr_remez: insufficient extremals -- cannot continue

Besides that, what IQ file format does GNUradio/Red Pitaya expect ? 

Klaus

pavel-demin, 2016-01-19:
The error message indicates that the the 'AM Demod' block can't design a FIR filter corresponding to the 'Audio Pass' and 'Audio Stop' parameters. Since you previously mentioned that you did not need AM modulation, you could solve this problem by just removing the 'AM Demod' block from your flow graph.

If I'm not mistaken, the I/Q file format is very simple. It's just a binary dump of an array of float. The I and Q samples are interleaved: I, Q, I , Q, ....

KlausLiebschner, 2016-01-19:
OK - I just have the AM demod part in it as a verification backup - I basically short my TRX path with the RX path to verify what I am sending.
The IQ format is clear to me. The file is a 16bit one and I have read, that "File source" uses a 32bit format. So I use now "File Source [Output type:short]" -> "IShort To Complex" -> "Throttle" -> "Red Pitaya Sink". The GUI Frequency Sink now looks as expected and I can also hear the AM station at the expected frequency. However, the quality is very bad. I think, that the format conversion is not yet correct - there seems to be an incorrect bitshift, which hurts my RF Signal to Noise ratio. Maybe I need to cut off the header of the .wav file and use a RAW .PCM file.....

KlausLiebschner, 2016-01-20:
Pavel, sorry to bother you. During my investigation on the format problem, I am encountering a weird situation. I have two SD-cards. One with your SDR receiver and one with your SDR Transceiver. With the SDR receiver SD-card , I  can connect with HDSDR to Red Pitaya and can record IQ files. With the SDR Tranceiver SD-card I can connect with Gnuradio to the Red Pitaya - but I can NOT connect with the SDR Tranceiver SD-card to HPSDR and vice versa. Are those two development projects not compatible ?

Klaus

pavel-demin, 2016-01-20:
There are two different ExtIO plug-ins:
 - ExtIO_RedPitaya that works with the SDR receiver,
 - ExtIO_RedPitaya_TRX that works with the SDR tranceiver.

Are you using the ExtIO_RedPitaya_TRX plug-in when you're testing HDSDR with SDR transceiver?

KlausLiebschner, 2016-01-20:
Yes, I am using both of them with either SDR Receiver or Transceiver.I meanwhile found the problem.... a network problem with the IP address....
I now wonder if the IQ file format difference is regarding the data type difference between the Windows PC and the data types on the LINUX/GNUradio platform. Little & Big  endian format and bit&byte shitf or rotation......

pavel-demin, 2016-01-20:
> I now wonder if the IQ file format difference is regarding the data type difference between the Windows PC and the data types on the LINUX/GNUradio platform.

As far as I know, the endianness is CPU dependent.

BTW, GNU Radio can be used under Windows thanks to the Pothos project. You can find the instructions at https://github.com/pothosware/PothosSDR/wiki/GNURadio

KlausLiebschner, 2016-01-21:
WOOOWW - super, I already have it running. However, there are some questions, which are not 100% clearly explained:
1.) To use portaudio you first need to edit config in %appdata%/.gnuradio/config.conf (if the folder doesn't exist then create it) Where does the config.conf file need to be saved ? C:\PothosSDR\config.conf  ?
2.) Why using gnuradio-companion when having the Pothos GUI ? Differences? Advandages? Disadvantages?
3.) If using gnuradio-companion - How to get the Red-Pitaya Devices into the right side Category Page ?
4.) If using the Pothos GUI - How to get the SDR-sink/source configured for Red-Pitaya ?
Would be wonderful, if you could get me up to speed :)
Klaus

pavel-demin, 2016-01-21:
> 1.) To use portaudio you first need to edit config in %appdata%/.gnuradio/config.conf (if the folder doesn't exist then create it) Where does the config.conf file need to be saved ? C:\PothosSDR\config.conf ?

No need to copy this file to `C:\PothosSDR`. Just keep it in `%appdata%\.gnuradio\config.conf` (`AppData\Roaming\.gnuradio\config.conf`)

> 2.) Why using gnuradio-companion when having the Pothos GUI ? Differences? Advandages? Disadvantages?

Both are great tools developed by the same author (Josh Blum). Personally, I slightly prefer the gnuradio-companion user interface. However, Pothos has nicer plotting widgets. Maybe I just need to get more experienced with Pothos.

> 3.) If using gnuradio-companion - How to get the Red-Pitaya Devices into the right side Category Page ?

Under Windows, I'm using the osmocom blocks. An example flow graph can be downloaded from [this link](https://googledrive.com/host/0B-t5klOOymMNfmJ0bFQzTVNXQ3RtWm5SQ2NGTE1hRUlTd3V2emdSNzN6d0pYamNILW83Wmc/SDR/trx_ssb_osmo_wx.grc).

> 4.) If using the Pothos GUI - How to get the SDR-sink/source configured for Red-Pitaya ?

Most probably, [Red Pitaya driver](https://github.com/pothosware/SoapyRedPitaya) will be included in the next release. Until then, you can try to download my pre-built RedPitaya.dll from [this link](https://googledrive.com/host/0B-t5klOOymMNfmJ0bFQzTVNXQ3RtWm5SQ2NGTE1hRUlTd3V2emdSNzN6d0pYamNILW83Wmc/SDR/RedPitaya.dll) and copy it to `C:\Program Files\PothosSDR\lib\SoapySDR\modules`

An example flow graph can be downloaded from [this link](https://googledrive.com/host/0B-t5klOOymMNfmJ0bFQzTVNXQ3RtWm5SQ2NGTE1hRUlTd3V2emdSNzN6d0pYamNILW83Wmc/SDR/pothos_example.pth).

KlausLiebschner, 2016-01-25:
Thanks Pavel,
I still have the problem, that I am not able to replay HDSDR 16bit PCM files with the Red Pitaya under gnuradio. I have summarized my setup and my test procedure in tha attached .pdf document. Maybe you can have a look at it and give me a feedback on it.

Klaus
[Evaluation of the Red Pitaya as SDR IQ Streaming Device.pdf](https://github.com/pavel-demin/red-pitaya-notes/files/103168/Evaluation.of.the.Red.Pitaya.as.SDR.IQ.Streaming.Device.pdf)

pavel-demin, 2016-01-25:
Thanks for the detailed description of the problem.

I'd say that the problem is with PCM16. The I/Q samples sent from Red Pitaya to HDSDR are in the float format. It would be better to use this format directly without converting it to PCM16.

Could you try the following?
 - set Sample Type to FLT32 in the HDSDR recording settings,
 - set Output Type to Complex in the File Source settings,
 - remove IShort to Complex and Throttle  blocks.

KlausLiebschner, 2016-01-26:
Ok that works !
But then I have a real problem. In our Lab we have an S1000 IZT Recorder/Player (https://www.izt-labs.de/en/products/category/signal-generators/product/izt-s1000/). It has an option ARB16, which can only play 16bit files (besides the IZT format). We already have recorded lots of 16bit files (Terabytes) - and I wanted to establish the Red Pitaya as an additional mobile record/play device.My options are now to convert the files, which doubles the amount of storage - which is impossible.
I was a little bit confused that the format is a FLT32 format, because the Red Pitaya Source/Sink block has a COMPLEX IN/OUT connector. Is there a possibility to have a 16bit input format ?

Klaus

pavel-demin, 2016-01-26:
Comparing the GRC screenshots on pages 4 and 5 of the PDF document, it's clear that Red Pitaya Sink works when the signal's power is below zero. So, to make your configuration with IShort to Complex block working, it should be enough to divide the samples by (2^15). The easiest would be to add Multiply Const block and set its Constant parameter to 3.051758e-05 (1/2^15).

KlausLiebschner, 2016-01-26:
You are the best ! :-) 
By the way - what's your profession ?

Thanks again, Klaus

pavel-demin, 2016-01-26:
Thanks for your kind words. I'm glad to know that it finally worked for you. Is it OK to close this issue?

> By the way - what's your profession?

I am a particle physicist and IT engineer.

pavel-demin, 2016-01-26:
> I wanted to establish the Red Pitaya as an additional mobile record/play device

BTW, it's possible to modify the [sdr-transceiver server](https://github.com/pavel-demin/red-pitaya-notes/blob/master/projects/sdr_transceiver/server/sdr-transceiver.c) and read/write the files from/to a USB disk directly connected to Red Pitaya. It's also possible to mount a network share or read files from a web server.

The command line could be as simple as
```
sdr-recorder channel rate frequency filename.dat
sdr-player channel rate frequency filename.dat
```
Reading files from a web server could be done as
```
curl http://192.168.1.2/filename.dat | sdr-player channel rate frequency
```

closed 2016-02-22 by pavel-demin