pavel-demin/red-pitaya-notes #125: Enhancement request: provide SoapySDR module for Red Pitaya

dc1rdb, 2015-12-04:
A vendor and platform neutral SDR support library including remote access: https://github.com/pothosware/SoapySDR/wiki

pavel-demin, 2015-12-04:
Since Pothosware/SoapySDR works with gr-osmosdr and gr-osmosdr works with Red Pitaya SDR transceiver, it should be possible to use Red Pitaya SDR transceiver with Pothosware/SoapySDR via the gr-osmosdr interface:
https://github.com/pothosware/SoapyOsmo/wiki

You can download gr-osmosdr with support for the Red Pitaya SDR transceiver from my fork of the gr-osmosdr repository:
https://github.com/pavel-demin/gr-osmosdr

Then you can build gr-osmosdr following the instructions from:
http://sdr.osmocom.org/trac/wiki/GrOsmoSDR

dc1rdb, 2015-12-14:
Issue is being followed at https://github.com/pothosware/SoapySDR/issues/55, so I'll close this one, ok?

closed 2015-12-14 by dc1rdb