pavel-demin/red-pitaya-notes #135: Question: Using IN2 and OUT2 for audio processing

dc1rdb, 2015-12-09:
Hi Pavel, on your "Embedded SDR Transceiver", would it be possible to configure IN2 and OUT2 as microphone input and headphone/speaker output to get rid of the external sound card?
Really appreciate your work - thanks a lot!
Kind regards,
Robert DC1RDB

pavel-demin, 2015-12-09:
Hi Robert,

Thanks for your interest in my embedded SDR transceiver project.

I think that it's already possible to test this kind of configuration with the current version.

It should be enough to set the frequency to zero for the source and sink blocks connected to IN2 and OUT2.

Best regards,

Pavel

dc1rdb, 2015-12-10:
Thanks Pavel, will try this as soon as I get the chance!

closed 2016-03-10 by pavel-demin