pavel-demin/red-pitaya-notes #136: cfg_test write to /dev/mem hangs pitaya

mhanuel26, 2015-12-09:
Dear pavel,

First of all,
I have to thank you for all the hard work made for this framework to work.

I have a problem that might be silly but I really would appreciate your comment,
I am trying to use the example code for blinking led on RP accessing /dev/mem just to confirm it's working and for the case of cfg_test project the RP hangs after loading the cfg_test.bit file generated using make command. 
Other projects such packetizer_test or red_pitaya_0_92 it works. For those I do the same, load PL and then run ./blink example.

It might be some missing AXI configuration that is present on the top .tcl files for those projects that cfg_test don't have?

Best,

pavel-demin, 2015-12-09:
It's been quite some time since I last tested cfg_test.

What is strange is that packetizer_test works. Since packetizer_test includes cfg_test, they should behave similarly.

Could you please remind me what is the blink example? If it's one of the Red Pitaya examples, then it only works with red_pitaya_0_92 or with the default Red Pitaya ecosystem.

pavel-demin, 2015-12-09:
I've just checked that the address range for cfg_test is 0x40000000-0x40000080.

mhanuel26, 2015-12-10:
Hello Pavel,

I am actually writing to address 16, since you have a slice that takes out 7 bits from [134:128] feeded from cfg IP output [1023:0]. I am using your example at 

https://0c2d3ba9cf2a7be9395f36aa24dd59103ca884e9.googledrive.com/host/0B-t5klOOymMNfmJ0bFQzTVNXQ3RtWm5SQ2NGTE1hRUlTd3V2emdSNzN6d0pYamNILW83Wmc/cfg_test/

I actually did a fresh make NAME=cfg_test all but since I have already a previous generated image running on RP (the basic led_blinker demo image) I was just testing the PL by uploading the .bit file. Due to my lack of knowledge about how the PL was integrated with the image I first thought it was something bothering the cfg_test PL, but then I did the same but this time  I uploaded the packetizer_test .bit file (same way as with cfg_test) and this time it worked. Actually I was wanting to start my app from cfg_test, but due to this issue I start using the packetizer app, and my PL app it;s working so far.

Other thing is that the image of base system is patched as your suggestion for UIO to work. It might not have anything to do, I just did

tmp/linux-xlnx-xilinx-v2015.3/scripts/dtc/dtc -I dts -O dtb -o devicetree.dtb -i tmp/led_blinker.tree patches/uio.dts

then I generate the image.

pavel-demin, 2015-12-26:
Thanks for the details about this problem! It's fixed with the following commit: https://github.com/pavel-demin/red-pitaya-notes/commit/e8daef4a732fa84eb61467fea203d2901acce860

closed 2015-12-26 by pavel-demin