pavel-demin/red-pitaya-notes #167: Accessing Digital IO using dac_player PL

JBosc, 2016-01-18:
Hi,

I'm trying to modify your dac_player's block design so I can use Digital IO. The memory mapping of your block design seems different from the original red pitaya and I can't find the good memory address to set direction then read or write IO. It seems also that register for DIO aren't mapped in block design, but I might be wrong, I don't have a lot of experience in fpga design.

What should I do to access DIO?

Thank you!

pavel-demin, 2016-01-18:
Could you please explain how you are planning to use the digital IO?

JBosc, 2016-01-18:
I want to generate a pulse of approx. 100ms each second. It will act as a chip select. Right now I already send signal with DAC each second and I want to trigger CS just before.

pavel-demin, 2016-01-18:
Please have a look at the [Scanning system](http://pavel-demin.github.io/red-pitaya-notes/scanner) project. It does almost exactly that you describe. It continuously outputs signals to DAC and generates trigger pulses on the GPIO pins. The code that calculates the DAC samples starts from line 182 in [projects/scanner/server/scanner.c](https://github.com/pavel-demin/red-pitaya-notes/blob/master/projects/scanner/server/scanner.c#L182).

JBosc, 2016-01-18:
Oh I'll check. Thanks for your help!

closed 2016-01-18 by JBosc