pavel-demin/red-pitaya-notes #168: Two buffers

YevgeniyIvanov, 2016-01-21:
Hi Pavel,
Can you help me understand how change axis_ram_writer for use two buffers.
First buffer from 0x1E000000, second buffer from 0x1E3FFFFF.
I want use trigger for change the buffer.
Yevgeniy.

pavel-demin, 2016-01-21:
Hi Yevgeniy,

I'd set the starting address of the writer to 0x1E400000 (507510784) and limit the accessible address space to 4 MB by setting ADDR_WIDTH to 19.

Then it's possible to use one or two writers.

In case of one writer, you need to set writer_0/cfg_data to 503316480 or to 507510784 depending on the state of your trigger.

In case of two writers, the second writer can be connected to S_AXI_HP1 and you need to use some kind of AXI4-Stream switch to redirect the data stream to the first or to the second writer depending on the state of your trigger.

For the switch functionality, it's possible to use two [axis_validator](https://github.com/pavel-demin/red-pitaya-notes/tree/master/cores/axis_validator_v1_0) modules:

```
# Create axis_validator
cell pavel-demin:user:axis_validator:1.0 vldtr_0 {
  AXIS_TDATA_WIDTH 32
} {
  S_AXIS input_0/M_AXIS
  trg_flag <enable vldtr_0>
  aclk ps_0/FCLK_CLK0
}

# Create xlconstant
cell xilinx.com:ip:xlconstant:1.1 const_1 {
  CONST_WIDTH 32
  CONST_VAL 503316480
}

# Create axis_ram_writer
cell pavel-demin:user:axis_ram_writer:1.0 writer_0 {
  ADDR_WIDTH 19
} {
  S_AXIS vldtr_0/M_AXIS
  M_AXI ps_0/S_AXI_HP0
  cfg_data const_1/dout
  aclk ps_0/FCLK_CLK0
  aresetn slice_1/Dout
}

# Create axis_validator
cell pavel-demin:user:axis_validator:1.0 vldtr_1 {
  AXIS_TDATA_WIDTH 32
} {
  S_AXIS input_1/M_AXIS
  trg_flag <enable vldtr_1>
  aclk ps_0/FCLK_CLK0
}

# Create xlconstant
cell xilinx.com:ip:xlconstant:1.1 const_2 {
  CONST_WIDTH 32
  CONST_VAL 507510784
}

# Create axis_ram_writer
cell pavel-demin:user:axis_ram_writer:1.0 writer_1 {
  ADDR_WIDTH 19
} {
  S_AXIS vldtr_1/M_AXIS
  M_AXI ps_0/S_AXI_HP0
  cfg_data const_2/dout
  aclk ps_0/FCLK_CLK0
  aresetn slice_2/Dout
}
```

YevgeniyIvanov, 2016-01-22:
Hi Pavel,
Thank you for fast reply.
I try read writer_0/sts_data for check now many samples wrote to memory but value is always zero.
Can you tell why “int_addr_reg“ is not incremented?
I change you code and insert this part

  input  wire [AXI_ADDR_WIDTH-1:0]   buff1_addr, // connect to const_1

  input  wire [AXI_ADDR_WIDTH-1:0]   buff2_addr, // connect to const_2

  output wire [AXI_ADDR_WIDTH-1:0]   start_addr, // head of buffer (this output for check only)

  output wire [AXI_ADDR_WIDTH-1:0]   end_addr,   // tail of buffer (this output for check only)

  output wire [31:0]                 samples,    // number of samples (this output for check only)

  …

    if(trg_flag)

    begin

      if(int_pulses_next)

      begin

         start_addr_next = buff_addr_reg;

         end_addr_next = buff_addr_reg + int_addr_reg;

         samples_next = int_addr_reg;

         if(buff_num_next)

         begin

           buff_addr_next = buff1_addr;

           buff_num_next = 1'b0;

         end

         else

         begin

           buff_addr_next = buff2_addr;

           buff_num_next = 1'b1;

         end

      end

      int_addr_next = {(ADDR_WIDTH){1'b0}};

      int_pulses_next = int_pulses_reg + 1'b1;

    end

    else

    begin

      if(int_rden_wire)

      begin

        int_addr_next = int_addr_reg + 1'b1;

      end

    end
May be this code is not correct.

P.S.
Pavel, can I write to you on Russian language?
Kind regards,

YevgeniyIvanov, 2016-01-23:
Павел, доброй ночи.
Хотел бы уточнить насчет предложенной Вами схемы записи с использованием axis_validator<https://github.com/pavel-demin/red-pitaya-notes/tree/master/cores/axis_validator_v1_0>, насколько я понимаю, после переключения с одного буфера на другой запись будет производится с текущего указателя, который после первого переключения уже не будет указывать на начало буфера. И необходимо добавить сигнал сброса указателя.
Я правильно понимаю?
Евгений.

closed 2016-02-22 by pavel-demin