pavel-demin/red-pitaya-notes #169: SDR receiver and SDR#

cedricboudinet, 2016-01-22:
Hello
I'm trying to use the example of SDR receiver and SDR#.
I followed the 'getting stared' instructions and when I run SDR#, Red Pitaya doesn't appears in Source list.
I noticed that there is a plugins.xml in sdr# directory. This file seems to list some sharpPlugins.
Should I add an entry for the plugin in that file? If so, how to proceed? I didn't manage to found out.
Thanks in advance.
Cedric

pavel-demin, 2016-01-22:
Hello Cedric,

What SDR# version are you using?

SDR# stopped supporting ExtIO plug-ins in September 2015. The latest version that supports ExtIO plug-ins is 1.0.0.1361. It can be downloaded from the [SDR receiver page](http://pavel-demin.github.io/red-pitaya-notes/sdr-receiver) or directly from [this link](https://googledrive.com/host/0B-t5klOOymMNfmJ0bFQzTVNXQ3RtWm5SQ2NGTE1hRUlTd3V2emdSNzN6d0pYamNILW83Wmc/SDR/sdrsharp_v1.0.0.1361_with_plugins.zip).

Best regards,

Pavel

cedricboudinet, 2016-01-26:
Hi
It works with the link. 
Thanks a lot

closed 2016-01-26 by cedricboudinet