pavel-demin/red-pitaya-notes #170: HamVNA instructions

TacHawkes, 2016-01-22:
Hi,

could you give a short introduction how to use HamVNA with your SDR app?
I'm using the HPSDR transceiver App from the Bazaar and installed the HamVNA software. The connection works so far but I do not get any meaningful data from the software. I assume the DUT ist connected to OUT 1 and IN 1 for measurements in transmission mode? For testing purpose I've hooked up a small LCR circuit with about 200 kHz resonance frequency which I can measure using the Bode analyzer.

pavel-demin, 2016-01-22:
I suppose that the instructions at http://www.dxatlas.com/hamvna/ could be directly applied when using this program with Red Pitaya. 

Looking at these instructions I'd say that connecting a DUT directly to OUT1 and IN1 won't work.It should be connected via a series fixture or a reflection bridge.

Last week, I found that the VNA functionality required some additional code in the FPGA configuration and in the server running on Red Pitaya. I've added the missing parts and updated the SDR/HPSDR bazaar application and the SD card image. Please make sure that you're using the latest version (0.94-861) of the SDR/HPSDR application.

I'd be very interested to make HamVNA program usable with Red Pitaya. So, it would be great if you could test it with a series fixture or a reflection bridge and explain me what do you think is going wrong.

TacHawkes, 2016-01-23:
Hi,

I will try if with a series fixture as soon as I have one available. For now I just wanted to give the software a quick test. I should mention that I've connected 50 Ohms in parallel to IN1 to have impedance matching between OUT 1 and IN 1 (actually I do not understand why the RP has no internal "50 Ohms mode") but of course this does not replace the fixture. Still I would have expected to measure at least something for the magnitude/gain parameter but instead I just get the flat noise-floor.

pavel-demin, 2016-01-23:
I've just done three frequency sweeps with the following elements connected to OUT1 and IN1:
[image]

Here are the frequency responses that I've obtained:

[image]
[image]
[image]

TacHawkes, 2016-01-23:
Ok this looks completely different ;) As I said I get basically no signal, just the noisefloor. I will try it again.

pavel-demin, 2016-01-23:
If you get something flat, then it's probably the SDR/HPSDR version which is not up-to-date on your Red Pitaya.

TacHawkes, 2016-01-23:
Ok I've updated the Ecosystem and your Bazaar App. I still get the flat response using your first circuit (OUT 1 to IN 1 with 50 Ohms between signal and GND)

TacHawkes, 2016-01-24:
Just to be sure: I do not need you custom image from your blog, do I? I have updated to the most recent 0.94 ecosystem, and installed all apps from Bazaar again, so your SDR/HPSDR app is now on version 0.94-861. I have no clue what could be wrong at the moment, as analyzing a DUT with the bode analyzer app works flawless.

pavel-demin, 2016-01-25:
> I do not need you custom image from your blog, do I?

No, not necessary. I used SDR/HPSDR 0.94-861 installed via bazaar to produce the above plots.

Could you check with an oscilloscope if you have any signal coming from OUT1?

pavel-demin, 2016-05-12:
After several only partially successful attempts to make my SDR application reliably work with HamVNA I developed a new VNA application for Red Pitaya:
http://pavel-demin.github.io/red-pitaya-notes/vna/

It's now available from the Red Pitaya application marketplace.

closed 2016-05-12 by pavel-demin

stevec5000, 2017-07-07:
Pavel can you tell me how to get the Red Pitaya to work with PowerSDR?
In your Red Pitaya Notes, SDR Receiver, you say:"The implementation of the SDR receiver is quite straightforward" and "The I/Q data is transmitted via TCP to the SDR programs such as SDR# and HDSDR".
but it doesn't seem to be any of these!  I know of no way to get it working with SDR# so I've been trying to get HDSDR to work with the Red Pitaya SDR app but haven't been able to find any instructions or anyone who will admit to knowing anything about it!
There are video showing it working on Youtube but they seem to all be fakes, like the ones for quadcopters that can't really fly such as the Lily drone.
As for HDSDR, I've installed the program in Windows 10 but when I start the Red Pitaya app the SDR program can't seem to be able to find the Red Pitaya.
The hardest part about this is finding anyone who will admit to knowing anything about it!
When I tried to run the Red Pitaya SDR radio app it says: "HAMlab is now running in SDR mode. To use it as a radio please connect to it with PowerSDR HAMlab edition. That can be downloaded from this link." and they give a link to the download site at Flex Radio.
So I installed Power SDR Open HPSDR like it wanted on the PC but when I try to run it by clicking the Power button I get a message saying: 
"No Charly 25/Hamlab device detected on the network, Please power up your Charly 25 / Hamlab before using the autosensing mode!"
Do you know what this means, what a Hamlab is or what a Charly 25 is or what autosensing mode is?
I have an SDR USB dongle running with SDR Sharp and that was easy compared to this!

In Power SDR/OPENHPSDR under Setup, Hardware Config., when I click the Hamlab option it has a window for the RedPitaya IP and Mac Address but it's blank and there's no way to enter the settings into it!
The program, which was downloaded from Flex Radio Systems, like it says to do in the Red Pitaya SDR app, is ver. 2017-02-18 Commit 15680ad.
Did I get a bad version of the program that doesn't work?
How can I get PowerSDR to recognize and work with the Red Pitaya?
Thanks.

pavel-demin, 2017-07-09:
To @stevec5000:
Since your questions aren't related to HamVNA, I've moved them to a new issue: https://github.com/pavel-demin/red-pitaya-notes/issues/490.

Rolf-L, 2017-07-11:
Hi Pavel,
I tried to use HamVNA at IN1 and OUT1. 
I used SDR/HPSDR 0.94-861 installed via bazaar, but there is no output signal at OUT1 or at OUT2. 
I checked this with my oscilloscope running "Sweep". There is  just the red noisefloor  and a blue signal line at -110dBFS. Any idea what's wrong?

Your RP VNA program works fine. 
Is there a RP VNA version available, using IN2 and OUT2 instead of IN1 and OUT1?. 
I'm using IN1 and OUT1 with my HPSDR TRX.

[image]

pavel-demin, 2017-07-23:
Hi Rolf,

> I used SDR/HPSDR installed via bazaar, but there is no output signal at OUT1 or at OUT2.

As I mentioned in my previous comment, my attempts to make my SDR application reliably work with HamVNA were only partially successful. Then I removed all the VNA related code from the 'SDR transceiver compatible with HPSDR' application for Red Pitaya. So, HamVNA can't be currently used with the 'SDR transceiver compatible with HPSDR' application for Red Pitaya.

> Is there a RP VNA version available, using IN2 and OUT2 instead of IN1 and OUT1?.

Using IN2 requires a very simple modification of the Python code in vna.py: `adc1` should be replaced with `adc2` [on line 212 in vna.py](https://github.com/pavel-demin/red-pitaya-notes/blob/master/projects/vna/client/vna.py#L212).

Using OUT2 requires some modifications of the FPGA configuration.