pavel-demin/red-pitaya-notes #171: debian ecosystem question

mhanuel26, 2016-02-04:
Hello Pavel,

I have build the SD image for ecosystem but I was looking for scpi server without success. 
Could you please let me know what are the differences regarding the ecosystem with the RP build process. 
https://github.com/RedPitaya/RedPitaya/blob/master/OS/debian/README.md
I will appreciate your answer,

Best.

pavel-demin, 2016-02-04:
Hello Manuel,

My Debian based SD card image was a prototype of the current Red Pitaya SD card image. The Red Pitaya developers took my scripts and adapted them to their requirements.

To build my Debian based SD card image with the Red Pitaya ecosystem, I'm still using version 0.92 of the Red Pitaya ecosystem. The latest Red Pitaya ecosystem is a mix of 0.93 and 0.94.

As a workaround to the missing scpi server, you can try to download the following zip file and copy `bin/scpi` and `lib/libscpi.so` from this file to the Debian based SD card:
http://downloads.redpitaya.com/downloads/ecosystem-0.92-65-35575ed.zip

Best regards,

Pavel

mhanuel26, 2016-02-04:
Dear Pavel, 

I tried the workaround copying both files (to /opt/bin and /opt/lib) and it gave me the following message
_
Runtime error: calib_ReadParams(&calib) returned 1 at calib.c:29Runtime error: calib_Init() returned 1 at rp.c:40_

Notice that in your 0.92 ecosystem version the library file was there in /opt/lib/libscpi.so, I replaced with the one in the .zip file.

In terms of Vivado project and TCL files, the difference is only the version (0.92 and 0.93/0.94) ? Do you have some twists to the 0.92. What would be the way to port the 0.94 ?

I really appreciate your response,

pavel-demin, 2016-02-04:
Looks like something is missing for the SCPI application. Unfortunately, I don't have any idea how it could be fixed.

> Do you have some twists to the 0.92.

I've added a couple of fixes for the timing problems and renamed a few ports to match my other projects. You can find the history of the modifications at:
https://github.com/pavel-demin/red-pitaya-notes/commits/master/projects/red_pitaya_0_92

> What would be the way to port the 0.94 ?

The latest versions of the Red Pitaya ecosystem are a little bit chaotic. There are at least two versions of the FPGA configurations: 0.93 and 0.94. Some applications require 0.93:
https://github.com/RedPitaya/RedPitaya/blob/master/apps-free/spectrum/fpga.conf
some other applications require 0.94:
https://github.com/RedPitaya/RedPitaya/blob/v0.94-RC23/Applications/scopegenpro/fpga.conf

What is not clear is which Verilog code is used for which version. We could try to assume that the Verilog code from the latest release candidates corresponds to 0.94.

As a starting point, you can try to take the Verilog code from the latest release candidate:
https://github.com/RedPitaya/RedPitaya/tree/v0.94-RC23/fpga/rtl

Then system_bd.tcl from
https://github.com/RedPitaya/RedPitaya/blob/v0.94-RC23/fpga/ip/system_bd.tcl
should be converted to something like block_design.tcl from
https://github.com/pavel-demin/red-pitaya-notes/blob/master/projects/red_pitaya_0_92/block_design.tcl
Probably, copying the code related to XADC, S_AXI_HP0 and S_AXI_HP1 would be enough.

mhanuel26, 2016-02-04:
Thank you for such detailed response,

I will try to make some progress based on your suggestion,

To conclude, is fair to say that using your version of ecosystem to build up from there might be better than starting from the debian OS of RP developers?
I am more interested in your version since I think it provides more flexibility, just one last question regarding the XADC, for DMA would it need the UIO driver right or is what you would recommend in case DMA is needed? 
I am not sure if ecosystem has DMA but so far I have been able to play around with your code and bring UIO drivers up with interrupts too.

I appreciate your comments,

Best,

pavel-demin, 2016-02-04:
It all depends on what kind of application you want to build and what kind of data you want to copy via DMA.

If you're planning to add new functionality to the original Red Pitaya FPGA configuration, then it's probably better to use the building system from the Red Pitaya developers.

If you're panning to build a new FPGA configuration following the design flow based on the IP cores, then my scripts could be useful.

XADC, DMA and UIO can be used with both systems.

For information. I've once written a long comment about DMA on the Red Pitaya forum:
http://forum.redpitaya.com/viewtopic.php?t=551&p=2740#p2200

Since then, I discovered (but did not test) another driver that could be used with the Xilinx DMA IP cores:
https://github.com/pothosware/pothos-zynq/

closed 2016-02-22 by pavel-demin