pavel-demin/red-pitaya-notes #174: dds_compiler and redpitaya DAC

cedricboudinet, 2016-02-08:
Hello

I am trying to send the output of a dds bloc (generating a 1MHz sine wave) to the DAC (using axis_red_pitaya_dacIP bloc) in Vivado.

I tried a direct connection and I had no output.
Then I tried to insert a width converter and a clock converter between the two blocs, then the dac has an output but it seems erratic (frequency much higher than 1MHz, very noisy, ...).

We set these parameters in the dds compiler :
 - Fixed phase increment
 - 1MHz Output Frequancy (Phase Increment is 0h83126E)

Regards
Cedric

pavel-demin, 2016-02-08:
Hello Cedric,

I have a test project with dds_compiler connected directly to axis_red_pitaya_dac:
https://github.com/pavel-demin/red-pitaya-notes/tree/master/projects/dac_test

Last time when I tested it, it worked quite well.

In this project, both DDS and DAC are clocked from a PLL connected to the ADC clock.

Here is how it looks like with all the unneeded blocks removed:
[image]

Best regards,

Pavel

cedricboudinet, 2016-02-10:
I managed to make it working. Thanks.

closed 2016-02-10 by cedricboudinet