pavel-demin/red-pitaya-notes #181: cic_compiler and dac

cedricboudinet, 2016-03-03:
Dear Pavel

I am trying to send the output of a CIC filter to the DAC you developed for the RedPitaya.
To try this I used a DDS generating a 2MHz sine and a 2 MHz cosine. The output of the DDS is on 32 bits (the cosine on bits 0 to 13, the sinus on bits 16 to 29).
The DDS output is send on an axis_subset_converter in order to keep just the cosine (bits 0 to 15).
Then a CIC compiler. We tuned it with the following parameters:
- decimator, 4 stages, differential delay 1, 1 channel, fixed rate 4
- input sample frequency :  125 MHz
- clock frequency : 125 MHz
- input data width : 14 bits
- quantization : truncation
- output data width : 14 bits
- all other parameters are unchanged

Output of the CIC are send to an axis_clock converter followed by the DAC.
I attached a screenshot of the project.
The trouble is that we always have +1 Volt as an output.
Do you have an idea of where the problem might come from?

Thanks in advance for any help
Cedric
[image]

pavel-demin, 2016-03-03:
Hi Cedric,

I'm not sure if I understand your configuration. What is the purpose of the CIC filter?

Looking at your description, I'd say that CIC filter does not produce enough samples for DAC. With "input sample frequency: 125 MHz" and "decimator, fixed rate 4", the CIC output sample rate is four times lower than the DAC sample rate. The clock converter is empty most of the time. When clock converter is empty, its output TVALID signal is low and my DAC interface puts DAC into the reset state. The code is on line 52 in [cores/axis_red_pitaya_dac_v1_0/axis_red_pitaya_dac.v](https://github.com/pavel-demin/red-pitaya-notes/blob/master/cores/axis_red_pitaya_dac_v1_0/axis_red_pitaya_dac.v#L52).

Looks like this configuration keeps DAC in the reset state, DAC outputs -1 V in this state and it's inverted to +1 V by the output amplifier.

Best regards,

Pavel

cedricboudinet, 2016-03-16:
Hi Pavel
Sorry for the delay.
The purpose of the CIC in this case is to downsample the input signal (generated by the DDS) and send to the DAC.
We also tried to generate a slower clock with a PLL (same frequency that the CIC output) and connect it to the DAC but we have some errors.
Do you have any hints for resolving this problem.
Best regards.
Cedric

pavel-demin, 2016-03-16:
Hi Cedric,

Thanks for the explanation.

I'd try the following:
 - add PLL
 - remove CIC filter
 - set DDS/DDS_CLOCK_RATE to the PLL output frequency
 - connect PLL/clk_out1 to DDS/aclk and DAC/aclk

Best regards,

Pavel

cedricboudinet, 2016-03-21:
Dear Pavel
Thanks for the hint. But in fact this was just a test case for what I would like to do.
We would like to remove a frequency component contained in a signal using a cascade of CIC filter and FIR filter (like the principle of lock in amplifier).

In our configuration:
  - the adc acquires sample data at 125MHz
  - the carrier wave is removed via a CIC filter with a decimation rate of 500. The frequency of the carrier is 2MHz.
  - the CIC output is send to a decimator FIR Filter  (with a decimation rate of 2)
  - finally the FIR output is send to the DAC

Preliminary tests:
First, we inputted to the ADC a signal of 100 Hz amplitude modulated by a carrier of 2 MHz. After sampling, the CIC filter is used to remove the carrier at 2 MHz. The FIR filter is typically  used to remove the ripples created by the CIC filter over the Nysquist frequency.

The FIR filter is sampled at 125 MHz/500 and the DAC at 125MHz/500/2.

After loading the bitstream, the DAC output always remains at 1 Volt.

Best regards and thanks in advance for any help

pavel-demin, 2016-03-21:
In the configuration that you describe, do you have a PLL that outputs 125 and 250 kHz?

closed 2016-03-30 by pavel-demin