pavel-demin/red-pitaya-notes #203: SDR receiver

Traore91, 2016-03-22:
Dear Pavel
    I want to use your source code SDR receiver. I would like to connect after demodulation and filtering the output of the FIR filter to the redpityaya's DAC. The decimation filter reports of CIC and FIR filters are fixed in my configuration and respectively equal to 1000 (for CIC filter) and 2 (for RIF filter).
I would like to use a PLL to sample the DAC at 62.5 kHz. Except that using the IP clocking wizard (on PLL mode), the minimum output frequency is 6 MHz (upper than the desired frequency).

Do you have a solution on this.

Best regards.

pavel-demin, 2016-03-22:
Thanks for information about the minimal PLL frequency. I didn't know about it. There is already a very similar question https://github.com/pavel-demin/red-pitaya-notes/issues/181.

Since I don't know your requirements, could you please explain me why DAC should be clocked at such a low frequency?

I can propose to use one of the SDR transceiver configurations and connect DDC output to DUC input.

Traore91, 2016-03-22:
It's almost the same problem.
By sampling a modulated signal by the ADC 125 MHz for example. After demodulation and filtering , I want to have an output bandwidth of 500 kHz after filtering. After the passage of the CIC decimator filters, sampling frequency is divided by the decimation ratio of 1000. 
Now I would like to transfer the data output of the filters to the DAC.
The sampling frequency of output data of the two filters is normally 125 MHz/1000/2 = 62.5 kHz.
Is - it imperative to have a clock at 62.5 kHz to subsequently view the data output of the DAC?

Best regards.

pavel-demin, 2016-03-22:
> Is - it imperative to have a clock at 62.5 kHz to subsequently view the data output of the DAC?

No, not at all. It's what you initially proposed. I've never tried to run the Red Pitaya DAC at low frequencies.

Please have a look at my SDR transceiver configuration. In these configurations the DAC is clocked at 125 MHz but there is not much problem to output signals at low frequency.

I'd say that with the following project it's possible to do what you describe:
http://pavel-demin.github.io/red-pitaya-notes/sdr-transceiver/

issmaf44, 2016-03-23:
Dear Pavel,

I think it is possible with your configuration because you used a DUC after the DDC.
As in your configuration, is it possible to store the data outputs of the two filters in the RAM of the fpga and write the next data stored in the slow DAC 100ks/S of the redpitaya?

Best regards.

pavel-demin, 2016-03-23:
> As in your configuration, is it possible to store the data outputs of the two filters in the RAM of the fpga and write the next data stored in the slow DAC 100ks/S of the redpitaya?

Writing to RAM should be possible.

As far as I know there is no real slow DAC on the Red Pitaya board. There are just some pins that can be used for PWM. I've never used them and I don't know if it's possible to use them at 100ks/S.

issmaf44, 2016-03-23:
Are you think it's possible to store the ouput data of the filter in the RAM of the FPGA and read the stored data by the DAC sampled at 125 MHz (without using a DUC)?

pavel-demin, 2016-03-23:
I'm still not sure if I understand what you're going to achieve and what are your requirements.

If your project requires to read samples with ADC, filter the samples and write them to DAC, then it can be done without writing anything to RAM. Is writing to RAM required by your project?

If your filters require decimation to lower sample rates, then before writing the samples to DAC they should be somehow interpolated to the DAC sample rate.

pavel-demin, 2016-03-23:
It's just occurred to me that a USB sound card could be used to output samples at low rates. There are sound cards like Creative Sound Blaster X-Fi and Asus Xonar U5 that work at 96 and 192 kHz.

issmaf44, 2016-03-25:
Dear Pavel,
    I am trying to test your proposal.

[image]

The dac is attacked by an amplitude modulated signal of 30 Hz. The frequency is  2MHz. I want to use a CIC decimator filter followed by a FIR decimator filter to remove the component at 2 MHz.
Reports decimation filters are: Filter CIC (1000) and FIR filter (2).

the sampling frequency of The output of these filters is 125MHz / 1000/2 = 62.5 kHz. At the exit of the two filters (CIC and FIR) another CIC interpolation filter is used to increase the sampling frequency up to 125 MHz. The data output of the interpolation filter are sent to the DAC sampled at 125 MHz .

By visualizing, the DAC output is like it was saturated.

Do you have a solution?

pavel-demin, 2016-03-28:
> By visualizing, the DAC output is like it was saturated.

Do you have an oscillogram to illustrate the saturated DAC output?

> Do you have a solution?

I've never tried anything similar to your configuration.

closed 2016-03-30 by pavel-demin

issmaf44, 2016-04-01:
Hello,
I am trying another configuration using single rate fir compiler in order to avoid down sampling and sampled the dac at 125 MHz.
I want to realize a FIR filter using the ip Vivado available. I want to implement a single rate filter sampled at 125 MHz and whose input data sampled is to 125 MHz. The band pass of the filter is 100 kHz. For this, I calculated 500 coefficients allowing me to have this frequency response.

Except it shows me the following error:
Validation failed on parameter 'Column Configuration(ColumnConfig)' for Column 0 length (501) is greater than the corresponding available column (39)
. BD Cell '/fir_compiler_0'

You can see the attached picture.
[image]

Do you know how to avoid this problem?
Thanks for your help.
Best regards.

pavel-demin, 2016-04-01:
I'd say that the error message means that there is not enough FPGA resources to accommodate your filter design requirements.

> Do you know how to avoid this problem?

Design filters with smaller number of coefficients and/or running at a lower sample rate.

issmaf44, 2016-04-01:
Thank you Pavel,
In order to running at a lower sample rate, I try to use a CIC compiler with a decimation rate of 5. The sampling frequency at the output of the CIC filter is 125MHz / 5=50 MHz.
Given that I'm limited by the PLL , I would like to sample the DAC at 50 MHz using a PLL. Is it necessary to put an axis clock converter between the CIC compiler and the DAC.
In this case,  the s-axis clock of the CIC compiler is normaly 50 MHz (corresponding to the output data sampling of the CIC).

But it displays this error
 [BD 41-237] Bus Interface property FREQ_HZ does not match between /axis_clock_converter_0/s_axis_aclk(50000000) and /cic_compiler_0/M_AXIS_DATA(142857132)

pavel-demin, 2016-04-01:
> Is it necessary to put an axis clock converter between the CIC compiler and the DAC.

If CIC compiler and DAC interface are in different clock domains, then it's necessary.

The error message indicates that there is a problem with the clock signals connected to the  clock converter.

pavel-demin, 2016-04-01:
I'm locking this issue, since the last few questions are not related to my projects or to any problems in my code.