pavel-demin/red-pitaya-notes #204: how to run wi-fi usb dongle as wlan0.client

dragniko, 2016-03-23:
Ethernet works fine, but I have hard time understanding steps like where to put wpa_supplicant.conf file and what needs to be in it such that wi-fi behave like client not the server. Right now I have in /boot/wpa_supplicant.conf the following

# Red Pitaya WiFi configuration file
ctrl_interface=/var/run/wpa_supplicant
network={
		ssid="myssid"
#		proto=RSN
		key_mgmt=WPA-PSK
#		pairwise=CCMP TKIP
#		group=CCMP TKIP
		psk="myssidpassword"
}

This setting works fine on the latest RP OS image found at http://redpitaya.com/quick-start/
but there /etc/network/interfaces file looks like

# interfaces(5) file used by ifup(8) and ifdown(8)
# Include files from /etc/network/interfaces.d:

# source-directory /etc/network/interfaces.d
source interfaces.d/eth0
source interfaces.d/wlan0

and interfaces.d/wlan0 is a link to wlan0.ap, which looks like

##############################################
# wlan0: Wireless USB adapter AP config      #
##############################################

allow-hotplug wlan0
iface wlan0 inet static
  address 192.168.128.1
  netmask 255.255.255.0
  post-up service hostapd restart
  post-up service isc-dhcp-server restart
  post-up iptables-restore < /etc/iptables.ipv4.nat
  pre-down iptables-restore < /etc/iptables.ipv4.nonat
  pre-down service isc-dhcp-server stop
  pre-down service hostapd stop

However, there is also a wlan0.client file that looks like
##############################################
# wlan0: Wireless USB adapter client config  #
##############################################

allow-hotplug wlan0
iface wlan0 inet dhcp
    pre-up service hostapd stop
    pre-up service isc-dhcp-server stop
    pre-up wpa_supplicant -B -D wext -i wlan0 -c /boot/wpa_supplicant.conf
    post-down killall -q wpa_supplicant
    udhcpc_opts -t7 -T3

which looks for user's local configuration in /boot/wpa_supplicant.conf.
Which scripts are used to create wlan0.client and to read it? Where do I tell that wlan0 is in client not the server mode?
Thanks for your time and effort
Dragan

pavel-demin, 2016-03-23:
Hi Dragan,

I'm afraid I don't know how to answer your questions. I'm not the author of the client Wi-Fi configuration for Red Pitaya.

I'd suggest to ask this question the Red Pitaya support or to copy this ticket to the Red Pitaya repository at https://github.com/RedPitaya/RedPitaya.

There are also some instructions at https://github.com/RedPitaya/RedPitaya/blob/master/NETWORKING.md but I don't know if they are up-to-date.

Best regards,

Pavel

dragniko, 2016-03-27:
you were correct, in the latest RP image wlan0 is the soft link to either wlan0.ap (access point mode) or wlan0.client (client mode). Personal preferences are read from /boot/wpa_supplicant.conf, so all works fine now.

closed 2016-03-27 by dragniko