pavel-demin/red-pitaya-notes #209: More in depth explaination of the SDR design decisions.

G4GUO, 2016-04-06:
Hello Pavel,

I am trying to understand your FPGA block diagram and I have some questions.
1) What does the LFSR that drives the complex mixer control channel do?
2) Why do you have 3 FIR stages inside your RX modules
3) Why do you split the the signal into 2 CIC blocks rather than use a single dual channel CIC block?

The other questions I have are Vivado related and I can probably Google them.

I am trying to understand your design decisions because I hope to do something similar with my Zedboard  and AD9253 EVM (mainly as a learning exercise)

I know these are not really issues but I can't find any other suitable way of asking you.

- Charles

pavel-demin, 2016-04-06:
Hello Charles,

Thank you for interest in my SDR projects and for trying to understand how they work.

I'll try to briefly answer your questions and give some links for more information.

> 1) What does the LFSR that drives the complex mixer control channel do?

LFSR generates a pseudo-random sequence used by the complex multiplier for random rounding.

http://www.xilinx.com/support/documentation/ip_documentation/cmpy/v6_0/pg104-cmpy.pdf
http://www.acasper.org/2011/10/02/my-sdr/

> 2) Why do you have 3 FIR stages inside your RX modules

The additional FIR filters are used for the fractional sample rate conversion.

http://www.xilinx.com/support/documentation/ip_documentation/fir_compiler/v7_2/pg149-fir-compiler.pdf

> 3) Why do you split the the signal into 2 CIC blocks rather than use a single dual channel CIC block?

The ADC sample rate and the FPGA frequency are both about 125M. For the CIC filter to process two channels of the ADC samples it should run at twice higher frequency w.r.t. the ADC sample rate.

http://www.xilinx.com/support/documentation/ip_documentation/cic_compiler/v4_0/pg140-cic-compiler.pdf

I've recently tried to clock the data processing chain with a 250 MHz clock and got some timing problems in the CIC filters. Looks like CIC filters have to use DSP blocks when running at higher frequencies and I need the DSP blocks for the DDS and FIR modules.

Best regards,

Pavel

G4GUO, 2016-04-06:
Thanks Pavel, that makes some sense I am using a Zedboard which has a 7020 so I have a bit more PL fabric to play with. 
All I have managed to do so far is to extract the frame clock from the AD9253 and use it to blink a led.
At least that shows the hardware mod I had to make to the AD interposer board works.

After wasting a lot of time creating a separate 125 MHz clock domain using the clock wizard I am now 
at the stage where I am starting to do the receiver modules so I have been looking at your design 
for inspiration.

Thanks again

- Charles

closed 2016-04-06 by pavel-demin