pavel-demin/red-pitaya-notes #212: AXI Stream to Memory mapped

jeanminet, 2016-04-13:
Hello Pavel,

Thanks a lot for your very helpful work !
In your `sdr_transceiver` project, you use a Xilinx `FIFO generator` core in combination with two custom cores (`axi_axis_reader` and `axi_axis_fifo`) in order to have a memory mapped access to the data stream: 

[image]

It looks like this functionality could be implemented with the Xilinx [`AXI-Stream FIFO`](http://www.xilinx.com/support/documentation/ip_documentation/axi_fifo_mm_s/v4_1/pg080-axi-fifo-mm-s.pdf) core:

[image]

I'm wondering if I missed something. Does your combination provide better throughput ?

Best,
Jean

pavel-demin, 2016-04-13:
Hello Jean,

> Does your combination provide better throughput ?

Throughput - I don't know, maybe. Features - yes.

I've once simplified the AXI4-Lite interface in all my AXI4-Lite cores with the following commit: https://github.com/pavel-demin/red-pitaya-notes/commit/663e313b6ccf3cb0a866e85cce2f7de34704d633.

I've tried several AXI and AXIS FIFO cores provided by Xilinx but without much success. They behaved strangely, had some features that I didn't need, missed some features that I needed.

My projects often require the following features:
 - mixed ports widths,
 - s_axis_tready and m_axis_tvalid always high,
 - m_axis_tdata low if FIFO is empty.

Both [axi_axis_reader](https://github.com/pavel-demin/red-pitaya-notes/tree/master/cores/axi_axis_reader_v1_0) and [axis_fifo](https://github.com/pavel-demin/red-pitaya-notes/tree/master/cores/axis_fifo_v1_0) are very simple and do exactly that I need.

Best regards,

Pavel

jeanminet, 2016-04-13:
That makes sense.
I didn't realize the `axis_fifo` core was so simple !

closed 2016-04-13 by pavel-demin