pavel-demin/red-pitaya-notes #213: Adding WSPR transceiver to sd card with other apps

df8oe, 2016-04-15:
Hello Pavel,

is there a howoto where I can find how to add this *very nice* project to my existing SD-card? I think this should be possible... I just have received my red pitaya yesterday and took a look at the card which actually contains the two default applications (and now additional the two sdr transceiver projects brought by you installed yesterday via bazaar). I am working with Linux and I am familar with doing such things manually :) Actually very interesting feature would be to upload prebuild packages via web interface so that you can add apps which are not available via bazaar...

vy73 de
Andreas, DF8OE

pavel-demin, 2016-04-15:
Hello Andreas,

There is now such a howto. At the moment, it's easier for me to provide a ready to use SD card image than writing such a howto.

You can have a look at the script that I'm using to build the SD card image:
https://github.com/pavel-demin/red-pitaya-notes/blob/master/scripts/debian-wspr.sh

First, it copies the following files to the SD card image:
```
projects/sdr_transceiver_wspr/transmit-wspr-message.c
projects/sdr_transceiver_wspr/transmit-wspr-message.cfg
projects/sdr_transceiver_wspr/write-c2-files.c
projects/sdr_transceiver_wspr/write-c2-files.cfg
projects/sdr_transceiver_wspr/decode-wspr.sh
projects/sdr_transceiver_wspr/README
projects/sdr_transceiver_wspr/Makefile
```
Next, it installs packages and builds programs with the following commands
```
apt-get -y install openssh-server ca-certificates ntp ntpdate fake-hwclock \
  usbutils psmisc lsof parted curl vim wpasupplicant hostapd isc-dhcp-server \
  iw firmware-realtek firmware-ralink ifplugd ntfs-3g \
  build-essential subversion libfftw3-dev libconfig-dev parallel

svn co svn://svn.code.sf.net/p/wsjt/wsjt/branches/wsjtx/lib/wsprd
make -C wsprd CFLAGS='-O3 -march=armv7-a -mcpu=cortex-a9 -mtune=cortex-a9 -mfpu=neon -mfloat-abi=hard -ffast-math -fsingle-precision-constant -mvectorize-with-neon-quad' wsprd
make
```

Then, you'll need a couple of scripts to start and to stop the WSPR transceiver application. Something like the [start.sh](https://github.com/pavel-demin/red-pitaya-notes/blob/master/projects/sdr_transceiver/bazaar/start.sh) and [stop.sh](https://github.com/pavel-demin/red-pitaya-notes/blob/master/projects/sdr_transceiver/bazaar/stop.sh) from the SDR transceiver application. 

The start script should look like the following:
```
devcfg=/sys/devices/soc0/amba/f8007000.devcfg
test -d $devcfg/fclk/fclk0 || echo fclk0 > $devcfg/fclk_export
echo 1 > $devcfg/fclk/fclk0/enable
echo 143000000 > $devcfg/fclk/fclk0/set_rate

cat sdr_transceiver_wspr.bit > /dev/xdevcfg

(crontab -l ; echo "1-59/2 * * * * cd /dev/shm && /root/decode-wspr.sh >> decode-wspr.log 2>&1") | crontab -
```

The stop script should kill all the WSPR related processes (`killall decode-wspr.sh write-c2-files wsprd`) and somehow disable the crontab job.

Making the WSPR transceiver application compatible with the default Red Pitaya SD card would seem to be trivial to implement but unfortunately it's not so. There are some small but important details that don't fit well together. The problems that I see are the following:
- Some pacjages are missing on the default  Red Pitaya SD card.
- Programs and FPGA configurations should be stopped and started in the right order. For example, stopping WSPR scripts and programs is already not very trivial and involves some careful crontab editing.
- To solve the previous problem, the WSPR scripts and programs should probably be converted into a single monolithic program that would be easy to start and stop.

Best regards,

Pavel

df8oe, 2016-04-15:
Hello Pavel,

many thanks for your support. That seem to be "solvable" for me :) I will give it a try at the weekend and report what I get for results. I am in contact with Michael, DK5HH and we will discuss results, too. The application seems to me *very* interesting and I want to get it running as fast as possible :)

If I have done anything (scripting e.g.) which can help others I will send you a pull request.

vy 73
Andreas

df8oe, 2016-04-16:
Hello Pavel,

I have investigated that all packages which are neccessary for run-time are already installed onactual rp-sd. Packages that are missing are: subversion, ntfs-3g, both -dev - packages and parallel...

I have mounted your wspr-image as loop and copied all (??) necessary binaries 8I found in /root) to actual sd-card. Now I must take a look to startup- und stop- scripts  - I'll report when finished :)

vy73
Andreas

pavel-demin, 2016-04-16:
For information. I've just updated the SD card image for the WSPR transceiver project and uploaded [sdr_transceiver_wspr.bit](https://googledrive.com/host/0B-t5klOOymMNfmJ0bFQzTVNXQ3RtWm5SQ2NGTE1hRUlTd3V2emdSNzN6d0pYamNILW83Wmc/SDR/sdr_transceiver_wspr.bit) that could be useful for your development.

df8oe, 2016-04-17:
Yes - of course it is :) Many thanks.

My idea is to create a tarball where all binaries and scripts are included. So users only need to login via ssh, fetch tarball via scp and start "configuration script" what will be bash script using gtk. So you can enter callsign, locator, tx power and all other necessary informations. Script installs additional packages thatare needed and does changes to cron file. When started next time it allows modify settings. Script will start second time running on "screen" so that you can leave ssh connection and fetch back running script again after ssh login.

Because of I am heavy investigated in firmware grogramming for SDR standalone transceiver mcHF that will take some time but I definitely will do that.

Great work, Pavel! It works like a charm...

vy 73
Andreas

pavel-demin, 2016-05-23:
There is now a new version of my SD card image based on Debian 8.4 and Red Pitaya ecosystem 0.95:
http://pavel-demin.github.io/red-pitaya-notes/debian-ecosystem/

It could be used as a starting point for a SD card including WSPR and other SDR applications.

I've just realized that replacing crontab with /etc/cron.d could simplify the start and stop scripts:

start.sh:
```
devcfg=/sys/devices/soc0/amba/f8007000.devcfg
test -d $devcfg/fclk/fclk0 || echo fclk0 > $devcfg/fclk_export
echo 1 > $devcfg/fclk/fclk0/enable
echo 143000000 > $devcfg/fclk/fclk0/set_rate

cat sdr_transceiver_wspr.bit > /dev/xdevcfg

ln -sf /root/wspr.cron /etc/cron.d/wspr.cron
```

stop.sh
```
devcfg=/sys/devices/soc0/amba/f8007000.devcfg
test -d $devcfg/fclk/fclk0 || echo fclk0 > $devcfg/fclk_export
echo 1 > $devcfg/fclk/fclk0/enable
echo 125000000 > $devcfg/fclk/fclk0/set_rate

rm -f /etc/cron.d/wspr.cron

killall decode-wspr.sh write-c2-files wsprd
```

df8oe, 2016-05-24:
Hello Pavel,

nice new Debian image for RP - I have installed it and enjoy it :) Many thanks for your work!

I have already built wsprd on RP - works out-of-the-box. Building transmit-wspr-message throws a wanrning about argument 3 of config_lookup_string ("incompatible pointer string") - is it ok indeed?

Next I will create startup scripts and then test it :)

vy 73
Andreas

pavel-demin, 2016-05-24:
> Where can I find wspr.cron (needed in start.sh)?

It should contain a line very similar to the crontab line:
```
1-59/2 * * * * root cd /dev/shm && /root/decode-wspr.sh >> decode-wspr.log 2>&1
```

df8oe, 2016-05-24:
I tried that already - does not work...

It seems that file is *not allowed* to have extention ".cron".Tried now with symlink name wspr instead of wspr.cron and it works...

vy 73
Andreas

pavel-demin, 2016-05-25:
Thanks for the test. I've fixed the cron.d file names in my previous comment.

df8oe, 2016-05-25:
Many thanks. I will start "whispering" next week and contact Michael DK5HH with informations about thisnew and very good operating system image for RP.

vy 73
Andreas

pavel-demin, 2016-05-25:
For information. Just found the following in `man cron`:

> Files  in  this  directory (/etc/cron.d) must consist solely of  upper-  and  lower-case  letters,  digits, underscores, and hyphens. This means that they cannot contain any dots.

KI7MT, 2016-08-23:
Hi Pavel,

I'm very interested in this thread, in particular, creating formal Debian (.deb ) packages for the various applications you've written up / designed ( unless somebody is already doing this for you of course ).

I've read through a fare bit of the ecosystem framework, but I am not clear on what is modified at the system level for each of your pitaya-note projects or if they all work together somehow.

For example, if all I wanted to run was a minimal system + Multi-Band WSPR, what steps would need to be taken from a package perspective to accomplish that. I can do the Debian packaging work, I just need to better understand the dependencies based on a default image ( e.g. a Debian / Ubuntu Bootstrap ). Once that is understood, the packages could be put together ( I think ) so an end user can simply install a Meta Package in much the same way Debian / Ubuntu desktop environments are installed using apt-get, e.g.

sudo apt-get install ubuntu-desktop - The Ubuntu desktop system
sudo apt-get install edubuntu-desktop - educational desktop for Ubuntu
sudo apt-get install  kubuntu-desktop - Kubuntu Plasma Desktop/Netbook system
sudo apt-get install  lubuntu-desktop - Lubuntu Desktop environment
sudo apt-get install build-essential

and so on ..

All of those are Meta packages, which install all the packages required for a given environment / function, but the end user simply types: < sudo apt-get install ... >. There are hundreds of meta packages for development, non-Foss applications, restricted extra codecs, the lists goes on forever.

If possible, can you email me off-line to discuss this, when your time allows of course: 

ki7mt@yahoo.com

Thanks

73's
Greg, KI7MT

pavel-demin, 2016-08-23:
Hi Greg,

Thanks for your interest in this development. My e-mail address can be found in my [profile](https://github.com/pavel-demin).

> I'm very interested in this thread, in particular, creating formal Debian (.deb ) packages for the various applications you've written up / designed ( unless somebody is already doing this for you of course ).

As far as I know, nobody's doing the Debian packages for my Red Pitaya projects. Red Pitaya has its kind of package system with all the packages available from http://archives.redpitaya.com. They can be installed via the application marketplace web interface.

> For example, if all I wanted to run was a minimal system + Multi-Band WSPR, what steps would need to be taken from a package perspective to accomplish that. I can do the Debian packaging work, I [...]

Here are the dependencies for the WSPR project:
 - libc6
 - libconfig
 - libfftw3 (for the WSPR decoder)
 - curl
 - parallel
 - ntp (recommended)
 - cron (recommended)

As I mentioned in one of my previous comments, I still think that before packaging this application, all its scripts and programs should be merged into a single monolithic program that would be easy to start and stop. In its current form, this application won't coexist very well with other applications.

Best regards,

Pavel

KI7MT, 2016-08-23:
Hi Pavel,

Thanks for the quick response. I need to study things a good bit more, but, I did notice in the Red Pitaya Changelog ( https://github.com/RedPitaya/RedPitaya/blob/master/CHANGELOG.md ) , they have migrated to Ubuntu 16.04 which allows direct access / use of Launchpad PPA and and they've switched to Systemd for various tasks.

Using Systemd may be an alternative way to control the Multi-WSPR application as well as move the uploading functionality to a service daemon, but as said above, I need to study this a good bit more.

As you said, making multi-wspr a monolithic app would certainly be beneficial. When I get things up and running on my end and do a bit of poking around, I'll email you off-line to discuss things in more detail.

73's
Greg.

SpudGunMan, 2017-01-03:
downloaded the image but get this error on a new install/run
`Recording ...
Killed write-c2-files(1875) with signal 15
Terminated
Decoding ...
Cannot open data file 'wspr_*_170103_2045.c2'`

SpudGunMan, 2017-01-04:
I see the dependencies listed - I didnt see any dependencies listed for wsprd or the huge 20gb SDK package.. wsprd is easy.. what is the components used from the xlinux SDK?

I would also like to get this code working on another (the main SD card) for RP

pavel-demin, 2017-01-04:
> I didnt see any dependencies listed for wsprd or the huge 20gb SDK package..
> what is the components used from the xlinux SDK?

Please see my previous comment for all the dependencies. Here is a line related to `wsprd`:
 - libfftw3 (for the WSPR decoder)

I build `wsprd` on Red Pitaya. The Vivado SDK package isn't required for this.

> I would also like to get this code working on another (the main SD card) for RP

Before running `write-c2-files` the FPGA should be reprogrammed by loading the [sdr_transceiver_wspr.bit](https://www.dropbox.com/sh/5fy49wae6xwxa8a/AACSA6RMQm4OxU5yqfDz8-rHa/sdr/sdr_transceiver_wspr.bit?dl=1) bitstream file into `/dev/xdevcfg`:
```
cat sdr_transceiver_wspr.bit > /dev/xdevcfg
```

SpudGunMan, 2017-01-04:
Thanks for the bit file .. the old link was bad and this corrects the "bus error" that I would get otherwise.

to build the project it requires the WSPRD code files as per build errors 
`wsprsim_utils.o', needed by 'transmit-wspr-message'.`
a source for the files is..
https://github.com/pavel-demin/wsprd 

here is a quick copy paste of commands to make this work on the official RP card.. I am looking at other ways to make this more user friendly like others.

```
apt-get install libc6 curl parallel ntp cron libfftw3-bin libfftw3-dev libconfig-dev

git clone https://github.com/pavel-demin/red-pitaya-notes
git clone https://github.com/pavel-demin/wsprd

wget https://www.dropbox.com/sh/5fy49wae6xwxa8a/AACSA6RMQm4OxU5yqfDz8-rHa/sdr/sdr_transceiver_wspr.bit
cat sdr_transceiver_wspr.bit > /dev/xdevcfg
```

pavel-demin, 2017-01-04:
As it was mentioned in my first comment, all the installation commands can be found in [scripts/debian-wspr.sh](https://github.com/pavel-demin/red-pitaya-notes/blob/master/scripts/debian-wspr.sh). I don't think that they should be duplicated comments of this issue.

> a source for the files is..
> https://github.com/pavel-demin/wsprd

This is wrong. Please don't use this repository. I'm removing my test wsprd repository and the related lines from your comment to prevent confusion.

The correct commands are in [scripts/debian-wspr.sh](https://github.com/pavel-demin/red-pitaya-notes/blob/master/scripts/debian-wspr.sh#L133).

pavel-demin, 2017-09-10:
There is now a new SD card image with all the pre-built applications including the WSPR transceiver. It's based on the lightweight [Alpine Linux](https://alpinelinux.org) distribution.

For me, this SD card image has the following advantages:
 - fast to build (much faster than the SD card images based on Debian and Ubuntu),
 - fast to download (it's much smaller than the SD card images based on Debian and Ubuntu),
 - easy to install (just unpack the .zip file to a FAT formatted micro SD card),
 - easy to maintain (since it contains all the pre-built applications, I'll have only one SD card image to maintain).

The SD card image and some instructions can be found at the following link:
http://pavel-demin.github.io/red-pitaya-notes/alpine

closed 2017-09-23 by pavel-demin