pavel-demin/red-pitaya-notes #234: invalid AXI memory map range values

AnttiLukats, 2016-05-15:
during the process of converting the IP cores to full Vivado IP Catalog cores I found that at least some AXI IP cores have invalid-meaningless memory range settings the way they are created by the tcl scripts. This does make the "auto assign" address in Vivado IPI to fail. This is minor issue of course, and not relevant for the non-gui build process.

pavel-demin, 2016-05-15:
Could you please provide more details? What cores? How to reproduce the problem?

pavel-demin, 2016-05-15:
I've just tried to add two AXI cores: AXI-Stream FIFO by Xilinx and AXI Configuration Register from my cores repository. Here is what I obtained:
[image]

AXI Configuration Register gets a 512 MB address range. It works in the same way in my Tcl scripts. I use the following commands to assign a required address range:
```
set_property RANGE 4K [get_bd_addr_segs ps_0/Data/SEG_cfg_0_reg0]
set_property OFFSET 0x40001000 [get_bd_addr_segs ps_0/Data/SEG_cfg_0_reg0]
```

Any idea how it could be improved?

AnttiLukats, 2016-05-15:
Hi,

sorry I was not precise enough: I used the sdr-receiver that has 3 IP cores, if they are de-assigned and then you try auto-assingn then it fails badly..

BEST solution: you specify the range, but I am not sure how to-do it from the "IP create" script.

I would suggest setting 64K for axi_cfg and axi_sts, but your examples already use fixed assignments in 4K blocks, so I am setting the Address range to 4K (not 64K) in my commits (in the fork..)

AnttiLukats, 2016-05-16:
closed as the issue does not exist when not using the GUI.

closed 2016-05-16 by AnttiLukats

pavel-demin, 2016-11-24:
Just discovered a simple solution for this problem.

The range for axi_cfg and axi_sts is automatically calculated as  `pow(2,(AXI_ADDR_WIDTH - 1) + 1)`.

So, to set the default range to 64K, it's enough to set the "AXI ADDR WIDTH" parameter to 16. It can be done via the IP core GUI.

I'll set the default AXI_ADDR_WIDTH value to 16 for all my IP cores with the AXI4-Lite slave interface.