pavel-demin/red-pitaya-notes #235: lfsr AXI width should be not editable

AnttiLukats, 2016-05-15:
very very minor, TDATA width is user editable, but the core only supports 64 bits, so it should not be user editable in the GUI.

very minor thing

pavel-demin, 2016-05-15:
Solved with [this commit](https://github.com/pavel-demin/red-pitaya-notes/commit/d9d9d7841c8041b36e0281a44592fcf9cffe70d0).

closed 2016-05-15 by pavel-demin

AnttiLukats, 2016-05-15:
ups, this may have broken the IP now. Sorry for the trouble, the "editable" in the gui is not that big of an issue, now the problem seems to be worse. Well initially there was no bi problem, now there is.

pavel-demin, 2016-05-15:
Strange. I can build the new version of the core and a project using this core without any problem.

What exactly is broken in the new version?

AnttiLukats, 2016-05-15:
it MAY build if all started from non-gui scripting. but if you open the IP in IP packager:

ERROR>>>
[IP_Flow 19-627] Port 'm_axis_tdata': XPath expression failed: Undefined parameter "AXIS_TDATA_WIDTH" used in XPATH expression "(AXIS_TDATA_WIDTH - 1)".

WARNING>>>
[IP_Flow 19-3834] Port 'm_axis_tdata': Port mapped to the logical port 'TDATA' of 'xilinx.com:interface:axis:1.0' interface must have a width of multiples of 8 bits.  The current port width is '1'.

pavel-demin, 2016-05-15:
I've removed the last commit.

Since all my cores work quite well with my projects, I'm not much interested in solving all these minor issues.

I'd prefer if you create a pull request once you find solutions to all these minor issues or create a **single** issue listing all these minor problems.

AnttiLukats, 2016-05-16:
understood your point.