pavel-demin/red-pitaya-notes #237: axi_axis_reader, ARADDR port map missing

AnttiLukats, 2016-05-15:
opening this in IP Packager yields in RED Warning:

[IP_Flow 19-3256] Bus Interface 'S_AXI': The slave AXIMM interface used as a read channel requires port maps to interface signals ARADDR, ARVALID, ARREADY, RDATA, RVALID and RREADY.  Found missing port map(s) for: ARADDR.

ARADDR signal is missing, even if not used it should be present as the interface has marked it as "required" signal

pavel-demin, 2016-05-15:
It's just a warning. I'd prefer to not add ports that are not used in the module.

Does  IP Packager continue to work after this warning?

AnttiLukats, 2016-05-15:
I am sorry, if the IP Interface IP-XACT defintions declares some ports as MANDATORY, they must be present, no matter if used or not.

I think it does still package, but you have to manually CONFIRM that you want to package an IP that includes an ERROR (not warning!) by the IP-XACT DRC checks.

pavel-demin, 2016-05-15:
I'd say that it's not a bad thing that the axi_axis_reader and axi_axis_writer cores have problems with the GUI tools. I use them only with axis_fifo and fifo_generator. I would not include them in a general purpose library.

AnttiLukats, 2016-05-16:
I see, this was not so clear.

closed 2016-05-16 by AnttiLukats