pavel-demin/red-pitaya-notes #238: Red Pitaya File Saving and suggestions

ptamike, 2016-05-17:
First of all thank you for the excellent support you're providing for the Red Pitaya board. It is very much appreciated.

I've been experimenting with your VNA application and have spotted a couple of minor bugs/suggestions in the Windows control program.

1 - When you attempt to save the configuration or s1p data, the Save As dialog shows the expected file suffix (*.ini or *.s1p) but this isn't automatically added to the file-name when you save it. You have to manually add the suffix. It's not a major issue but it confused me for a while so may confuse others.

2 - If you change the Start/Stop/Points settings then attempt a DUT measurement without re-calibrating, the program hangs. It might be better if it gave a message alerting the user to re-calibrate.

3 - Is it possible to save the Red Pitaya's IP address when saving a configuration?

Once again, thanks for some great software.

Mike - G4WNC

pavel-demin, 2016-05-17:
Thank you for your tests and suggestions!

The points 1 and 3 are addressed in [this commit](https://github.com/pavel-demin/red-pitaya-notes/commit/ee9f2545825f063f606ee693d9f1a4cb12d83e26).

If you want to give this new version a try, the you can download [vna.py](https://raw.githubusercontent.com/pavel-demin/red-pitaya-notes/ee9f2545825f063f606ee693d9f1a4cb12d83e26/projects/vna/client/vna.py) and copy it to the directory where the vna client is installed.

ptamike, 2016-05-17:
Excellent - points 1 and 3 resolved. Thank you.

One other refinement to consider. When running a detailed plot with several thousand points, it would be helpful to have a progress indicator of some sort.

pavel-demin, 2016-05-19:
There is now a new version (0.94-1117) of the VNA application. It has a progress indicator and attempts to address the point 2. At least, it doesn't hang when changing Start/Stop/Points settings without recalibration.

To use new version, both the Red Pitaya application and the client program should be updated.

ptamike, 2016-05-19:
Hi Pavel,

Thanks for the update. Just tried this and I'm unable to connect. I have VNA version 0.94-1117 installed via the Bazaar and have downloaded and installed the control program: vna-win32-20160519 using the link that appears when the Bazaar app runs.
When I enter the RP's IP address I get a Connection refused error.
I tried reinstalling the app and client with the same result.
Any suggestions?

Thanks,

Mike

ptamike, 2016-05-19:
Just built the control program on a Ubuntu PC and I get the same connection refused error.

pavel-demin, 2016-05-19:
Indeed, the bazaar application doesn't work with the default Linux kernel coming with the Red Pitaya SD card. I've removed the VNA application from bazaar and trying to fix the problem. The source of the problem is the Vivado 2016.1 tool chain.

dl1sdz, 2016-05-19:
Hello Pavel,

I always had trouble with the original Red Pitaya Linux Kernel. On the
others side your Kernel versions worked without problems.

Couldn't you provide your Linux with your Apps which could be started with
a shell script?

73 de Hajo

ptamike, 2016-05-19:
I can confirm that the new control program is working well here with your SD card image on github.

Thanks,

Mike

pavel-demin, 2016-05-19:
Hi Mike,

Thanks for testing the SD card image!

I think that my new compilation chain works better now. I've updated the bazaar application. The version number is the same. To update the application, it should be uninstalled and then installed again.

Best regards,

Pavel

ptamike, 2016-05-20:
Hi Pavel,

I see you've just posted a new application (0.94-1117). I've just tested it here and all seems to be fine so far. I'll do some more testing over the weekend.

Regards,

Mike - G4WNC

closed 2016-05-24 by ptamike