pavel-demin/red-pitaya-notes #248: Starting hpsdr via web fails

G4DCP, 2016-05-22:
If you build all and then build the image, followed by installing HPSDR version of transeiver then try to start the image / server end by using the web interface, then the hermes code is not started:-

root@red-pitaya:~# ps -ef|grep sdr
root      1639  1625  0 17:28 pts/0    00:00:00 grep sdr

if you go to the app directory and start it by running the start script it all works:-

root@red-pitaya:~# cd /opt/redpitaya/www/apps/
root@red-pitaya:/opt/redpitaya/www/apps# ls
assets                    error_bazaar_install.html  idfile.id  index.html  scope      sdr_transceiver_hpsdr  style.css
error_bazaar_access.html  error_bazaar_proto.html    images     info        scope+gen  spectrum               ui
root@red-pitaya:/opt/redpitaya/www/apps# cd sdr_transceiver_hpsdr/
root@red-pitaya:/opt/redpitaya/www/apps/sdr_transceiver_hpsdr# ls
controllerhf.so  fpga.conf  index.html  js                     sdr_transceiver_hpsdr.bit  start.sh
css              img        info        sdr-transceiver-hpsdr  src                        stop.sh
root@red-pitaya:/opt/redpitaya/www/apps/sdr_transceiver_hpsdr# ps -ef|grep sdr
root      1763  1625  0 17:32 pts/0    00:00:00 grep sdr
root@red-pitaya:/opt/redpitaya/www/apps/sdr_transceiver_hpsdr# ./start.sh
-bash: ./start.sh: Permission denied
root@red-pitaya:/opt/redpitaya/www/apps/sdr_transceiver_hpsdr# . ./start.sh
root@red-pitaya:/opt/redpitaya/www/apps/sdr_transceiver_hpsdr# ls -al
total 1892
drwxr-xr-x  7 root root    4096 Apr 27 12:18 .
drwxr-xr-x 10 root root    4096 May 22 17:27 ..
-rwxr-xr-x  1 root root    3616 Apr 27 12:18 controllerhf.so
drwxr-xr-x  2 root root    4096 Jan  1 00:22 css
-rw-r--r--  1 root root      72 Jan  1 00:22 fpga.conf
drwxr-xr-x  2 root root    4096 Jan  1 00:22 img
-rw-r--r--  1 root root    3051 Jan  1 00:22 index.html
drwxr-xr-x  2 root root    4096 Apr 27 12:18 info
drwxr-xr-x  2 root root    4096 Jan  1 00:22 js
-rwxr-xr-x  1 root root  698040 Apr 27 12:18 sdr-transceiver-hpsdr
-rw-r--r--  1 root root 1186312 Apr 27 12:18 sdr_transceiver_hpsdr.bit
drwxr-xr-x  2 root root    4096 Jan  1 00:22 src
-rw-r--r--  1 root root     260 Jan  1 00:22 start.sh
-rw-r--r--  1 root root     220 Jan  1 00:22 stop.sh
root@red-pitaya:/opt/redpitaya/www/apps/sdr_transceiver_hpsdr#

is this due to a missing execute permission on the two scripts?

pavel-demin, 2016-05-22:
Hello Peter,

What image are you using? Normally, with the SD card images provided by Red Pitaya the SDR applications start without problem.

Anyway, I've also noticed this problem yesterday when testing the VNA application with my new Debian based image and fixed the permissions in [this commit](https://github.com/pavel-demin/red-pitaya-notes/commit/ae960cf2462f5b98788dba49fea58bddad0cfd0c).

I'm currently rebuilding all the images and applications. When they are ready, I'll upload the applications to the application market place and I'll also update all my images.

Best regards,

Pavel

G4DCP, 2016-05-22:
I was using an image built from your master git repository. I have built a vmware build environment using debian and have been testing some of your recent commits to master. So the answer is basically your repository but built from source.

pavel-demin, 2016-05-22:
Thanks for the clarification. So, you observe exactly the same problem that I've found and fixed yesterday.

A temporary workaround would be to manually fix the permissions of the start.sh and stop.sh scripts:
```
chmod +x start.sh stop.sh
```

Once I upload the new versions of the applications to the application market, the applications will work with your image without any manual fixes.

G4DCP, 2016-05-22:
Thanks we can close this then.

Now I only need to work out how someone got the wspr code to coexist!

-- 
Peter

closed 2016-05-22 by G4DCP

pavel-demin, 2016-05-22:
> Somehow `. ./start.sh` seems to bypass this lack of permission?

Just noticed the dot in front of `./start.sh`.

`. ./start.sh` is equivalent to `source ./start.sh`. It reads the script and executes it line by line. The execution permission is not needed in this case.

Without dot or source in front of `./start.sh`, the scripts should have a correct execution permission.

pavel-demin, 2016-05-22:
> Now I only need to work out how someone got the wspr code to coexist!

There are some hints in https://github.com/pavel-demin/red-pitaya-notes/issues/213#issuecomment-210410340.

To better coexist with other application, the WSPR transceiver application should be refactored into a  monolithic program with eventually a web interface for configuration of call, grid, bands, etc.