pavel-demin/red-pitaya-notes #250: VNA Enhancements

ptamike, 2016-05-24:
I've been testing the VNA with a resistive series fixture and also with a Mini-Circuits ZFBDC20-6HP+ directional coupler and the results have been very encouraging. The series fixture in particular works very well.

However, when making measurements it would be helpful to have a more detailed readout. I can think of two options but I'm sure there are others:

1 - A tool-tip box that appears at the mouse pointer with a readout of the exact measurement at the pointer tip.
or
2 - Measurement cursors that can be moved on the display to show precise results.

Thanks,

Mike

pavel-demin, 2016-05-24:
I've found a library called [mpldatacursor](https://github.com/joferkington/mpldatacursor) that provides cursors for matplotlib. Here is how these cursors look like with the VNA client:
[image]

ptamike, 2016-05-24:
Hi Pavel,

That looks interesting and probably the basis of a solution. My ideal would be to be able to click the mouse on a point and see the reading in whatever measurement mode I've selected, i.e. SWR and Freq, Impedance and Freq, Return loss and Freq, etc. The alternative could be a pair of measurement cursors with a value readout at the point where they intersect.

Thanks for your persistence on this project. I'm sure it will be worthwhile.

PS: I've just been using the VNA to measure a home-made balun and also an antenna tuner. So far the results have all been as expected so it is already a very useful tool.

Mike

ptamike, 2016-05-24:
Hi again Pavel,

Just had a closer look at that library and I think it could be just right. I'd wrongly assumed from your drawing that it was automatically displaying the peak value of each trace. If I've understood it properly it will display the measurement wherever you click the mouse; just what I wanted. Thanks,  Mike

pavel-demin, 2016-05-25:
There is now a new version with mpldatacursor, variable bandwidth and frequency correction. Both Red Pitaya and PC applications should be updated.

pavel-demin, 2016-05-28:
There is now a version of vna.py with improved Smith chart. It can be zoomed/panned and it has cursors displaying frequency and impedance. If you want to give this new version a try, then you can download [vna.py](https://raw.githubusercontent.com/pavel-demin/red-pitaya-notes/master/projects/vna/client/vna.py) and copy it to the directory where the vna client is installed.

Here is how the new Smith chart looks like:
[image]

ptamike, 2016-05-29:
Hi Pavel,

A very helpful enhancement that makes the Smith chart much more usable. I see you've also tidied-up the measurement cursors so they give a more meaningful readout - very nice. One point though, could you amend the cursors to report frequency in MHz to 3 decimal places?
I'm currently using the VNA to retune my Butternut 9-band HF vertical antenna. As the adjustments interact between bands, I set the VNA to sweep from 3MHz to 33MHz with 3,000 points thus giving 10kHz resolution. I can then quickly zoom-in to any band to check the measurement.

I'm using a reflectance bridge with the Mini-Circuits 1-60MHz directional coupler as shown in the attached drawing. As a precaution, I have put a 50R termination on the unused Red Pitaya input 2. I've also added a 7.5dB matching attenuator between the Red Pitaya's output and the coupler as I'm a bit suspicious of the Red Pitaya's output impedance. I'm calibrating using OSL loads on the DUT (OUT) port of the Mini-Circuits coupler. The results are very good so far. I should have a ZX30-20-4+ single port coupler next week so that will give a simpler layout.

Thanks again for your work on this project - it's coming along very nicely.

Mike - G4WNC 
[Red Pitaya VNA bridge.pdf](https://github.com/pavel-demin/red-pitaya-notes/files/288282/Red.Pitaya.VNA.bridge.pdf)

pavel-demin, 2016-05-29:
Many thanks for the detailed description of your measurements. Very interesting.

I've just replaced all the format expressions with `%g`. This format outputs at most six significant digits and removes trailing zeroes. I'd say that this format works better than the previous one.

The new version is now available from the Red Pitaya application marketplace. Both Red Pitaya and PC applications should be updated.

This new version includes the following improvements and changes:
 - improved Smith chart with zoom, pan and cursors
 - twice faster sweep speed (1000 points/s)
 - changed bandwidths (1000, 100, 10, 1 Hz)
 - better formatted axes labels and cursors

ptamike, 2016-05-31:
Hi Pavel,

Very useful enhancements - thanks. However, I've noticed one glitch. Occasionally, the progress bar will freeze on 99% and after a wait will disappear without displaying any results or messages. It seems to be random but I'll keep monitoring to see if I can spot a pattern.
I thought you might like to see the results I'm getting using the test set-up I mentioned previously with my HF9V multi-band vertical antenna. I've also attached the s1p file so you see the detail.

[image]

pavel-demin, 2016-05-31:
> Occasionally, the progress bar will freeze on 99% and after a wait will disappear without displaying any results or messages.

Does it happen with the latest version or with one of the previous ones?

ptamike, 2016-05-31:
I only noticed it with the latest version.

pavel-demin, 2016-05-31:
Most probably, the freezes happen when CPU does not have enough time to process all the samples. The buffer between FPGA and CPU overflows and some samples get lost. I've tried to add some more optimizations but the the progress bar still freezes once in a while. Looks like switching to 1000 points/s was not a good idea.

ptamike, 2016-06-01:
Hi Pavel,

[image]
I think you're right. I'm also getting very different readings with this version. The attached image shows the same scan as the earlier posting but with very different results. There's a chance something has happened to the antenna but I don't think so. I'll do some more testing. Unfortunately I can't switch back to the old version to check.

Regards,

Mike

ptamike, 2016-06-06:
Hi Pavel,

I downloaded the latest build over the weekend and have been experimenting with reflectance bridges. I'm still getting the occasional sweep freeze but it seems to have improved a little.

I also built a variation of the 3-bead reflectance bridge mentioned in your VNA notes. The only difference with mine is that I have omitted the capacitors and loaded the co-ax from the bridge transformer to the output attenuator with ferrite beads.This has been working very well for HF antenna measurement from 1MHz through to 54MHz with repeatable results.
I have also built a bridge using a Mini-Circuits ZX30-20-4 directional coupler (see below). Although this is a forward coupler it is a symmetrical device so I am using it reversed, i.e. the Red-Pitaya output is connected to the Out and the DUT is connected to the coupler In. The CPL output feeds the Red Pitaya input which has a 50R termination. The Mini-Circuits coupler works really well and is producing similar results to the 3-bead bridge. However it's important to present all the coupler ports except the DUT with a 50R load, hence, the 50R terminating load on the RP input and the matching attenuator on the RP output.
Over the next couple of days I shall run some comparison measurements with the DG8SAQ VNWA to see how it fares.
[image]

**Extra Feature:** One feature that would be very convenient is a master calibration option, here's how I see it working.
The user does a calibration of the full frequency range using a step size of say 10kHz.
That calibration run is then stored as the master calibration.
The user can then define new scan ranges to take a closer look at the spectrum but using the previous calibration scan steps.
It would also be helpful if this could be combined with a continuous sweep and the display updated after each sweep.
This combination would be particularly powerful for antenna tuning work.

Hope you don't mind my requests but I think this application has lots of potential.

Regards,

Mike G4WNC

pavel-demin, 2016-06-07:
I've switched back to the 500 points/s max. sweep speed and added some optimizations to reduce the probability of buffer overflows and interface freezes. The new version is now available from the Red Pitaya application marketplace. Both Red Pitaya and PC applications should be updated.

ptamike, 2016-06-07:
Thanks - for the update. It's now working very well with repeatable measurements and no freezes.
As mentioned previously, I've now run some comparisons with the well respected DG8SAQ VNA.
For these tests, the Red Pitaya and DG8SAQ VNA were calibrated using the same test loads and then connected to my Butternut HF9V antenna (not at the same time!).
**Trace 1 - This is the result from the DG8SAQ VNA**
[image]

**Trace 2 - This is the Red-Pitaya VNA using the 3-bead reflection bridge and a 20dB attenuator between the Red-Pitaya Out and the bridge input**
[image]

**Trace 3 - This is the same setup as trace 2 but without the 20dB attenuator**
[image]

I re-calibrated the VNA after removing the 20dB attenuator but you can see that the reactive output impedance of the Red Pitaya is distorting the results.

The next task will be to use the DG8SAQ VNA to help design a matching network for the Red Pitaya outputs.

pavel-demin, 2016-06-07:
Thanks a lot for this comparison with DG8SAQ VNA! It's great to see that the results produced by the Red Pitaya VNA look very similar.

There is a known bug/feature of the Red Pitaya outputs. When the outputs aren't properly loaded they start to "resonate" at about 15-20 and 40-45 MHz. Maybe it could explain the results without the 20dB attenuator.

ptamike, 2016-06-07:
It's certainly odd. I've just done a VSWR run with the DG8SAQ looking into the reflectance bridge and as you can see, it's providing a pretty flat load from 1MHz through to 60MHz so you would expect the Red Pitaya to see that as a good 50R load.
**Trace 1 - 3-bead Reflectance Bridge input matching**
[image]
I'll do some more investigations looking into the RP.

ptamike, 2016-06-23:
Hi Pavel,

I've been away on holiday for a week but now back home. I thought it might be helpful if I attach details of my modified 3-bead balun reflectance bridge for use with the Red Pitaya. I built my prototype using Schmartboard with SMD (0805) resistors for the attenuators. As the Red Pitaya VNA only works up to 60MHz, construction is less critical than the original design.
This new design employs a 12dB input attenuator a) To provide a good termination for the RP's output and b) To isolate the RP output from the bridge circuit. On the other side of the bridge, I use an SMA tee connector with a 50R load connected to provide a good termination. I've run a number of tests using this bridge and the results compare very well with those from my DG8SAQ VNWA.

[image]

pavel-demin, 2016-06-26:
Hi Mike,

Many thanks for sharing schematics of your new bridge.

Last week, there was [a post](http://forum.redpitaya.com/viewtopic.php?t=266&p=5616#p5616) on the Red Pitaya forum with the following circuit:
[image]

I've just uploaded to the Red Pitaya application marketplace a new version of the VNA application with the following improvements and changes:
 - separate threads for readout and processing communicating via a ring buffer,
 - output level control that scales samples before sending them to DAC.

Best regards,

Pavel

Rolf-L, 2017-12-05:
Hi Pavel,

is it possible to set the GPIO-signal DIO0 always to "0", when you start the vna program?
"Connect" set DIO0 to "1" and my transmitter is activated via the PTT-line.
Everything else works great.

Best regards,
Rolf -DJ7TH-

pavel-demin, 2017-12-05:
Hi Rolf,

Thank you for testing the SD card image based on Alpine Linux.

> Is it possible to set the GPIO-signal DIO0 always to "0", when you start the vna program?

With the latest version (20171130), it can be done by editing [line 590 in vna.py](https://github.com/pavel-demin/red-pitaya-notes/blob/master/projects/vna/client/vna.py#L590) and replacing `self.set_gpio(1)` with `self.set_gpio(0)`.

I'll try to add a GUI element for controlling the GPIO pins to the next VNA version.

Best regards,

Pavel

Rolf-L, 2017-12-06:
Hi Pavel,

Thank you for the fast info.
Since I'm not a py expert, I will wait for the update :-)
It would be a great help if the user could easily set or reset the GPIO signals in an extended menu.

Best regards,
Rolf -DJ7TH-

satfan52, 2019-01-27:
Hello Pavel,

I am currently working on a filter and  in order to measure S11 and S21 successively, I have to calibrate in transmission mode, then when the measure is finished rewire everything in reflection mode to measure S11, then do another calibration in reflection mode. If for some reasons I change something to mu filter I have do come-back to  the transmission mode. Though S11 and S21 are of course directly correlated in an ideal filter, it would be much more convenient  if I could wire my set-up in transmission mode ONLY and then do 1 SINGLE calibration and then a single "sweep" get both S11 and S22. Even better would be to get also get S22 and S21 too as part of the same sweep, but I understand from what I have read that it is not a trivial exercise.

Regards
Peter