pavel-demin/red-pitaya-notes #254: HPSDR transceiver Attenuator reversed

ptamike, 2016-05-26:
Hi Pavel,

Just noticed a minor error in the HPSDR transceiver. The receive attenuator has two settings 0dB or -20dB and they seem to be reversed, i.e. choosing 0dB inserts 20dB of attenuation whilst choosing -20dB removes the attenuation.

Other than that it's working surprisingly well.

Mike - G4WNC

pavel-demin, 2016-05-26:
Hi Mike,

The following code on lines 324-326 in sdr-transceiver-hpsdr.c reads the preamplifier bit and sets pin DIO1_P to high or low:
https://github.com/pavel-demin/red-pitaya-notes/blob/master/projects/sdr_transceiver_hpsdr/server/sdr-transceiver-hpsdr.c#L324

In the current version, if the preamplifier bit is 1, then the pin DIO1_P is high. This way it can be used to connect or bypass a preamlifier. Are you proposing to invert it? I'm not sure how it would affect already existing setups that use this pin.

I can think of a solution with DIO1_P remaining as it is and with DIO2_P outputting inverted preamplifier signal.

Best regards,

Pavel

pavel-demin, 2016-09-24:
The outputs for 10 dB and 20 dB attenuators control are now connected to the pins DIO2_P - DIO3_P of the extension connector E1. So, the preamp output can be used to control a pream and the attenuator outputs can be used to control attenuators.

[image]

closed 2016-09-24 by pavel-demin