pavel-demin/red-pitaya-notes #264: Save data as textfile

sdrck, 2016-06-01:
First of all I want to thank you for this great work, Pavel. I think this application will be very helpful for me in the future. It would be great if one was able to save the data as a textfile to plot the data with another tool later. I think it is not possible at the moment, is it? Thank you in advance for a short answer.

Kind regards,
Sebastian

pavel-demin, 2016-06-01:
If you're talking about the VNA application, then it exports the data in s1p and s2p formats.

sdrck, 2016-06-01:
Hey Pavel,
thank you for your quick reply. I just measured a quick transfer function. I did'nt calibrate anything. I tried to save the data wit "Write s1p" and "Write s2p" but the files only contain a frequency vector and a lot of columns of nan, inf and zeros but not the DUT data (magnitude and phase). Do I have to calibrate the VAN (Open, short and load) to be able to get the data?
Thank you in advance again.

pavel-demin, 2016-06-01:
> Do I have to calibrate the VNA (Open, short and load) to be able to get the data?

Yes, you have to calibrate the VNA. To have some numbers in the s2p file, it's enough to do the 'Short' and 'DUT' measurements.

Don't hesitate to modify vna.py and add all the functionality that you need.

sdrck, 2016-06-02:
Thank you very much. I plotted the data in Matlab but I don't have access to the measured DUT data, right? The MA column contains magnitude and phase of the gain. Is it possible to get the DUT data without any modifications of the vna.py? Otherwise I will have to do that. Thank you for permitting modifications.

pavel-demin, 2016-06-02:
Out of curiosity, how the raw DUT data could be usable without calibration?

The saved .ini files contain all the data. But the format is probably not very easy to import to matlab.

Is where a preferred format for this kind of data? Otherwise I could add a button to write all the raw data in simple csv-like format.

sdrck, 2016-06-02:
Just in my measurement case the gain data looks strange. Of course the reason is my calibration. It seems that my "short" is not really a short but the DUT data looks as supposed. If I was able to get the raw data that would be great (csv would be perfect). In my case it is not important to get exact values because I just want to match a resonant circuit where I am interested in the resonance frequency. Thank you for your effort!

pavel-demin, 2016-06-02:
It just occurred to me that maybe you can communicate with the VNA application running on Red Pitaya directly from Matlab. Something like the following untested code should do the job:

```
t = tcpip('192.168.1.100', 1001);
fopen(t);

% start frequency in Hz
start = 100000;
command = bitor(bitshift(0, 28, 'uint32'), start, 'uint32');
fwrite(t, command, 'uint32');

% stop frequency in Hz
start = 60000000;
command = bitor(bitshift(1, 28, 'uint32'), stop, 'uint32');
fwrite(t, command, 'uint32');

% points
points = 600;
command = bitor(bitshift(2, 28, 'uint32'), points, 'uint32');
fwrite(t, command, 'uint32');

% decimation rate
rate = 1
command = bitor(bitshift(3, 28, 'uint32'), rate, 'uint32')
fwrite(t, command, 'uint32')

% sweep
command = bitshift(5, 28, 'uint32')
fwrite(t, command, 'uint32')

adc1 = complex(ones(points), ones(points));
adc2 = complex(ones(points), ones(points));
dac1 = complex(ones(points), ones(points));
for i = 1 : points
  adc1(i) = complex(fread(t, 4, 'float32'), fread(t, 4, 'float32'));
  adc2(i) = complex(fread(t, 4, 'float32'), fread(t, 4, 'float32'));
  dac1(i) = complex(fread(t, 4, 'float32'), fread(t, 4, 'float32'));
end

fclose(t);

data = adc1 / dac1;
```

sdrck, 2016-06-02:
Thank you very much for that script. I will test it! Just of pure interest, I would like to know if it is possbile to measure S11 and S21 as well as S21 and S22? I'm not quite sure how to connect the RP to my network correctly.
Thank you for your effort again.

pavel-demin, 2016-06-02:
>I would like to know if it is possible to measure S11 and S21 as well as S21 and S22?

You can find a schematics of a 2-port VNA on the Red Pitaya forum:
http://forum.redpitaya.com/viewtopic.php?t=266&p=5266#p5266
[image]

I think that with such an extension board it's possible to measure all the 2-port S-parameters.

pavel-demin, 2016-06-07:
New version has now a write csv button.

closed 2016-06-07 by pavel-demin