pavel-demin/red-pitaya-notes #265: Skimmer Spots on both 15 and 10 m

w2naf, 2016-06-02:
Hi, Pavel.

I set up a Red Pitaya-based Reverse Beacon Network skimmer this weekend, and overall was extremely impressed. For the most part, it worked beautifully.

However, I received a report from another ham saying that my receiver spotted them on 10 m even though they were transmitting on 15 m. They said that the reported kHz was correct, but the MHz was not.

I was able to replicate this today. By calling CQ de K2MFF on 21035 kHz. My skimmer (W2NAF) reported spots on both 21035 and 28035 kHz (see attached screenshot).

Do you know how to resolve this?

Thanks and 73,
Nathaniel, W2NAF

[image]

pavel-demin, 2016-06-02:
Hi Nathaniel,

Thanks for spotting this problem. Could you please send me screenshots of the CW skimmer settings?

Best regards,

Pavel

pavel-demin, 2016-06-02:
Found it. `data4` should be replaced with `data5` on lines 348 and 380 in [sdr-receiver-hpsdr.c](https://github.com/pavel-demin/red-pitaya-notes/blob/master/projects/sdr_receiver_hpsdr/server/sdr-receiver-hpsdr.c#L348).

pavel-demin, 2016-06-02:
> I think I have seen this on the wspr receiver as well, is that possible given what you have discovered?

If you're talking about the multiband WSPR transceiver, then the code in [write-c2-files.c](https://github.com/pavel-demin/red-pitaya-notes/blob/master/projects/sdr_transceiver_wspr/write-c2-files.c) isn't the same. It's currently running on my Red Pitaya without any problem.

Could you please provide more details about the problem that you observed?

pavel-demin, 2016-06-02:
I've fixed the problem with commit https://github.com/pavel-demin/red-pitaya-notes/commit/c76ef2ff918b296b177eee7b563f5da89e6e43d4 and updated the customized SD card image zip file.

Could you please check if the new version works better?

w2naf, 2016-06-03:
> I've fixed the problem with commit c76ef2f <https://github.com/pavel-demin/red-pitaya-notes/commit/c76ef2ff918b296b177eee7b563f5da89e6e43d4> and updated the customized SD card image zip file.

Hi, Pavel. Thank you so much. I just tested it out, and it looks like it is working great! I really appreciate you writing this SDR program.

Incidentally, I am working on a project called HamSCI, the Ham radio Science Citizen Investigation. We are going to encourage more people to set up RBN nodes and then work on getting ionospheric science out of the data set. See http://hamsci.org <http://hamsci.org/> and http://hamsci.org/eclipse <http://hamsci.org/eclipse>.

73,
Nathaniel, W2NAF

pavel-demin, 2016-06-03:
Thanks for the test.

> We are going to encourage more people to set up RBN nodes

Are you planning to use only CW skimmer or some other systems/protocols like for example WSPR?

w2naf, 2016-06-03:
You're welcome.

I'm going to start by focusing on Skimmer, but hopefully will fold in
things like WSPR eventually, too.

closed 2016-06-03 by pavel-demin