pavel-demin/red-pitaya-notes #267: Is SPI enabled in the Debian build?

XYEFA, 2016-06-02:
Hello Pavel,

great work with the Debian build. Can you please tell me if SPI and I2C are enabled in your kernel? If not, how do I enable them.

Regards!

pavel-demin, 2016-06-02:
Yes, SPI and I2C are enabled.

XYEFA, 2016-06-02:
Pavel - Thank you for your quick reply! Much appreciated. A few more questions:

1) Where is SPI configured in the kernel?

2) Are there any links to some quick SPI or I2C tests (in C/C++)?

Regards!

pavel-demin, 2016-06-03:
> Where is SPI configured in the kernel ?

https://github.com/pavel-demin/red-pitaya-notes/blob/master/patches/linux-xlnx-xilinx-v2016.1.patch#L74
https://github.com/pavel-demin/red-pitaya-notes/blob/master/patches/devicetree.patch#L58

> Are there any links to some quick SPI or I2C tests (in C/C++)?

SPI:
http://redpitaya.com/examples-new/spi/
http://armbedded.eu/node/318

I2C:
http://redpitaya.com/examples-new/i2c/
https://github.com/pavel-demin/red-pitaya-notes/blob/master/projects/sdr_transceiver_hpsdr/server/sdr-transceiver-hpsdr.c

> For instance is this example applicable - http://redpitaya.com/examples-new/spi/ ?

I think that this example should work after replacing `spidev1.0` with `spidev2.0`.

closed 2016-06-03 by pavel-demin

XYEFA, 2016-06-09:
This was very helpful and it looks like I can now open and write to the device. I am still looking into the various SPI MODE settings ... I am trying to do loopback test.

Regards!

XYEFA, 2016-06-13:
I connected the MOSI pin to the MISO pin and ran the test and saw the bytes come out. Everything is working.