pavel-demin/red-pitaya-notes #269: Audio from the HPSDR

on3vna, 2016-06-06:
Hey Pavel,
Maybe i missed somehow somewhere something, but how can we get audio from the red-pitaya?

eg. on the flex-1500 we have a jack to connect a speaker how can this be achieved on the red-pitaya?

I have been playing around in the weekend with the HPSDR transceiver , great job.

73 de ON3VNA
Johan

pavel-demin, 2016-06-06:
With the SDR transceiver compatible with HPSDR, the only option at the moment is to connect a speaker to a computer running the SDR programs (PowerSDR mRX PS, QUISK, etc).

With the embedded SDR transceiver, I use a USB audio adapter or a USB headset. For example, the following two work for me:
https://www.amazon.de/gp/product/B000JSDOMO
https://www.amazon.de/gp/product/B005625U1Q

Just yesterday, I've learned an easy way to input and output audio to a C program via standard input and output using the `arecord` and `aplay` programs. I'm using the following command in the [DiscoRedTRX](https://github.com/pavel-demin/DiscoRedTRX) project:
```
arecord -fFLOAT_LE -c2 -r48000 | ./main | aplay -fFLOAT_LE -c2 -r48000
```

Something similar could be added to the SDR transceiver compatible with HPSDR. This way, PowerSDR mRX PS receives I/Q samples from Red Pitaya, demodulates them and sends audio samples back to Red Pitaya. These audio samples could be sent to the `aplay` command for a playback via a USB audio adapter.

on3vna, 2016-06-06:
Is it possible to have that same USB device connected on the RED-PITAYA in the HPSDR version of the software?

on3vna, 2016-06-06:
Pavel,
That would be great, as I use VAC to connect to DM780 and JT65 software.
It is nice to hear some audio when moving around the HF waves.
73
Johan

pavel-demin, 2016-06-06:
Are you using both VAC1 and VAC2 interfaces in PowerSDR mR PS?

on3vna, 2016-06-06:
It seems that VAC1 only works with RX1 and VAC2 only works with RX2.
A dirty trick was turn on both RX and VAC1 to DM780 and VAC2 to speakers and keep the vfo sync.

pavel-demin, 2016-06-06:
> VAC2 only works with RX2

I didn't know about it.

I've just realized that the USB audio won't work with the SD card image provided by Red Pitaya. It doesn't have some of the necessary drivers enabled.

You'll need to use my SD card image from [this link](http://pavel-demin.github.io/red-pitaya-notes/debian-ecosystem).

on3vna, 2016-06-07:
I have a SD card image with the debian eco image in use

pavel-demin, 2016-06-08:
I've just added the audio playback to sdr-transceiver-hpsdr.c with the commit https://github.com/pavel-demin/red-pitaya-notes/commit/7a4c72ee869957a7f9719dba2fce3a0494864b16

You can try to compile and run sdr-transceiver-hpsdr with the following commands:
```
apt-get install alsa-utils
gcc sdr-transceiver-hpsdr.c -o sdr-transceiver-hpsdr -D_GNU_SOURCE -lm -lpthread
./sdr-transceiver-hpsdr | aplay -fS16_BE -c2 -r48000
```

pavel-demin, 2016-06-09:
I've just uploaded the latest version (0.94-1165) of the SDR transceiver compatible with HPSDR to the Red Pitaya application marketplace.

The USB audio works with my SD card image from this [this link](http://pavel-demin.github.io/red-pitaya-notes/debian-ecosystem). The latest version (20160609) of the SD card image comes with alsa-utils installed out of the box. With the previous version of the SD card image, alsa-utils should be installed manually: `apt-get install alsa-utils`.

on3vna, 2016-06-10:
Thanks Pavel.
Just have to buy a USB sound device and as soon as I have it I will test it.
Let you know the results.
73

on3vna, 2016-06-13:
Just installed the new eco image.
Inserted the usb sound card device Teratec Aureon Dual USB and works like a charme.
Thanks a lot.
73

closed 2016-06-13 by on3vna