pavel-demin/red-pitaya-notes #276: Multiband WSPR - Tune option

ptamike, 2016-06-09:
Hi Pavel,

I'm doing some work with your WSPR transceiver at the moment and it would be very useful to have a command-line accessed 'Tune' facility for the Transmitter. This would force the transmitter into WSPR transmit mode for say 30 seconds or even an on/off toggle. This would be very useful for setting-up and measuring transmit filters and amplifiers.

Thanks,

Mike - G4WNC

pavel-demin, 2016-06-09:
Hi Mike,

I'm not sure if I understand why this functionality should be added to the WSPR transceiver. There are already several existing options to generate continuous sine waveforms:
 - Generator & Oscilloscope application
 - VNA application (with start and stop set to the same frequency) 
 - generate command line utility

Anyway, it's possible to create a new C program (tune.c) based on [transmit-wspr-message.c](https://github.com/pavel-demin/red-pitaya-notes/blob/master/projects/sdr_transceiver_wspr/transmit-wspr-message.c). It would be enough to remove all the WSPR related code and set number of loop iterations to `seconds*125000000/85333332`.

Best regards,

Pavel

ptamike, 2016-06-09:
Hi Pavel,

I asked because I'm experimenting with PAs and filters for the WSPR Tx and wanted to be able to quickly switch the Tx to transmit for testing purposes as I didn't know what output levels you were using.
You've provided the answer in your response so I've just created an executable bash script that calls transmit-wspr-message.

In case anyone else wants to use it, I've attached the script. Before you can use the file you will need to install sudo (apt-get install sudo) and then make the script executable (chmod +x tune.sh).
To run the script and start transmission just type ./tune.sh
Once you've run the file for the first time you can run it again by pressing the up-arrow/return

[tune.zip](https://github.com/pavel-demin/red-pitaya-notes/files/307950/tune.zip)

Thanks for your help as always,

Mike

closed 2016-06-25 by ptamike