pavel-demin/red-pitaya-notes #279: Changing the drive level of tune doesn't work.

on3vna, 2016-06-11:
When setting the TUNE level in the transmit options to 10 we would assume that the output power would be 10% of the maximum output level.
There is no change at all in the output level it is always fulll transmit power.

pavel-demin, 2016-06-11:
I don't think that I can do anything about this problem. It's a known limitation of the Red Pitaya board.

Red Pitaya does not have any hardware for the drive control. I suppose that some kind of variable or fixed amplifier or attenuator should be used to adjust the output level. I've seen a nice set of fixed value SMA attenuators at the FUNKAMATEUR online shop:
http://www.box73.de/advanced_search_result.php?keywords=Dampfungsglied+sma

Another idea discussed on the Red Pitaya forum was to use the OUT2 output to control an external amplifier:
http://forum.redpitaya.com/viewtopic.php?t=265&p=5292#p5292

on3vna, 2016-06-13:
Maybe this could be used : AD8321 or a replacement for it. I have still 2 in the my junkbox.

on3vna, 2016-06-13:
AD8324 could be used 3.3V supply or AD8325 5V supply.

pavel-demin, 2016-06-13:
A discussion of variable attenuator based on [BAP70Q](http://www.nxp.com/documents/data_sheet/BAP70Q.pdf) (quad PIN diode attenuator) can be found at  [this link](http://www.qrpforum.de/index.php?page=Thread&threadID=10089&pageNo=12).

on3vna, 2016-06-13:
Advantage of the ADXXXX is that it is to be programmed by SPI.

pavel-demin, 2016-06-13:
I've just found [an interesting article](http://mwrf.com/semiconductor/diode-quad-foundation-pin-diode-attenuator) about a variable attenuator based on [HSMP-3816](http://www.avagotech.com/docs/AV02-0407EN).

Looks like it works quite well between -4 and -20 dB.

pavel-demin, 2016-06-13:
> Advantage of the ADXXXX is that it is to be programmed by SPI.

I think that a pin diode attenuator could be combined with a small SPI or I2C DAC with an internal reference. Something like MCP4821, LTC2630, LTC2631, TLV5636, etc.

on3vna, 2016-06-13:
Both solutions could be tried out and see what is best. Always available for testing -;)

on3vna, 2016-06-13:
Pavel,
I was looking around and found this:

PE4302 Digital RF step attenuator.

This could be used in the TX path and also in both receiver paths and is not so expensive.
you can find them on ebay .
http://www.aliexpress.com/store/product/PE4302-Digital-RF-Attenuator-Module-High-Linearity-0-5dB/1829717_32473840163.html

on3vna, 2016-06-14:
Hello Pavel,

http://wiki.redpitaya.com/index.php?title=Command_line_utilities

Here i found that via the command line utility it is possible to set the amplitude of the generated signal:

redpitaya> generate
generate version 0.90-299-1278

Usage: generate   channel amplitude frequency <type>

       channel     Channel to generate signal on [1, 2].
       amplitude   Peak-to-peak signal amplitude in Vpp [0.0 - 2.0].
       frequency   Signal frequency in Hz [0.0 - 6.2e+07].
       type        Signal type [sine, sqr, tri].

Table 7: Parameters of Signal generator utility Name 	Type 	Range 	Description
channel 	int 	1 / 2 	Output channel selection
amplitude 	float 	0 - 2 [V] 	Maximal output signal is 2 V peak to peak
freq 	float 	0 - 62000000* [Hz] 	Frequency can be generated from 0 Hz (DC signal) on*.
<type> 	string 	sine / sqr / tri 	Optional parameter. Signal shape type (sine – sine wave signal, sqr – square signal, tri – triangular signal). If omitted, sine is used.

* To generate smooth signals, not exceeding Back-End bandwidth, limitations are:

    62 MHz (62000000) for sine wave 

Is that not something that can be implemented in the transceiver also for the output drive ?

Rgds
Johan

pavel-demin, 2016-06-14:
The generate program scales the amplitude digitally before it's sent to DAC.

In case of SDR applications, I don't think that scaling the input to the DAC is a good idea. It decreases the effective DAC resolution, SNR, etc.

Then it depends on what exactly you want to achieve.

If it's just for the test signals, then it's possible to generate test signals with digitally scaled amplitude via Setup/Tests. Level controls amplitude. Freq #1 and Freq #2 can be set to 0 to generate only one frequency.

If it's for the transmitted signals, then with Radio Model set to HPSDR the signal amplitude  can be digitally scaled with the Drive slider.

on3vna, 2016-06-14:
Well we had a discussion here at the QRL and indeed scaling the DAC, you will pay in resolution SNR etc.

So you say that with the type of radio set to HPSDR iso hermes the output level can be changed with the drive slider ?

pavel-demin, 2016-06-14:
> So you say that with the type of radio set to HPSDR iso hermes the output level can be changed with the drive slider ?

Here is what is written in [HPSDR - USB Data Protocol](http://svn.tapr.org/repos_sdr_hpsdr/trunk/Documentation/USB_protocol_V1.58.doc) :
When using Hermes or PennyLane the Transmitter output level is set by the drive level value (C0 = 0b0001001x, C1 = 0x00 to 0xFF) and not by the amplitude of the I&Q signals. These are held to a peak value of +/-1.0 by AGC action in the PC DSP code. When using Penelope the amplitude of the I&Q signals controls the output level. Set the drive level (C1) to 0x00 when Penelope is selected and whilst the transmitter is not active.

on3vna, 2016-06-16:
Pavel,

The output of the DAC could be changed by putting in serie with the resistors used in AVIRES and BVIRES (R47 and R48 ) a FET like is done with some DDS chips. Hence it would be possible to drive the fet with a DC voltage and control the output voltage of the DAC and still keep the resolution etc.

Much simpler then using the AD chips or PIN diodes. Of course this needs some soldering on the red pitaya.

Rgds
Johan

pavel-demin, 2016-06-16:
Yes, it's similar to how it's done on the Hermes board:
 - PowerSDR sends drive level (0-255) to Hermes (see protocol description, C0 = 0001001x, C1 = Hermes/PennyLane Drive Level)
 - then see the code in http://svn.tapr.org/repos_sdr_hpsdr/trunk/Hermes/Source/Hermes_v3.1/Hermes.v
 - the value of the drive level is stored in the IF_Drive_Level register
 - this value is used to generate a PWM signal that is output from FPGA as DAC_ALC
 - then see page 9 of the schematics in http://openhpsdr.org/support/Hermes/Hermes_Final_Changes.pdf
 - the DAC_ALC (PWM) signal is buffered, filtered and connected to the REFIO and FSADJ pins of the DAC chip

If I'm not mistake FSADJ is similar to AVIRES and BVIRES.

on3vna, 2016-07-03:
I have given the info to RED PITAYA team concerning the implementation with the FET. I would love to try it out but the components on the pba are to small for my tools and eyes. I dont want to screw up my red pitaya.

on3vna, 2016-09-02:
code changed for the HW modification.

pavel-demin, 2016-09-02:
Great! I've added this code with a few minor changes to my repository: https://github.com/pavel-demin/red-pitaya-notes/commit/384c6b035ec3220565f649e038f96afafc58df0a.

I've just replaced the 8-bit command 0xAF with the 16-bit command 0xA9.

on3vna, 2016-09-15:
Pavel,
I think we can close this now as we have 3 ways of setting the drive level.
2 hardware solutions and 1 software solution

73 de ON3VNA
rgds
Johan

closed 2016-09-15 by on3vna

ka6s, 2016-09-17:
Pavel,

You have us all very curious.  I was wondering if you could publish a
future pin-out utilization possibly?

1. You have checked--in connecting the A/Ds recently - what are these going
to source SWR F/R?? What is the pin assignment?
2.  If you are going to bring in/out I2S channels for an external Codec -
what is the pin assignment there?
3. What is the Software solution for the Drive?

Inquiring minds are REALLY curious!

Thanks,

Steve KA6S

pavel-demin, 2016-09-17:
> 1. You have checked--in connecting the A/Ds recently - what are these going to source SWR F/R?? What is the pin assignment?

Yes, the slow analog inputs are used for the Fwd and Rev power measurement. Fwd - Analog input 0, Rev - Analog input 1.

> 2. If you are going to bring in/out I2S channels for an external Codec - what is the pin assignment there?

The I2S interface is sharing pins with the ALEX interface. So, the two can't be used simultaneously. 

> 3. What is the Software solution for the Drive?

It's what I didn't want to implement [initially](https://github.com/pavel-demin/red-pitaya-notes/issues/279#issuecomment-225893923) but after all I've added a multiplier just before the DAC interface. This way the DAC samples are scaled down digitally. The scale factor is controlled by the Drive slider in PowerSDR mRX PS.

ka6s, 2016-09-18:
>    1. If you are going to bring in/out I2S channels for an external Codec
>    1. What is the Software solution for the Drive?

Thank you Pavel - that helps greatly!

Steve