pavel-demin/red-pitaya-notes #280: Pure signal is not working with HPSDR transceiver

on3vna, 2016-06-11:
I have tried the pure signal but it doesn't seems to work with the HPSDR transceiver.

The Two tone is generated but the feedback level stays black and also the correcting stays black when starting the autocorrection.

pavel-demin, 2016-06-11:
Does your configuration look like it's shown on the screenshots at [this link](http://pavel-demin.github.io/red-pitaya-notes/sdr-transceiver-hpsdr#amplifier-linearization)? Have you tried to adjust the SetPk value?

More screenshots with various SetPk settings can be found on the CQ-NRW forum:
http://forum.cq-nrw.de/viewtopic.php?f=17&t=88#p723
http://forum.cq-nrw.de/viewtopic.php?f=11&t=67

There is also a nice summary of possible problems and solutions on the Red Pitaya forum:
http://forum.redpitaya.com/viewtopic.php?t=265&p=5473#p5473

on3vna, 2016-06-11:
The cntrl-alt-i shouws
[image]
 the info as in the links above, but the pure signal goes to state 4 and stays there notting happens.

on3vna, 2016-06-11:
I am using a directional coupler HP-779D that has -20 dB connected to IN2 and 50 ohm at the output of the coupler.

pavel-demin, 2016-06-11:
The feedback level is critical, and should be between 1 and 2 Vpp.

on3vna, 2016-06-12:
Indeed the feedback level is critical.
With straight connection between TX1 and RX2 we are close to ADC Overload.
With a 6 dB attenuator we do not have enough signal Yellow led. 
The other attenuators i have for the moment are 10 dB and 20 dB. 
Have to look for a 3 dB attenuator.
I will keep you posted of the result.

pavel-demin, 2016-06-13:
According to [this post](http://forum.cq-nrw.de/viewtopic.php?f=18&t=34&p=606&hilit=edwin#p606), setting the input jumper at the 2-5 position would provide a 7 dB gain.

If I'm not mistaken, it could be combined with a 10 dB attenuator to obtain 3 dB attenuation.

on3vna, 2016-06-13:
Ok, I will try again this evening with the changed jumpers.
I will do again the calibration first.

on3vna, 2016-07-03:
The jumpers have been changed, but unfortunately did have any spare time yet to try it out.

pavel-demin, 2016-09-24:
I'm closing this issue because it's now confirmed that Pure Signal works well with Red Pitaya.

Here is a link to a very detailed video by Yevgeni RZ3QS showing how Pure Signal works with his Red Pitaya:
https://www.youtube.com/watch?v=eECiwHu0vOY

closed 2016-09-24 by pavel-demin