pavel-demin/red-pitaya-notes #289: Compiling a Raspberry Pi LCD Device Driver (Device Tree) - How?

XYEFA, 2016-06-16:
Hello:

I am trying to adapt a Raspberry Pi Device Tree and make it work on the Red Pitaya. You can see my thread here: http://forum.redpitaya.com/viewtopic.php?f=14&t=1409 ... I am working the issue slowly and trying to do this.

This is the Display: https://www.adafruit.com/product/2097

This is the Device Driver: https://github.com/notro/fbtft

How would I build this on your Debian build for the Red Pitaya?

<<

pavel-demin, 2016-06-17:
I'd follow the instructions from the fbtft/README:
>cd drivers/video/fbdev
git clone https://github.com/notro/fbtft.git
Add to drivers/video/fbdev/Kconfig:   source "drivers/video/fbdev/fbtft/Kconfig"
Add to drivers/video/fbdev/Makefile:  obj-y += fbtft/

XYEFA, 2016-06-17:
Hello Pavel - I am assuming that I need to make some updates to the Device Tree (since it is for the Raspberry Pi). Is this thinking correct?

pavel-demin, 2016-06-17:
Sorry, but I have no idea how to make this driver working with Red Pitaya. Why not read the driver documentation or ask the authors of the driver?

pavel-demin, 2016-06-17:
I've just realized that your question was already answered on the Red Pitaya forum:
http://forum.redpitaya.com/viewtopic.php?t=1409&p=5580#p5580

I don't think that I can add anything to this answer.

closed 2016-06-17 by pavel-demin

XYEFA, 2016-06-17:
I think I have a working plan of how to migrate this driver from a Raspberry Pi to the Red Pitaya. Since I used the build environment from your tree, I wanted to make sure I was on the right track. I will update this thread if all goes well.

pavel-demin, 2016-06-17:
I had a quick look at the fbtft documentation and found a solution based on the devicetree overlays:
https://github.com/notro/fbtft/wiki/FBTFT-RPI-overlays

Some more information about devicetree overlays can be found at the following link:
https://www.raspberrypi.org/documentation/configuration/device-tree.md

I also found the fbtft driver in `drivers/staging/fbtft`.

I enabled devicetree overlays and all the fbtft components with commit https://github.com/pavel-demin/red-pitaya-notes/commit/f0742db41309a1e244b6d8f3544e689f84e9553a.

XYEFA, 2016-06-17:
Thanks Pavel, I am reading this and the Xilinx SPI documentation: http://www.wiki.xilinx.com/Linux+SPI+Driver