pavel-demin/red-pitaya-notes #293: Multiband WSPR Bands List

Ch-6, 2016-06-20:
Hello Pavel,
First of all may I congratulate you for your great work in designing such a brilliant and interesting project for the Red Pitaya.
If I may ask a question, I have the WSPR receiver working really well and it is uploading the spots as it should do.
The problem I have is when I change any of the band settings using WinSCP it stops working. If I then restore it back to default everything works again.
For example if I disable the 18MHz band using // nothing works.  If I then remove the // everything works again. .Maybe I am missing something.
The same thing happens when I add a comma to the 21MHz band.
Any help would be greatly appreciated.
Many thanks,
Steve

pavel-demin, 2016-06-20:
Hello Steve,

Thank you for your kind words. I'm glad that you like my WSPR receiver project.

When receiver stops working, looking at the error messages in `/dev/shm/decode-wspr.log` sometimes helps to understand the problem.

Current version is quite strict about the configuration file format. It requires exactly eight bands to be enabled. So, if you want to disable the 18MHz band, then you should enable any other band.

The bands should be comma separated without trailing commas.

Best regards,

Pavel

Ch-6, 2016-06-20:
Many thanks Pavel, the 8 band selection sorted it out.
Kind regards, Steve

closed 2016-07-03 by pavel-demin