pavel-demin/red-pitaya-notes #305: sound gets lost by the soundcard device plugged in USB of the red

on3vna, 2016-07-01:
I have seen on several occasions that the USB sound get lost and is not to be recovered:
So far i'm not able to perform a reproduction when or what caused it.
It seems to me once yoy have entered this you only get out of this error by reflashing the SD card

root@red-pitaya:~# tail -f /var/log/redpitaya_nginx/debug.log
2016/07/01 18:01:39 [notice] 1596#0: signal 17 (SIGCHLD) received
2016/07/01 18:01:39 [info] 1596#0: waitpid() failed (10: No child processes)
2016/07/01 18:01:47 [notice] 1596#0: signal 17 (SIGCHLD) received
2016/07/01 18:01:47 [info] 1596#0: waitpid() failed (10: No child processes)
2016/07/01 18:01:47 [notice] 1596#0: signal 17 (SIGCHLD) received
2016/07/01 18:01:47 [info] 1596#0: waitpid() failed (10: No child processes)
2016/07/01 18:01:55 [notice] 1596#0: signal 17 (SIGCHLD) received
2016/07/01 18:01:55 [info] 1596#0: waitpid() failed (10: No child processes)
2016/07/01 18:01:55 [notice] 1596#0: signal 17 (SIGCHLD) received
2016/07/01 18:01:55 [info] 1596#0: waitpid() failed (10: No child processes)
2016/07/01 18:02:03 [notice] 1596#0: signal 17 (SIGCHLD) received
2016/07/01 18:02:03 [info] 1596#0: waitpid() failed (10: No child processes)
2016/07/01 18:02:03 [notice] 1596#0: signal 17 (SIGCHLD) received
2016/07/01 18:02:03 [info] 1596#0: waitpid() failed (10: No child processes)
Calling application: start
Loading specific FPGA from: '/opt/redpitaya/www/apps/sdr_transceiver_hpsdr/sdr_transceiver_hpsdr.bit'
Loading application: '/opt/redpitaya/www/apps/sdr_transceiver_hpsdr/controllerhf.so'
Cannot resolve 'ws_set_params_interval' function.
Cannot resolve 'ws_set_signals_interval' function.
Cannot resolve 'ws_get_params_interval' function.
Cannot resolve 'ws_get_signals_interval' function.
Cannot resolve 'ws_set_params' function.
Cannot resolve 'ws_get_params' function.
Cannot resolve 'ws_set_signals' function.
Cannot resolve 'ws_get_signals' function.
Cannot resolve 'ws_set_demo_mode' function.
Cannot resolve 'verify_app_license' function.
Cannot resolve 'ws_gzip' function.
Starting HPSDR transceiver server.
2016/07/01 18:02:07 [notice] 1596#0: signal 17 (SIGCHLD) received
2016/07/01 18:02:07 [info] 1596#0: waitpid() failed (10: No child processes)
Application loaded succesfully!random: nonblocking pool is initialized
Playing raw data 'stdin' : Signed 16 bit Big Endian, Rate 48000 Hz, Stereo
underrun!!! (at least 0.042 ms long)
underrun!!! (at least 0.015 ms long)
underrun!!! (at least 0.015 ms long)
underrun!!! (at least 0.013 ms long)
underrun!!! (at least 0.015 ms long)
underrun!!! (at least 0.013 ms long)
underrun!!! (at least 0.012 ms long)

pavel-demin, 2016-07-01:
> you only get out of this error by reflashing the SD card

I didn't realize that nginx writes a lot of logs to /var/log/redpitaya_nginx/debug.log. Could it be that the SD card gets full and it somehow slows down the SDR server running on Red Pitaya?

on3vna, 2016-07-02:
root@red-pitaya:~# df
Filesystem     1K-blocks   Used Available Use% Mounted on
/dev/root        1000568 728496    204028  79% /
devtmpfs          230276      0    230276   0% /dev
tmpfs             238608      0    238608   0% /dev/shm
tmpfs             238608   4400    234208   2% /run
tmpfs               5120      4      5116   1% /run/lock
tmpfs             238608      0    238608   0% /sys/fs/cgroup
/dev/mmcblk0p1     11222   5608      5614  50% /boot
root@red-pitaya:~#

on3vna, 2016-07-02:
root@red-pitaya:/# fdisk -l

Disk /dev/ram0: 16 MiB, 16777216 bytes, 32768 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes
Disk /dev/ram1: 16 MiB, 16777216 bytes, 32768 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes
Disk /dev/ram2: 16 MiB, 16777216 bytes, 32768 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes
Disk /dev/ram3: 16 MiB, 16777216 bytes, 32768 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes
Disk /dev/ram4: 16 MiB, 16777216 bytes, 32768 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes
Disk /dev/ram5: 16 MiB, 16777216 bytes, 32768 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes
Disk /dev/ram6: 16 MiB, 16777216 bytes, 32768 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes
Disk /dev/ram7: 16 MiB, 16777216 bytes, 32768 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes
Disk /dev/ram8: 16 MiB, 16777216 bytes, 32768 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes
Disk /dev/ram9: 16 MiB, 16777216 bytes, 32768 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes
Disk /dev/ram10: 16 MiB, 16777216 bytes, 32768 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes
Disk /dev/ram11: 16 MiB, 16777216 bytes, 32768 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes
Disk /dev/ram12: 16 MiB, 16777216 bytes, 32768 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes
Disk /dev/ram13: 16 MiB, 16777216 bytes, 32768 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes
Disk /dev/ram14: 16 MiB, 16777216 bytes, 32768 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes
Disk /dev/ram15: 16 MiB, 16777216 bytes, 32768 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes
Disk /dev/mmcblk0: 7.4 GiB, 7948206080 bytes, 15523840 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x6f7fd5bb

Device         Boot Start     End Sectors  Size Id Type
/dev/mmcblk0p1       8192   30719   22528   11M  e W95 FAT16 (LBA)
/dev/mmcblk0p2      30720 2097151 2066432 1009M 83 Linux

root@red-pitaya:/#

pavel-demin, 2016-07-03:
```
root@red-pitaya:~# df
Filesystem 1K-blocks Used Available Use% Mounted on
/dev/root 1000568 728496 204028 79% /
```

The SD card isn't full. So, it should be something else.

on3vna, 2016-07-08:
Pavel,

Here a trace on the ALSA when starting up HPSDR on the RED PITAYA and then starting up HPSDR pc:

There you will see pop up:
read(0, "\372\5\372\5\367\272\367\272\367y\367y\371\4\371\4\373\244\373\244\376|\376|\0\323\0\323\2L\2L"..., 4096) = 4096
ioctl(5, SNDRV_PCM_IOCTL_SYNC_PTR, 0x3c420) = 0
ioctl(5, SNDRV_PCM_IOCTL_SYNC_PTR, 0x3c420) = 0
ioctl(5, SNDRV_PCM_IOCTL_SYNC_PTR, 0x3c420) = 0
ioctl(3, SNDRV_TIMER_IOCTL_STOP, 0x54a1) = 0
clock_gettime(CLOCK_MONOTONIC, {986, 326250127}) = 0
ioctl(5, SNDRV_PCM_IOCTL_SYNC_PTR, 0x3c420) = 0
clock_gettime(CLOCK_MONOTONIC, {986, 326547410}) = 0
clock_gettime(CLOCK_MONOTONIC, {986, 326707030}) = 0
write(2, "underrun!!! (at least 0.457 ms l"..., 37) = 37

Starting HPSDR transceiver server.
2016/07/08 15:06:43 [notice] 1593#0: signal 17 (SIGCHLD) received
2016/07/08 15:06:43 [info] 1593#0: waitpid() failed (10: No child processes)
Application loaded succesfully!Playing raw data 'stdin' : Signed 16 bit Big Endian, Rate 48000 Hz, Stereo
underrun!!! (at least 0.544 ms long)
underrun!!! (at least 0.457 ms long)
underrun!!! (at least 0.538 ms long)
underrun!!! (at least 0.872 ms long)
underrun!!! (at least 0.802 ms long)
underrun!!! (at least 0.489 ms long)
underrun!!! (at least 0.533 ms long)
underrun!!! (at least 0.675 ms long)
underrun!!! (at least 0.393 ms long)
underrun!!! (at least 0.516 ms long)
underrun!!! (at least 0.524 ms long)
underrun!!! (at least 0.387 ms long)

root     11473     1  1 15:06 ?        00:00:00 aplay --format=S16_BE --channels
root     11475  1719  0 15:06 ttyPS0   00:00:00 ps -ef
root@red-pitaya:~# strace -p 11473
Process 11473 attached
read(0, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0", 24) = 24
read(0, "\0\0", 2)                      = 2
write(2, "Playing raw data 'stdin' : ", 27) = 27
write(2, "Signed 16 bit Big Endian, ", 26) = 26
write(2, "Rate 48000 Hz, ", 15)         = 15
write(2, "Stereo", 6)                   = 6
write(2, "\n", 1)                       = 1
brk(0x6e000)                            = 0x6e000
ioctl(3, SNDRV_TIMER_IOCTL_PARAMS or TIOCSTI, 0xbeaa2638) = 0
read(0, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 4070) = 4070
ioctl(5, SNDRV_PCM_IOCTL_SYNC_PTR, 0x3c420) = 0
ioctl(5, SNDRV_PCM_IOCTL_SYNC_PTR, 0x3c420) = 0
ioctl(5, SNDRV_PCM_IOCTL_SYNC_PTR, 0x3c420) = 0
read(0, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 4096) = 4096
ioctl(5, SNDRV_PCM_IOCTL_SYNC_PTR, 0x3c420) = 0
ioctl(5, SNDRV_PCM_IOCTL_SYNC_PTR, 0x3c420) = 0
ioctl(5, SNDRV_PCM_IOCTL_SYNC_PTR, 0x3c420) = 0
read(0, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 4096) = 4096
ioctl(5, SNDRV_PCM_IOCTL_SYNC_PTR, 0x3c420) = 0
ioctl(5, SNDRV_PCM_IOCTL_SYNC_PTR, 0x3c420) = 0
ioctl(5, SNDRV_PCM_IOCTL_SYNC_PTR, 0x3c420) = 0
read(0, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 4096) = 4096
ioctl(5, SNDRV_PCM_IOCTL_SYNC_PTR, 0x3c420) = 0
ioctl(5, SNDRV_PCM_IOCTL_SYNC_PTR, 0x3c420) = 0
ioctl(5, SNDRV_PCM_IOCTL_SYNC_PTR, 0x3c420) = 0
read(0, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 4096) = 4096
ioctl(5, SNDRV_PCM_IOCTL_SYNC_PTR, 0x3c420) = 0
ioctl(5, SNDRV_PCM_IOCTL_SYNC_PTR, 0x3c420) = 0
ioctl(5, SNDRV_PCM_IOCTL_SYNC_PTR, 0x3c420) = 0
read(0, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 4096) = 4096
ioctl(5, SNDRV_PCM_IOCTL_SYNC_PTR, 0x3c420) = 0
ioctl(5, SNDRV_PCM_IOCTL_SYNC_PTR, 0x3c420) = 0
ioctl(5, SNDRV_PCM_IOCTL_SYNC_PTR, 0x3c420) = 0
[... 348 lines cut ...]
ioctl(5, SNDRV_PCM_IOCTL_SYNC_PTR, 0x3c420) = 0
ioctl(5, SNDRV_PCM_IOCTL_SYNC_PTR, 0x3c420) = 0
ioctl(5, SNDRV_PCM_IOCTL_SYNC_PTR, 0x3c420) = 0
semop(65536, {{0, 0, 0}, {0, 1, SEM_UNDO}}, 2) = 0
semop(65536, {{0, -1, SEM_UNDO|IPC_NOWAIT}}, 1) = 0
read(0, "\371\211\371\211\3737\3737\375\362\375\362\0\206\0\206\1\312\1\312\1\16\1\16\376a\376a\372\214\372\214"..., 4096) = 4096
ioctl(5, SNDRV_PCM_IOCTL_SYNC_PTR, 0x3c420) = 0
ioctl(5, SNDRV_PCM_IOCTL_SYNC_PTR, 0x3c420) = 0
ioctl(5, SNDRV_PCM_IOCTL_SYNC_PTR, 0x3c420) = 0
ioctl(5, SNDRV_PCM_IOCTL_SYNC_PTR, 0x3c420) = 0
ioctl(5, SNDRV_PCM_IOCTL_SYNC_PTR, 0x3c420) = 0
ioctl(5, SNDRV_PCM_IOCTL_SYNC_PTR, 0x3c420) = 0
ioctl(5, SNDRV_PCM_IOCTL_SYNC_PTR, 0x3c420) = 0
ioctl(5, SNDRV_PCM_IOCTL_SYNC_PTR, 0x3c420) = 0
semop(65536, {{0, 0, 0}, {0, 1, SEM_UNDO}}, 2) = 0
semop(65536, {{0, -1, SEM_UNDO|IPC_NOWAIT}}, 1) = 0
read(0, "\371G\371G\373\341\373\341\375\n\375\n\375\26\375\26\374\335\374\335\375F\375F\376\342\376\342\1\235\1\235"..., 4096) = 4096
ioctl(5, SNDRV_PCM_IOCTL_SYNC_PTR, 0x3c420) = 0
ioctl(5, SNDRV_PCM_IOCTL_SYNC_PTR, 0x3c420) = 0
ioctl(5, SNDRV_PCM_IOCTL_SYNC_PTR, 0x3c420) = 0
ioctl(5, SNDRV_PCM_IOCTL_SYNC_PTR, 0x3c420) = 0
ioctl(5, SNDRV_PCM_IOCTL_SYNC_PTR, 0x3c420) = 0
ioctl(5, SNDRV_PCM_IOCTL_SYNC_PTR, 0x3c420) = 0
ioctl(5, SNDRV_PCM_IOCTL_SYNC_PTR, 0x3c420) = 0
ioctl(5, SNDRV_PCM_IOCTL_SYNC_PTR, 0x3c420) = 0

XYEFA, 2016-07-19:
You may want to look at this: http://askubuntu.com/questions/129621/ubuntu-locks-up-when-swap-full

Since df does not show swap space, you can probe it (on a working system) as follows:

swapon -s 

and

foo@red-pitaya:/# free -m
total       used       free     shared    buffers     cached
Mem:           466        182        283          4         21        134
-/+ buffers/cache:         26        439
Swap:            0          0          0

on3vna, 2016-07-20:
Pavel and XYEFA,

I've got a break trough of this problem yesterday, changed code and did some initial testing,but I still want to test the solution under several conditions. Reboot restart, start stop the application monitor in transmit mode.
I also want to start first from a fresh image that will be in the coming days.
Keep you posted.

73
Johan

XYEFA, 2016-07-20:
@on3vna You may also want to read the notes relating to ALSA here: https://www.raspberrypi.org/documentation/configuration/device-tree.md ... I do not know if this is applicable to you.

Regards

on3vna, 2016-07-29:
Pavel , XYEFA

Still had to handle some rare conditions but they are also ok now.
Now to do is testing with transmit and see if it still is working.

73 Johan
ON3VNA

on3vna, 2016-07-29:
Pavel,

Did  testing of the solution seems ok to me.
In stead of using aplay is have implemented the sound in the server, so there is no more need for the aplay.

Extra needed on the red pitaya:

apt-get install libasound2 alsa-utils alsa-oss
apt-get install libasound2-dev

compile the application:
With Lasound:

arm-linux-gnueabihf-gcc -O3 -march=armv7-a -mcpu=cortex-a9 -mtune=cortex-a9 -mfpu=neon -mfloat-abi=hard projects/sdr_transceiver_hpsdr/server/sdr-transceiver-hpsdr.c -o sdr-transceiver-hpsdr -D_GNU_SOURCE -Iprojects/sdr_transceiver_hpsdr/server -lm -lpthread -v -lasound

How do I get my version now in the git? I have never used this before.

Rgds
Johan
73 de ON3VNA

pavel-demin, 2016-07-30:
Hi Johan,

Thanks for finding a solution to this problem.

I think that the easiest way to add your code to the git repository is to go to https://github.com/pavel-demin/red-pitaya-notes/blob/master/projects/sdr_transceiver_hpsdr/server/sdr-transceiver-hpsdr.c and edit the file via the web browser by pressing the 'Edit this file' icon.

Or follow the first two steps from the following tutorial:
https://help.github.com/articles/fork-a-repo/
then edit sdr-transceiver-hpsdr.c on your computer and the run `git commit` and `git push`

Best regards,

Pavel

on3vna, 2016-07-30:
Hey Pavel,

Today noticed some interesting thing: When i was testing yesterday I was using 100Mb connections and did not have any problem with loss of sound, but today i was using 1Gb connections and was confronted with the same sound loss even with the new code.
Just now connected HPSDR PC and RED HPSDR on a 100Mb hub and observe the behaviour.
It will be a hard one the find i guess why on 1Gb link it fails.... as there is not much debug output available.

Rgds
Johan
73 de ON3VNA

pavel-demin, 2016-07-30:
It's a known problem. The main source of the problem is the UDP protocol implementation in the modern operating systems (MS Windows and GNU Linux). Sometimes, there are long delays between the UDP packets sent from PC to Red Pitaya. When it happens, the buffer of audio samples on the Red Pitaya side underflows and you hear a click. It's a known problem and you can find more details at the following links:
http://www.tau.ac.il/~stoledo/Bib/Pubs/qex-jan-feb-2011-toledo.pdf (page 14)
https://groups.yahoo.com/neo/groups/n2adr-sdr/conversations/topics/101
https://groups.google.com/forum/#!topic/sdr-widget/UpLStkqTEoQ

For the I/Q samples the workaround is a quite deep (16k samples) FIFO buffer.

I think that something similar should work for the audio samples.

on3vna, 2016-07-31:
Good afternoon Pavel,

I will have a look into it. See what I can do.
Keep you posted.
Rgds
Johan
73 de ON3VNA

on3vna, 2016-08-02:
Hey Pavel,

From what I observed in the last testing, i strongly start to suspect that the communication between the USB part on the FPGA and the linux kernel gets lost due to the high traffic on the ip interface.
If the ip interface and the USB part on the FPGA are on the same bus then that would explain it.
IP block to busy to handle all the packets on the network.
I have put some extra syslog code in the audio part and there i don't see any anomalies concerning contact loss between the application and the ALSA.

I also found somewhere in Xilinx that the USB should be defined as full speed when using sound.
But i don't know where to check that.

Rgds
Johan
73 de ON3VNA

on3vna, 2016-08-05:
Hey Pavel,

This is what i found this morning when running on a 1gb link:
It seems no more interrupts are handled for usb when the sound is gone

root@red-pitaya:~# cat /proc/interrupts
           CPU0       CPU1
 16:          0          0       GIC  27 Edge      gt
 17:          0          0       GIC  43 Level     ttc_clockevent
 18:       8481      45729       GIC  29 Edge      twd
 21:         43          0       GIC  39 Level     f8007100.adc
141:          2          0       GIC  57 Level     cdns-i2c
143:          0          0       GIC  35 Level     f800c000.ocmc
144:        466          0       GIC  59 Level     xuartps
146:          0          0       GIC  58 Level     e0006000.spi
147:          0          0       GIC  81 Level     e0007000.spi
148:          0          0       GIC  51 Level     e000d000.spi
149:     421534          0       GIC  54 Level     eth0
150:       3172          0       GIC  56 Level     mmc0
151:          0          0       GIC  45 Level     f8003000.dmac
152:          0          0       GIC  46 Level     f8003000.dmac
153:          0          0       GIC  47 Level     f8003000.dmac
154:          0          0       GIC  48 Level     f8003000.dmac
155:          0          0       GIC  49 Level     f8003000.dmac
156:          0          0       GIC  72 Level     f8003000.dmac
157:          0          0       GIC  73 Level     f8003000.dmac
158:          0          0       GIC  74 Level     f8003000.dmac
159:          0          0       GIC  75 Level     f8003000.dmac
160:          1          0       GIC  40 Level     f8007000.devcfg
166:      16694          0       GIC  53 Level     e0002000.usb
167:          0          0       GIC  41 Edge      f8005000.watchdog
IPI1:          0          0  Timer broadcast interrupts
IPI2:       3014       6030  Rescheduling interrupts
IPI3:          0          0  Function call interrupts
IPI4:         54         22  Single function call interrupts
IPI5:          0          0  CPU stop interrupts
IPI6:          1          0  IRQ work interrupts
IPI7:          0          0  completion interrupts
Err:          0
root@red-pitaya:~# cat /proc/interrupts
           CPU0       CPU1
 16:          0          0       GIC  27 Edge      gt
 17:          0          0       GIC  43 Level     ttc_clockevent
 18:       9006      51003       GIC  29 Edge      twd
 21:         43          0       GIC  39 Level     f8007100.adc
141:          2          0       GIC  57 Level     cdns-i2c
143:          0          0       GIC  35 Level     f800c000.ocmc
144:        502          0       GIC  59 Level     xuartps
146:          0          0       GIC  58 Level     e0006000.spi
147:          0          0       GIC  81 Level     e0007000.spi
148:          0          0       GIC  51 Level     e000d000.spi
149:     474552          0       GIC  54 Level     eth0
150:       3172          0       GIC  56 Level     mmc0
151:          0          0       GIC  45 Level     f8003000.dmac
152:          0          0       GIC  46 Level     f8003000.dmac
153:          0          0       GIC  47 Level     f8003000.dmac
154:          0          0       GIC  48 Level     f8003000.dmac
155:          0          0       GIC  49 Level     f8003000.dmac
156:          0          0       GIC  72 Level     f8003000.dmac
157:          0          0       GIC  73 Level     f8003000.dmac
158:          0          0       GIC  74 Level     f8003000.dmac
159:          0          0       GIC  75 Level     f8003000.dmac
160:          1          0       GIC  40 Level     f8007000.devcfg
166:      16694          0       GIC  53 Level     e0002000.usb
167:          0          0       GIC  41 Edge      f8005000.watchdog
IPI1:          0          0  Timer broadcast interrupts
IPI2:       3021       6189  Rescheduling interrupts
IPI3:          0          0  Function call interrupts
IPI4:         54         22  Single function call interrupts
IPI5:          0          0  CPU stop interrupts
[... 13 lines cut ...]
147:          0          0       GIC  81 Level     e0007000.spi
148:          0          0       GIC  51 Level     e000d000.spi
149:     540517          0       GIC  54 Level     eth0
150:       3172          0       GIC  56 Level     mmc0
151:          0          0       GIC  45 Level     f8003000.dmac
152:          0          0       GIC  46 Level     f8003000.dmac
153:          0          0       GIC  47 Level     f8003000.dmac
154:          0          0       GIC  48 Level     f8003000.dmac
155:          0          0       GIC  49 Level     f8003000.dmac
156:          0          0       GIC  72 Level     f8003000.dmac
157:          0          0       GIC  73 Level     f8003000.dmac
158:          0          0       GIC  74 Level     f8003000.dmac
159:          0          0       GIC  75 Level     f8003000.dmac
160:          1          0       GIC  40 Level     f8007000.devcfg
166:      16694          0       GIC  53 Level     e0002000.usb
167:          0          0       GIC  41 Edge      f8005000.watchdog
IPI1:          0          0  Timer broadcast interrupts
IPI2:       3035       6445  Rescheduling interrupts
IPI3:          0          0  Function call interrupts
IPI4:         54         22  Single function call interrupts
IPI5:          0          0  CPU stop interrupts
IPI6:          1          0  IRQ work interrupts
IPI7:          0          0  completion interrupts
Err:          0
root@red-pitaya:~#

on3vna, 2016-08-05:
Pavel,
I also implemented in the code the use of syslog.
So i have added in /etc/rsyslog.conf the following line:

local0.*                         -/var/log/hpsdr.log

Also uploaded the file with the changes.

Rgds
Johan
73 de ON3VNA

on3vna, 2016-08-09:
Pavel,

Can I send you a PM concerning this ?

Rgds
Johan

pavel-demin, 2016-08-09:
Yes, sure. The address is in my [profile](https://github.com/pavel-demin).

on3vna, 2016-08-15:
Pavel,

New code is available. This with the interrupt routing of the USB to the second core should do the trick.
Now working on a 1Gb link between HPSDR and the RED PITAYA.

Also the start.sh (stop.sh) needs to be adopted, removing the alsa stuff.
Also the rc.local for routing the int to the second core.

Rgds
Johan

pavel-demin, 2016-08-15:
Thanks for the good news!

Have you checked if the interrupt routing also fixes the version with aplay? If possible, I'd prefer to keep the version with aplay because it works with the small customized SD card image.

on3vna, 2016-08-15:
Hey Pavel,

No I did not check the int routing with the aplay.
I just coded everything from scratch forgetting about the aplay.
I dont see any problem to run the code on the small SD card image.  
I still use the SD image you made with only the new code so should not be a problem .

73
Johan

on3vna, 2016-08-15:
This is the size of the new file 
-rwxr-xr-x  1 root root 21584 Aug 15 15:33 sdr-transceiver-hpsdr

pavel-demin, 2016-08-15:
> I dont see any problem to run the code on the small SD card image.

The problem is that the alsa/asound library isn't installed on the SD cards provided by Red Pitaya.

Another problem is that I'll have to cross-compile the alsa/asound library on my development machine. I'd prefer not to do it.

ka6s, 2016-08-15:
Pavel,

I saw somewhere that the asound was only unidirectional, i.e. can only be
used to play the audio out. Is that correct?

Steve KA6S

pavel-demin, 2016-08-15:
> I saw somewhere that the asound was only unidirectional, i.e. can only be used to play the audio out. Is that correct?

In my understanding, libasound is the name of the Debian package for the ALSA library:
https://packages.debian.org/jessie/libasound2

The ALSA library provides both snd_pcm_writei and snd_pcm_readi functions:
http://www.alsa-project.org/alsa-doc/alsa-lib/group___p_c_m.html

So, both playback and recording are available.

ka6s, 2016-08-18:
Pavel - I saw your email on this in another thread - and I've obtained your
development platform version and loaded the alsa libs on it.  I now have
sound with an $8 USB Sound dongle!

I've tried the current HPSDR link - and it doesn't seem to have the sound
system installed.  So to turn the development version into a booting sdr
transceiver - what files do I need to install where?

Thanks!

Steve KA6S

pavel-demin, 2016-08-18:
>  So to turn the development version into a booting sdr transceiver - what files do I need to install where?

I think that adding the following lines to /etc/rc.local should start the SDR application just after the boot:
```
cd /opt/redpitaya/www/apps/sdr_transceiver_hpsdr
cat sdr_transceiver_hpsdr.bit > /dev/xdevcfg
source start.sh
```

pavel-demin, 2016-10-02:
I'm closing this issue because SDR transceiver compatible with HPSDR now supports audio playback/capture and CW sidetone generation via the WM8731 and TLV320AIC23B I2S audio codecs.

If there are problems with the USB audio devices, then I'd suggest switching to the I2S audio codecs.

closed 2016-10-02 by pavel-demin