pavel-demin/red-pitaya-notes #312: WSPR image copy to SD card via win32 disk imager

DK8FT, 2016-07-12:
when doing so only 4 files appear on sd card

boot.bin    1,59MB
devicetree.dtb   11,8kB
uEnv.txt   319 bytes
uImage    4,67 MB

total 4gb  sd card is said to be only 11,5MB big

how to proceed, please ?

thanks and regards

Werner  DK8FT

pavel-demin, 2016-07-12:
Hello Werner,

The SD card contains two partitions. Only the first one (with 4 files) is visible under Windows.

I suggest the following next steps:
 - boot Red Pitaya with the WSPR SD card
 - connect to Red Pitaya using [WinSCP](http://winscp.net)
 - open decode-wspr.sh in an editor like [Notepad++](http://notepad-plus-plus.org/)
 - set the CALL and GRID variables
 - close the editor and WinSCP
 - check for reported WSPR spots via [WSPRnet](http://wsprnet.org)

Best regards,

Pavel

DK8FT, 2016-07-12:
Hello Pavel,

thank you for your quick answer.

It is working now !  You wrote an exciting software !

I've made a donation a couple of minutes ago.

Best regards

Werner

pavel-demin, 2016-07-13:
Many thanks for your feedback and for your donation! I'm glad that the WSPR receiver is working for you.

closed 2016-07-13 by pavel-demin