pavel-demin/red-pitaya-notes #314: PTT Line for Multi-Channel WSPR transceiver

ptamike, 2016-07-17:
Hi Pavel,

Could you add a PTT line using one of the Red Pitaya GPIOs to trigger a transmit/receive antenna switch?

Thanks,

Mike - G4WNC

XYEFA, 2016-07-19:
A Push To Talk to a GPIO would be straightforward if the gpio lines were accessible from the kernel layer so they could access the usual gpio drivers (I think the stock Red Pitaya has them going to LEDs only). 

However, I think it would be simple if you use the Red Pitaya API (user layer code) and read GPIO that way. 

Regards!

pavel-demin, 2016-07-19:
Hi Mike,

The GPIO pins were enabled in the FPGA configuration with the commit https://github.com/pavel-demin/red-pitaya-notes/commit/8ce36c84808403de661bc0085b0f393dee3d3ee2.

What is missing is some C code to control one of the GPIO pins. I'll try to add it in the coming days.

Best regards,

Pavel

XYEFA, 2016-07-19:
Hello Pavel - Will this code be at the kernel level or will be a user level API? Also from what I infer from your comments, the GPIOs are not Zynq MIO GPIOs but GPIOs synthesized from the FPGA. Is this correct?

pavel-demin, 2016-07-19:
It will be the user level code. Very similar to the PTT control in other SDR projects:
https://github.com/pavel-demin/red-pitaya-notes/blob/master/projects/sdr_transceiver/server/sdr-transceiver.c#L309

pavel-demin, 2016-09-21:
I've just added a small program called gpio-output.c with [this commit](https://github.com/pavel-demin/red-pitaya-notes/commit/b0e13be7980fa55929c158489a416880e13c75ec).

This program controls the DIO*_P pins of the [extension connector E1](http://wiki.redpitaya.com/index.php?title=Extension_connectors#Extension_connector_E1).

I've also added this program to the decode-wspr.sh and transmit-wspr.sh scripts. The pin DIO0_P is switched to high one second before the transmission and it's switched to low one second before the reception.

The program and all the modified files can be downloaded to Red Pitaya with the following commands:
```
wget https://raw.githubusercontent.com/pavel-demin/red-pitaya-notes/master/projects/sdr_transceiver_wspr/Makefile
wget https://raw.githubusercontent.com/pavel-demin/red-pitaya-notes/master/projects/sdr_transceiver_wspr/gpio-output.c
wget https://raw.githubusercontent.com/pavel-demin/red-pitaya-notes/master/projects/sdr_transceiver_wspr/decode-wspr.sh
wget https://raw.githubusercontent.com/pavel-demin/red-pitaya-notes/master/projects/sdr_transceiver_wspr/transmit-wspr.sh
```
The program can be compiled with the following command:
```
make
```

ptamike, 2016-09-22:
Hi Pavel,

Thanks for adding the PTT line. However, I think there may be an error in the link you sent to download the Makefile. I've tried it a couple of times and it returns the old Makefile. I can get round it here but it might confuse others.

Thanks again,

Mike

pavel-demin, 2016-09-22:
Hi Mike,

Thanks for testing this new code.

I think that wget saves new Makefile as Makefile.1. It's probably good that it doesn't overwrite the old file. At least, it could be useful for decode-wspr.sh because it could contain some local modifications. But I agree that in case of Makefile it might confuse.

I've added -O option to all the wget commands and -ptt suffix to the new versions of the shell scripts but I'm afraid that it's still confusing. 

Best regards,

Pavel

pavel-demin, 2016-12-23:
I've finally rebuilt the WSPR SD card image. The GPIO control program (gpio-output) is installed on this new WSPR SD card image. Closing this issue.

closed 2016-12-23 by pavel-demin

ptamike, 2016-12-23:
Hi Pavel,

Thanks again for all your support with the Red Pitaya projects – it’s very much appreciated.

Mike – G4WNC