pavel-demin/red-pitaya-notes #315: Where does one install overlays on the Red Pitaya?

XYEFA, 2016-07-19:
Hello again - I should have asked this question in my earlier thread. Where are Device Tree overlays installed on the Red Pitaya? I have a Device Tree overlay but I do not know the procedure of where to copy the compiled blob in the Red Pitaya. Do you know the Overlay install directory?

On the Raspberry Pi the overlay is compiled and copied to /lib/firmware or something like that. What is the process in Zynq based boards like the Red Pitaya? Is the method  the same?

Thanks!

<<

pavel-demin, 2016-07-19:
Unfortunately, I can't answer this question.

XYEFA, 2016-07-19:
I will probe and try to figure it out. It seems different platforms have various ways, usually copying the file to /lib/firmware

closed 2016-09-24 by pavel-demin