pavel-demin/red-pitaya-notes #317: HPSDR .ZIP file isn't an .IMG file?

ka6s, 2016-08-02:
Pavel,

I have a single confounding issue.  The instructions say to install the image in the HPSDR ZIP file on the web site.  However, when I unzip it I don't find the ecosystem .img file, but rather a zipped file system?  I understand taking the .img file and using dd to install it on a card. I don't understand taking the zipped file system and installing it. What am I missing here??

Thanks,

Steve KA6S

pavel-demin, 2016-08-03:
It's enough to copy the files from the .zip file to a FAT32 formatted SD card.

ka6s, 2016-08-03:
Pavel,

I found that there was cock-pit error on my part.  Things don't transmit very well when you only download the receiver!   Using the transceiver download works MUCH better!

Thanks!  I'm really enjoying the fruit of the work you have put in.  I've been working on a companion board to build up a communications grade transceiver with your code at the heart of it.  It has been a great deal of fun!

Steve KA6S

pavel-demin, 2016-08-04:
Thanks for your feedback. I'm glad that you've solved the problem and the transceiver works for you.

closed 2016-08-04 by pavel-demin