pavel-demin/red-pitaya-notes #320: DMA

vilmagsumov, 2016-08-07:
Hello Pavel,

in my work I need to write adc samples directly to DDR. As an example I use your adc-test project. The first few records are successfull, but after that the internal counter in RAM Writer stops working. Each record ends with RESET MODE.  Do you have any suggestions why it happens?

Thanks!

pavel-demin, 2016-08-07:
I'm not sure but could it be that you reset or reconfigure FPGA before the end of the writing burst? The AXI slave interface (S_AXI_HP0) hangs if a transaction is not correctly terminated.

I don't have Red Pitaya right now to test it. I'll try to do it next week.

The [MCPHA project](https://github.com/pavel-demin/red-pitaya-notes/tree/master/projects/mcpha) also writes samples to DDR. I tested it recently and didn't notice any problem.

The [DAC player project](https://github.com/pavel-demin/red-pitaya-notes/tree/master/projects/dac_player) also worked some months ago.

vilmagsumov, 2016-08-07:
Thank you for your quick response. I'll try to use MCPHA project or DAC player.

vilmagsumov, 2016-08-12:
Pavel,
I saw your projects, but they are not suitable for solving my problem.
I try to record ADC samples without any decimation directly to DDR. Could you tell me how to terminate transaction correctly?
My problem is as follows: to record 8MB samples to DDR, read them, record samples again, read and so on.
P.S. What is the packetizer in your IP core ?
Thank you!

pavel-demin, 2016-08-12:
> record ADC samples without any decimation directly to DDR
> record 8MB samples to DDR, read them, record samples again, read and so on

Thanks for the details about your requirements. The adc_recorder project almost matches your requirements:
https://github.com/pavel-demin/red-pitaya-notes/tree/master/projects/adc_recorder

I even have SD card images and FPGA bitstreams:

https://googledrive.com/host/0B-t5klOOymMNfmJ0bFQzTVNXQ3RtWm5SQ2NGTE1hRUlTd3V2emdSNzN6d0pYamNILW83Wmc/adc_recorder

Here is how it can be used:

1. download the customized SD card image zip file from
https://googledrive.com/host/0B-t5klOOymMNfmJ0bFQzTVNXQ3RtWm5SQ2NGTE1hRUlTd3V2emdSNzN6d0pYamNILW83Wmc/adc_recorder/ecosystem-0.92-65-35575ed-adc-recorder.zip

2. copy the content of the SD card image zip file to an SD card

3. boot Red Pitaya with the new SD card

4. run the following commands from the Red Pitaya command line:
```
# make /opt writable
rw

# record 1M ADC samples
adc-recorder > /opt/adc-samples.txt

# make /opt read-only
ro
```
The only problem is that adc_recorder decimates the samples by 64.

To solve this problem, you'll need to remove bcast_0, cic_0, cic_1, comb_0, fir_0 and connect fifo_0 directly to pktzr_0.

> What is the packetizer in your IP core.

The packetizer IP core is used to control the number of samples written to DDR.

vilmagsumov, 2016-08-17:
Thank you for your answer! After reading the source code of packetizer I found my mistake (in packetizer I set true to continous mode).
Now the ADC recorder works.
Thank you!

pavel-demin, 2016-08-17:
Thanks for your feedback!

closed 2016-08-17 by pavel-demin

falcon98, 2018-07-13:
Hi Vilmagsumov,

could you pls inform me how you were able to acquire 8M samples. What changes are needed in C program.
When I change the no of samples to 15*1024*1024, I get segmentation error. 

Thanks a lot

glnmurthy, 2022-08-09:
> Thanks for the details about your requirements. The adc_recorder project almost matches your requirements: https://github.com/pavel-demin/red-pitaya-notes/tree/master/projects/adc_recorder
>
> I even [...]

Unable to download