pavel-demin/red-pitaya-notes #322: Led Linker: Could not install required packages

tilman1, 2016-08-18:
Dear Pavel
I tried to compile the LED blinker example.  I have followed your procedure to setup a development VM. That has worked. Then I went on with the procedure of the led blinker example to install the required packages. When doing "sudo apt-get update", the armhf packages cannot be found. They seem not to be on the server any longer. Can you kindely advise? Below  the error messages.

Many Thanks 
Tilman

rr http://ftp.heanet.ie trusty-updates/main armhf Packages                    
  404  Not Found [IP: 193.1.193.64 80]
Err http://ftp.heanet.ie trusty-updates/restricted armhf Packages              
  404  Not Found [IP: 193.1.193.64 80]
Err http://ftp.heanet.ie trusty-updates/universe armhf Packages                
  404  Not Found [IP: 193.1.193.64 80]
Err http://ftp.heanet.ie trusty-updates/multiverse armhf Packages              
  404  Not Found [IP: 193.1.193.64 80]
Err http://ftp.heanet.ie trusty-backports/main armhf Packages                  
  404  Not Found [IP: 193.1.193.64 80]
Err http://ftp.heanet.ie trusty-backports/restricted armhf Packages            
  404  Not Found [IP: 193.1.193.64 80]
Err http://ftp.heanet.ie trusty-backports/universe armhf Packages              
  404  Not Found [IP: 193.1.193.64 80]
Err http://ftp.heanet.ie trusty-backports/multiverse armhf Packages            
  404  Not Found [IP: 193.1.193.64 80]
Err http://ftp.heanet.ie trusty/main armhf Packages                            
  404  Not Found [IP: 193.1.193.64 80]
Err http://ftp.heanet.ie trusty/restricted armhf Packages                      
  404  Not Found [IP: 193.1.193.64 80]
Err http://ftp.heanet.ie trusty/universe armhf Packages                        
  404  Not Found [IP: 193.1.193.64 80]
Err http://ftp.heanet.ie trusty/multiverse armhf Packages                      
  404  Not Found [IP: 193.1.193.64 80]
W: Failed to fetch http://security.ubuntu.com/ubuntu/dists/trusty-security/main/binary-armhf/Packages  404  Not Found [IP: 91.189.91.26 80]

W: Failed to fetch http://security.ubuntu.com/ubuntu/dists/trusty-security/restricted/binary-armhf/Packages  404  Not Found [IP: 91.189.91.26 80]

W: Failed to fetch http://security.ubuntu.com/ubuntu/dists/trusty-security/universe/binary-armhf/Packages  404  Not Found [IP: 91.189.91.26 80]

W: Failed to fetch http://security.ubuntu.com/ubuntu/dists/trusty-security/multiverse/binary-armhf/Packages  404  Not Found [IP: 91.189.91.26 80]

W: Failed to fetch http://ftp.heanet.ie/pub/ubuntu/dists/trusty-updates/main/binary-armhf/Packages  404  Not Found [IP: 193.1.193.64 80]

W: Failed to fetch http://ftp.heanet.ie/pub/ubuntu/dists/trusty-updates/restricted/binary-armhf/Packages  404  Not Found [IP: 193.1.193.64 80]

W: Failed to fetch http://ftp.heanet.ie/pub/ubuntu/dists/trusty-updates/universe/binary-armhf/Packages  404  Not Found [IP: 193.1.193.64 80]

W: Failed to fetch http://ftp.heanet.ie/pub/ubuntu/dists/trusty-updates/multiverse/binary-armhf/Packages  404  Not Found [IP: 193.1.193.64 80]

W: Failed to fetch http://ftp.heanet.ie/pub/ubuntu/dists/trusty-backports/main/binary-armhf/Packages  404  Not Found [IP: 193.1.193.64 80]

W: Failed to fetch http://ftp.heanet.ie/pub/ubuntu/dists/trusty-backports/restricted/binary-armhf/Packages  404  Not Found [IP: 193.1.193.64 80]

W: Failed to fetch http://ftp.heanet.ie/pub/ubuntu/dists/trusty-backports/universe/binary-armhf/Packages  404  Not Found [IP: 193.1.193.64 80]

W: Failed to fetch http://ftp.heanet.ie/pub/ubuntu/dists/trusty-backports/multiverse/binary-armhf/Packages  404  Not Found [IP: 193.1.193.64 80]

W: Failed to fetch http://ftp.heanet.ie/pub/ubuntu/dists/trusty/main/binary-armhf/Packages  404  Not Found [IP: 193.1.193.64 80]

W: Failed to fetch http://ftp.heanet.ie/pub/ubuntu/dists/trusty/restricted/binary-armhf/Packages  404  Not Found [IP: 193.1.193.64 80]

W: Failed to fetch http://ftp.heanet.ie/pub/ubuntu/dists/trusty/universe/binary-armhf/Packages  404  Not Found [IP: 193.1.193.64 80]

W: Failed to fetch http://ftp.heanet.ie/pub/ubuntu/dists/trusty/multiverse/binary-armhf/Packages  404  Not Found [IP: 193.1.193.64 80]

E: Some index files failed to download. They have been ignored, or old ones used instead.

pavel-demin, 2016-08-18:
Dear Tilman,

Thanks for finding this inconsistency in my notes.

The development machine notes explain how to install both Debian and Ubuntu. However, the required packages part of the led blinker notes only works with the Debian development machine. Actually, these commands are not needed if the development machine is installed using my automatic installation scripts. The automatic installation scripts install all required packages.

I'll remove the required packages part of the led blinker notes.

It think that the following steps should work with your installation:

- remove armhf architecture
```
dpkg --remove-architecture armhf
```  
- remove /etc/apt/sources.list.d/emdebian.list
```
rm /etc/apt/sources.list.d/emdebian.list
```
- skip the required packages part of the led blinker notes

Best regards,

Pavel

tilman1, 2016-08-18:
Dear Pavel

thank you for your quick  help.

I have executed the steps you suggested to fix the installation. I then skipped the required package part and the syntatic sugar for IP cores part, and continued with "Getting Started". The make command did not finish successfully -- see below. As the name suggest, I would assume that the example makes an LED blinking. Is it possible to execute the example without rebuilding the image on the sd card, i.e. transfer an executable via scp to the red pitaya and execute it there?

Many thanks
Tilman

make NAME=led_blinker all
mkdir -p tmp/cores
vivado -nolog -nojournal -mode batch -source scripts/core.tcl -tclargs axi_axis_reader_v1_0 xc7z010clg400-1
terminate called after throwing an instance of 'std::runtime_error'
  what():  locale::facet::_S_create_c_locale name not valid
/opt/Xilinx/Vivado/2016.2/bin/loader: line 164: 10333 Aborted                 "$RDI_PROG" "$@"
make: *** [tmp/cores/axi_axis_reader_v1_0] Error 134

pavel-demin, 2016-08-18:
Looks like a locale problem. Maybe you're connecting via SSH from a machine with a locale that is not installed on the development machine. Could you please check what the `locale` command returns on the development machine?

> Is it possible to execute the example without rebuilding the image on the sd card...

It's possible to build the FPGA configuration bitstream with the following command:
```
make NAME=led_blinker tmp/led_blinker.bit
```

Then the command to configure FPGA is
```
cat led_blinker.bit > /dev/xdevcfg
```

tilman1, 2016-08-18:
Hello Pavel

yes, I am connecting via ssh to the VM (as I did not manage to change 
the resolution of the VM display which results in buttons at the buttom 
of GUIs not being visible.).

Local compilation on the DEV VM works indeed.

root@ubuntu:/home/red-pitaya/red-pitaya-notes# locale
locale: Cannot set LC_ALL to default locale: No such file or directory
LANG=en_US.UTF-8
LANGUAGE=en_US:
LC_CTYPE="en_US.UTF-8"
LC_NUMERIC=de_DE.UTF-8
LC_TIME=de_DE.UTF-8
LC_COLLATE="en_US.UTF-8"
LC_MONETARY=de_DE.UTF-8
LC_MESSAGES="en_US.UTF-8"
LC_PAPER=de_DE.UTF-8
LC_NAME=de_DE.UTF-8
LC_ADDRESS=de_DE.UTF-8
LC_TELEPHONE=de_DE.UTF-8
LC_MEASUREMENT=de_DE.UTF-8
LC_IDENTIFICATION=de_DE.UTF-8
LC_ALL=

Many thanks

Tilman

pavel-demin, 2016-08-19:
Could you please try the following commands on the development machine and then relogin?
```
export LC_ALL="en_US.UTF-8"
sudo dpkg-reconfigure locales
```

tilman1, 2016-08-20:
That resolves the issue. It is however not permanent -- if I log out and back in, I need  to reexecute the export command.

pavel-demin, 2016-08-20:
Thanks for confirming that it almost works. So, the problem is indeed with the locale settings.

Looks like your main computer is using de_DE.UTF-8 and development machine is using en_US.UTF-8.  The SSH client on your main computer is configured to forward the locale settings. It forwards de_DE.UTF-8 to the development machine but de_DE.UTF-8 is not configured on the development machine.

If I'm not mistaken, the following commands should enable de_DE.UTF-8 on the development machine:
```
locale-gen de_DE.UTF-8
update-locale LANG=de_DE.UTF-8
```
Alternatively you can change the configuration of the SSH server on the development machine, by commenting out the following line in /etc/ssh/sshd_config on the development machine:
```
AcceptEnv LANG LC_*
```

tilman1, 2016-08-20:
Thanks for the advices -- that should do...

closed 2016-08-20 by tilman1