pavel-demin/red-pitaya-notes #326: I2C interface use

dcmk1mr2, 2016-08-22:
I have tried following DC2PD's instructions but I am loosing something in translation.  

I have replicated all of the PowerSDRsettings shown for General/Penny Out, General/Alex/Antenna and General/ALEX/HPF/LPF pages.  I don't seem to be getting anything out on the I2C GPIO pins on connector E2.  I don't understand jumper K2 and K4 discussed on page 4.  Can anyone explain where they are?  I have the ALEX signals on connector E1 and the latest software.

Thanks as always to Pavel and also to DC2PD for simplifying the interface.

73,

 Martin W6MRR

on3vna, 2016-08-22:
Martin,

I have noticed the same during the weekend, I assume we are missing the 4k7 resistors on the I2C lines.
They should be connected to the 3V. I did not have more time to investigate.

GL
73
Johan ON3VNA

pavel-demin, 2016-08-22:
Out of curiosity, why do you prefer the I2C interface and not the ALEX interface?

>  I don't seem to be getting anything out on the I2C GPIO pins on connector E2.

I've never tested the I2C code. I'll try to do it later today.

Normally, there should be some activity on the I2C bus when the program starts on Red Pitaya and initializes the PCA9555 chip:
https://github.com/pavel-demin/red-pitaya-notes/blob/master/projects/sdr_transceiver_hpsdr/server/sdr-transceiver-hpsdr.c#L165

>  I don't understand jumper K2 and K4 discussed on page 4. 

The schematics of these jumpers is shown on page 2. They are used to set the I2C address of the PCA9555 chip. This way, multiple PCA9555 chips can be connected to the same bus.

>  I assume we are missing the 4k7 resistors on the I2C lines.

The resistors should be already present on the Red Pitaya board. The on-board EEPROM chip is connected to the same I2C bus.

pavel-demin, 2016-08-22:
One more detail. If during the initialization the program doesn't find any PCA9555 chip with address 0x20 or 0x21, then it doesn't send anything to the I2C bus:

https://github.com/pavel-demin/red-pitaya-notes/blob/master/projects/sdr_transceiver_hpsdr/server/sdr-transceiver-hpsdr.c#L391

https://github.com/pavel-demin/red-pitaya-notes/blob/master/projects/sdr_transceiver_hpsdr/server/sdr-transceiver-hpsdr.c#L463

ka6s, 2016-08-22:
> transceiver_hpsdr/server/sdr-transceiver-hpsdr.c#L391
> transceiver_hpsdr/server/sdr-transceiver-hpsdr.c#L463

I've gotten the I2C  interface to work.  Took me a bit of time to find this
- the BIG secret was just turning on the interfaces in HPSDR that I
missed!  You got to the Configuration option to control what the outputs
are on the OC drivers for Transmit/receive and you MUST highlight the
buttons on the bottom of the form for it to send the information.  Maybe
that was obvious to others, but it was something I missed initially.

Hope that helps.

Steve KA6S

dcmk1mr2, 2016-08-22:
I'm using different filters than ALEX or SC2PD's so I need different pins to be activated.   I am planning on using a microcontroller to map the bits as needed.  That's why I prefer the I2C solution to the ALEX interface - it's simpler to decode two wires over four.

I see that the initialization code interrogates the PCA9555 chip.  I figure I can provide a respose with a microcontroller as well but I'm not getting that far.  I have a logic analyzer on the I2C pins and have never seen activity on HPSDR power up or band changes so I think it must be the HPSDR software settings. 

@Steve - could you provide a little more information on the Configuration option that you found to be important.  I've tried the settings show on SC2PD's screen shots to no avail. 

Thanks to all!

73 Martin W6MRR

ka6s, 2016-08-23:
Martin - please note the two highlighted buttons on the bottom of the form
{EXT CONTROL and HERCULES AMP} If you don't hit those - the function isn't
enabled in the software and it doesn't send the control codes.  My I2C
interface was done using a PSoC - and it powers up running the I2C code
that translates over to the filters I'm using, i.e. same problem that you
had.

[image: Inline image 1]

on3vna, 2016-08-23:
Good to know this . Thanks Steve for the info.That is again a very strange combination for activating the I2C. I was hoping to get the I2C out also with only ALEX enabled. hehehe. I was checking it with the logic analyzer and did see any movement on the I2C bus, but the bits where send out on the serial outputs on the ALEX interface so i did not spent more time on the I2C bus.
Anyhow it is something to write down in the user manual of the RED pitaya HPSDR. HI

73 de ON3VNA
Johan

ka6s, 2016-08-23:
I'll point out that the other option that I did originally was to modify
the server code and figure out band changes by itself - defeats the control
of TX/RX but I didn't really care about split operation. That was my
original solution until I found out about pushing those two buttons at the
bottom.

Steve

on3vna, 2016-08-23:
@ka6s 
If you only activate the external control, then you can manual select what should be turned on for TX/RX
right ?
Is the  information then send out on I2C for the ALEX ?
There are so many features buttons and boxes and bells in the HPSDR...
It seems that we have to do code reading of the HPSDR software before you want to use some feature.

Well not complaining about it, that the whole fun of HPSDR and open source. I love it in combination with the red pitaya and it keeps the brains fit. HI

Johan

ka6s, 2016-08-23:
Oh - I hit my head against the wall for over a week looking at every aspect
of how two wires were connected ;-)

1) I2C IS working from the RP hardware - tested by writing a test routine
replacing the SDR server.
2) Writing code on my interface using 1 I2C to talk to the other to prove
that the SoC was SEEING the traffic.
3) Beeping the wires out (did that first..)
4) Rewriting my I2C code a few dozen times on my side.  Turns out that was
wrong TOO.
5) Staring at the hpsdr Server code? It WAS there - and aligned with what
my test routine did...
6) Finally - followed the link to article in German.. and REALLY looked at
the pictures!

Overall - spent about 1 1/2 weeks in I2C Hell - I have quite a bit fewer
hairs on my head.

Steve

dcmk1mr2, 2016-08-28:
I got together with Steve KA6S today and tried a bunch of things to no avail.  We did rule out PowerSDR settings, Red Pitaya GPIO pins and logic analyzer.  It it possible DC2PD's code didn't make it into the current image available for download?

Thanks again to all!

Martin W6MRR

pavel-demin, 2016-08-28:
Do you see any activity on the I2C bus when the SDR application starts?

I've just connected a scope to the SDA and SCL pins and I've seen the address bits.

The following commands can be used to detect all devices connected to the I2C bus:
```
apt-get install i2c-tools
i2cdetect -y -r 0
```

> Is it possible DC2PD's code didn't make it into the current image available for download?
No.

dcmk1mr2, 2016-08-28:
Sorry - stupid newbie question:
How do I install apt-get?
Thanks!

pavel-demin, 2016-08-28:
> How do I install apt-get?

It's already installed on the SD card images available from the following links:
http://pavel-demin.github.io/red-pitaya-notes/debian-ecosystem/
http://redpitaya.com/quick-start/

dcmk1mr2, 2016-08-30:
I used the debian image and i2cdetect get activity on the I2C bus so the GPIO pins and analyzer are OK.

on3vna, 2016-08-31:
I have been using the following :
http://forum.redpitaya.com/viewtopic.php?f=9&t=554
And indeed the I2C bus is alive and kicking.
My newly added device is detected.

73 de ON3VNA
Johan

pavel-demin, 2016-08-31:
> I used the debian image and i2cdetect get activity on the I2C bus so the GPIO pins and analyzer are OK.

Good. Let's try to continue with the Debian image. As a next step, I'd suggest to install and run the SDR application via the Red Pitaya web interface. When the SDR application start, I'll most probably see some activity on the I2C bus.

dcmk1mr2, 2016-08-31:
Everything is working with the new image!

Thanks for everyone's help and patience.

73 de W6MRR
Martin

closed 2016-08-31 by dcmk1mr2