pavel-demin/red-pitaya-notes #346: Cannot install MCPHA client packages on Ubuntu

benbaltes, 2016-09-12:
I'm trying to install the client code for MCPHA on Ubuntu ( http://pavel-demin.github.io/red-pitaya-notes/mcpha/ ) 

In the instructions it says to fist install the required libraries:
sudo apt-get install tcl8.6-dev tk8.6-dev blt-dev tk8.6-blt2.5 tcllib

When I do this, i get an error saying that it can't find the tk8.6-blt2.5 libraries, so i manually installed those libraries from here:
https://packages.debian.org/jessie/amd64/tk8.6-blt2.5/download

I then try again to install, and i get an error saying that blt-dev depends on the blt package ( I guess tk8.6-blt2.5 isn't good enough).

I then tried to install blt using 'sudo apt-get install blt' and there is a conflict where 'tk8.6-blt2.5' and 'blt' break eachother.

So when I uninstall tk8.6-blt2.5 and install blt, i can finally move on to the next step and compile my control program...
gcc -shared -fPIC -I /usr/include/tcl8.6 mcpha.c -o mcpha.so -lBLT

This finally compiles, however it crashes with segfaults and seems to be missing some functionality as i can't get a histogram.  I'm guessing its because i'm not using the correct tk8.6-blt2.5 package?

How do i fix this when tk8.6-blt2.5 and blt conflict?

pavel-demin, 2016-09-12:
What Ubuntu version are you using?

I only tested the MCPHA client with Debian 8 and MS Windows 7.

According to [packages.ubuntu.com](http://packages.ubuntu.com/search?keywords=tk8.6-blt2.5) is available in Ubuntu 15.10 and 16.04.

So, the easiest and known to work solution is to install Debian 8.

It's not impossible that the MCPHA client will work with Ubuntu 16.04.

On other systems, I'd suggest to build both Tcl/Tk 8.6 and BLT 2.5 from sources. It can be done with [kitgen](https://github.com/pavel-demin/red-pitaya-notes/tree/master/kitgen).

benbaltes, 2016-09-12:
I'm using Ubuntu 14.04.4 LTS

pavel-demin, 2016-09-12:
Just tried it on 14.04.

The required packages can be installed with the following command:
{{{
sudo apt-get install tcl8.6-dev tk8.6-dev blt-dev blt tcllib
}}}

The problem is that BLT from 14.04 doesn't work. I can start the MCPHA client but when I try to zoom, it crashes. A very similar problem is reported at the following link:
https://bugs.launchpad.net/ubuntu/+source/blt/+bug/359857

benbaltes, 2016-09-12:
Thanks for the quick responses Pavel - i really appreciate it.

Was the command supposed to get the app working in a state where zoom works or so that it just launches?  It seems to still be crashing on me when i zoom.

Should i expect the histogram to still work?  I'm not seeing anything yet, but i'm not sure if i'm just not using the program correctly.

pavel-demin, 2016-09-13:
BTW. My applications don't work with the 0.96 SD card images currently provided by the Red Pitaya developers:
https://github.com/RedPitaya/RedPitaya/issues/77

As a workaround, I suggest to use the Debian based SD card images from the following links:
http://pavel-demin.github.io/red-pitaya-notes/debian-ecosystem
http://downloads.redpitaya.com/downloads/0.95

> Was the command supposed to get the app working in a state where zoom works or so that it just launches?

MCPHA client worked without crashes when I tested it with Debian 8 and BLT 2.5.

With Ubuntu 14.04,  MCPHA client crashes when I try to zoom.

> Should i expect the histogram to still work?

I've just tested it with Ubuntu 14.0. It works if I don't try to zoom.

pavel-demin, 2016-09-13:
> I'm not seeing anything yet, but i'm not sure if i'm just not using the program correctly.

If you uncheck the 'amplitude threshold' check-box, then the program should start to fill the lowest bins of the histogram with noise.

closed 2016-09-22 by pavel-demin