pavel-demin/red-pitaya-notes #352: CIC compiler Redpitaya

issmaf44, 2016-09-21:
Dear Pavel,
I am trying to send the output of a CIC filter to the DAC you developed for the RedPitaya.
To try this I used a external signal generating a 1kHz / 100 mV peak sine. The sinusoidal signal is transmitted toa CIC decimation filter through the ADC. The used CIC decimation have the following parameters: 
   decimator, 3 stages,  differential delay 1, 1 channel, fixed rate 100
    input sample frequency : 125 MHz
    clock frequency : 125 MHz
    input data width : 14 bits
    quantization : truncation
    output data width : 14 bits
    all other parameters are unchanged
This filter output is send on a CIC filter interpolator prior to transmit to the DAC. This filter are tuned with the following parameters:
interpolator, 3 stages,  differential delay 1, 1 channel, fixed rate 100
    input sample frequency : 1.25 MHz
    clock frequency : 125 MHz
    input data width : 14 bits
    quantization : truncation
    output data width : 14 bits
The gain due to the cascade of the two CIC filter is ((100)^3)².
The trouble is the visualized signal on the oscilloscope is of 20mV and i think it would be 0.1((100)^3)²
Do you have an idea of where the problem might come from?

Thanks in advance for any help

pavel-demin, 2016-09-21:
If I'm not mistaken, the normalized gain of the CIC filter can be calculated as ((R*M)^N)/(2^ceiling(N*log2(R*M))), where M is differential delay, N is number of stages, R is rate change factor.

So, for the filter that you describe, it should be 0.95.

100 mV  * 0.95 * 0.95 = 90 mV

Looks like a factor 4 is still missing. Maybe the CIC compiler shifts its results by 1 bit. 

Could you try to set output data width to 15 bits for both filters?

issmaf44, 2016-09-21:
I set the output data to 15 bits for both filters and its works well. 
 I have another a question, Because of the passband droop, and therefore narrow usable passband, i used a single rate FIR compiler. The filter is synthesized in Matlab with a unity gain in the band pass. 
The filter coefficient was:-7.109910e-04, -1.112864e-04, 1.108516e-03, 3.816352e-03, 8.833174e-03, 1.671161e-02, 2.754876e-02, 4.087898e-02, 5.567736e-02, 7.048093e-02, 8.360946e-02, 9.344460e-02, 9.871252e-02, 9.871252e-02, 9.344460e-02, 8.360946e-02, 7.048093e-02, 5.567736e-02, 4.087898e-02, 2.754876e-02, 1.671161e-02, 8.833174e-03, 3.816352e-03, 1.108516e-03, -1.112864e-04, -7.109910e-04

and the IP FIR filter has the following parameters.
single rate
input sample frequency : 1.25 MHz
clock frequency : 125 MHz
input data width : 15 bits
output rounding mode : convergent rounding to even
output data width : 15 bits.
The coefficient are quantized only in 32 bits
all other parameters are unchanged

When i visualize the output of the DAC, the amplitude was of 58 mV, the FIR filter reduce the amplitude of the input signal. while the gain should be equal to unity ( so the ouput should be equal to 100mV)

Thank you for your help

pavel-demin, 2016-09-21:
> I set the output data to 15 bits for both filters and its works well.

Have you checked that it also works with full scale amplitudes (2 Vpp)?

> input data width : 15 bits

What is connected to the FIR filter input? If it's the CIC filter from the previous question, then I'd set this input data width to 14 bits.

> When i visualize the output of the DAC, the amplitude was of 58 mV

Setting FIR's input data width to 14 bits should solve this problem.

The idea is to boost only the outputs, not the inputs. So, all the inputs should be 14-bit wide. And all the outputs should be 15-bit wide.

Moreover, I think that it's better to use more bits for intermediate results to decrease the effect of the rounding errors:

CIC1: input 14 bits, output 25 bits
subset1: input 25 bits, output 24 bits
FIR: input 24 bits, output 25 bits
subset2: input 25 bits, output 24 bits
CIC2: input 24 bits, output 14 bits

Here is a link to a subset configuration:
https://github.com/pavel-demin/red-pitaya-notes/blob/master/projects/sdr_transceiver/rx.tcl#L203

issmaf44, 2016-09-21:
Yes it works with 1.9vpp.

The CIC decimator filter (CIC1) is connected at the FIR filter input. The FIR filter output is send to the CIC interpolator filter (CIC2). 

in order to decrease the effect of the rounding errors. 
This is my configuration
CIC1: input 14 bits, output 29 bits
subset1: input 29 bits [31:0], output 24 bits [31:8]
FIR: input 24 bits, output 32 bits
subset2: input 32 bits [31:0], output 24 bits  [31:8]
CIC2: input 24 bits, output 24 bits
 subset2: input 24 bits [23:0], output 14 bits  [23:8]

In the Subset configuration, the data is processed by byte.
with this configuration , the amplitude of the DAC output was of 40 mV instead of 100mV.
What's wrong in my configuration?

pavel-demin, 2016-09-21:
I don't understand why you use [23:8] or [31:8].

Could you try the configuration from my previous comment?

Here it's with bit numbers:

CIC1: input 14 bits, output 25 bits
subset1: input 25 bits, output 24 bits [23:0]
FIR: input 24 bits, output 25 bits
subset2: input 25 bits, output 24 bits [23:0]
CIC2: input 24 bits, output 14 bits

pavel-demin, 2016-09-21:
> In the Subset configuration, the data is processed by byte.

Yes, the input width should be set to 4 bytes and the output width should be set to 3 bytes.

issmaf44, 2016-09-22:
I use [23:8]  because i thought that the data are in big endian mode.
In your configuration it seems that data are in little endian mode.
I test your configuration, but when the input width of the CIC2 filter is of 24 bits, the ouput width can't be lower than 24 bits.
So i try this configuration
CIC1: input 14 bits, output 25 bits
subset1: input 25 bits [31:0], output 24 bits [23:0]
FIR: input 24 bits, output 25 bits
subset2: input 25 bits [31:0], output 24 bits [23:0]
CIC2: input 24 bits, output 24 bits
subset3: input 24 bits [23:0], output 16 bits [15:0]
When i visualize the DAC output, i don't have a sine,

pavel-demin, 2016-09-22:
Thanks for the test.

I think that the problem is with subset3. If output of CIC2 is 24-bit wide, then the output of subset3 should indeed include the most significant bits [23:8] as you previously proposed.

Could you try the following modification?

CIC2: input 24 bits, output 25 bits
subset3:  input 25 bits [31:0], output 16 bits [23:8]

pavel-demin, 2016-09-22:
After some thought, I changed the subset3 configuration in my previous comment. The idea is that the most significant (MSB) DAC bit (bit number 13) should be the MSB-1 (bit number 25-1=24) of the CIC2 output.

issmaf44, 2016-09-22:
in this case could you think that the configuration should be:
CIC1: input 14 bits, output 25 bits
subset1: input 25 bits [31:0], output 24 bits [24:1]    // in order to take the 24 MSB of the CIC1 output
FIR: input 24 bits, output 25 bits
subset2: input 25 bits [31:0], output 24 bits [24:1]     // in order to take the 24 MSB of the CIC1 output
CIC2: input 24 bits, output 25 bits
subset3: input 24 bits [31:0], output 16 bits [26:11]

pavel-demin, 2016-09-22:
As you previously observed, the CIC filter should be multiplied by 2 to obtain the same amplitude. This multiplication by 2 is done by outputting 25 bits and taking only 24 of the least significant bits. 

So, I think that the configuration should look like the following:
CIC1: input 14 bits, output 25 bits
subset1: input 25 bits [31:0], output 24 bits [23:0] // CIC1 output multiplied by 2
FIR: input 24 bits, output 25 bits
subset2: input 25 bits [31:0], output 24 bits [23:0] // FIR output multiplied by 2
CIC2: input 24 bits, output 25 bits
subset3: input 24 bits [31:0], output 16 bits [26:11] // CIC2 output multiplied by 2 and converted to 14 bits

issmaf44, 2016-09-22:
your configuration works normally but when the amplitude of the sine at the ADC input is of 1vpp, the DAC output saturate.
Why its doesn't work with full scale amplitude of 2vpp?

and i don't understand your previous comment,
why the CIC filter should be multiplied by 2 to obtain the same amplitude?
I am agree with the fact that multiplication by 2 is done by left shifting to one bit (so 25 bits in our case)
but i would like to understand why we should take 24 LSB [23:0] and not 24 MSB [24:1] ?

pavel-demin, 2016-09-22:
> why the CIC filter should be multiplied by 2 to obtain the same amplitude?

I can propose two answers:
 - I don't know. It's a question for the Xilinx developers.
 - It's what your initial results showed. Without multiplying by 2 the outputs of the two CIC filter, the resulting amplitude was four times lower than expected.

pavel-demin, 2016-09-22:
> your configuration works normally but when the amplitude of the sine at the ADC input is of 1vpp, the DAC output saturate.
> Why its doesn't work with full scale amplitude of 2vpp?

Looks like there is an overflow somewhere.

I can propose to try one more configuration without intermediate shifts/multiplications:

CIC1: input 14 bits, output 24 bits
FIR: input 24 bits, output 24 bits
CIC2: input 24 bits, output 24 bits
subset: input 24 bits, output 16 bits [22:7] // CIC2 output multiplied by 8 and converted to 14 bits

issmaf44, 2016-09-23:
i try this configuration, the DAC output is of 2.1 Vpp when the sine amplitude is of 1Vpp.
why the bits  [22:7] are chosen in the ouput of the subset converter?

pavel-demin, 2016-09-23:
> why the bits [22:7] are chosen in the output of the subset converter?

To multiply CIC2 output by 8.

> DAC output is of 2.1 Vpp when the sine amplitude is of 1Vpp

Look like we're almost there :-) So, if multiplying by 8 is too much, then let's try to multiply by 4:

subset: input 24 bits, output 16 bits [23:8] // CIC2 output multiplied by 4 and converted to 14 bits

pavel-demin, 2016-09-24:
I'm closing this issue because I think that I've already answered all the questions about the CIC filters.

closed 2016-09-24 by pavel-demin

issmaf44, 2016-09-26:
I try your last configuration. Its work well. Thanks for your help.
However i have a last question. I see that the DAC ouput is not symetric, there is an offset of 25 mV.
I try to calibrate the redpitaya using the command calib -w but its change anything in the DAC output.
Could you think that it is possible to compensate this offset using the redpitaya DAC code that you have developed. 
or if you have another solution to eliminate this offset.
thanks for your help.

issmaf44, 2016-09-27:
Thanks for all I'm trying to do that.