pavel-demin/red-pitaya-notes #361: No TX power

ka6s, 2016-10-04:
Pavel,

I've hooked up the  WM8731 development board to try out the CODEC.  I have receive Audio but no TX power. I've tried to get SSB transmit power through using the VAC as well as through the CODEC now and have never seen any indication that the Transmit audio is reaching the HPSDR application, or that other than CW transmit works.  

Any suggestions?

Thanks,

Steve Wilson KA6S

pavel-demin, 2016-10-04:
Thanks for the test, Steve.

To get the microphone audio from the codec to PowerSDR mRX PS, I had to switch off VAC1 and VAC2.

on3vna, 2016-10-04:
@ka6s 
There is an option in the menu's to disable VAC automatically when using other modes then digimodes.
 Goto vac1 or vac2 and select auto enable.

73
Johan

pavel-demin, 2016-10-04:
Here is a screenshot showing my PowerSDR working with the  WM8731 codec. The important buttons and sliders are: MON, MOX, Master AF, VAC1, VAC2, MIC and COMP.
[image]

pavel-demin, 2016-10-04:
BTW. It's possible to switch on/off the on-board microphone amplifier via Setup->Transmit->Mic->20dB Mic Boost.

ka6s, 2016-10-04:
Thank you all for the suggestions.  I have used PowerSDR on a real HPSDR
setup, along with the Mic I'm trying to use.  I suspect my issue might be
the Mic in question is an Electret microphone, and there isn't a bias
available on the development board. So I'm going to get another Mic and
give that a try this evening.

I'll keep you posted on progress.

pavel-demin, 2016-10-04:
Do you have schematics of this development board?

I think that even without a microphone connected, the noise of the codec's ADC should be seen on the spectrum display in the SSB TX mode.

on3vna, 2016-10-04:
@pavel-demin ,
I'm planning to use the same microphone i use on my flex radio's. So I have to look how to wire them up.

Just send you some info .
73 de ON3VNA
Johan

ka6s, 2016-10-04:
The web page for the board is at:

http://www.mikroe.com/add-on-boards/audio-voice/audio-codec-proto/

There is a datasheet with schematic there.

Also - don't see ANY noise at the moment.  This has been a recurring
problem  - I've never seen transmit audio.

Steve

pavel-demin, 2016-10-04:
Thanks for the link. So, we're using exactly the same board. I can confirm that this board provides microphone bias for an electret microphone.

Just to be sure. Have you tried to disable VAC1 and VAC2 and to enable MIC as shown on the screenshot from one of my previous comments?

ka6s, 2016-10-05:
I've set it up very much like what you have suggested and I'm seeing a
-80dbm on the meter when I key up on MOX.  How do you enable the Mic Bias
output. Looks like it is a software controlled function?

Steve

pavel-demin, 2016-10-05:
> I've set it up very much like what you have suggested and I'm seeing a -80dbm on the meter when I key up on MOX.

Great! So, the codec's ADC works and the ADC's samples are sent to PowerSDR mRX PS.

>  How do you enable the Mic Bias output. Looks like it is a software controlled function?

Both the Microphone Input and Microphone Bias are powered up when the program starts. Here is the code line that does it:
https://github.com/pavel-demin/red-pitaya-notes/blob/master/projects/sdr_transceiver_hpsdr/server/sdr-transceiver-hpsdr.c#L348

Could you please try the following?
 1. Increase the MIC level using the MIC slider.
 2. Switch on the microphone amplifier via Setup->Transmit->Mic->20dB Mic Boost.

ka6s, 2016-10-05:
> transceiver_hpsdr/server/sdr-transceiver-hpsdr.c#L348

Well - got to -70dB on the Mic input level.

So the other piece of info that is likely relevant is that I have a High
SWR warning which MAY be the culprit.  I don't have the SWR_F/R hooked up
yet!  How would I defeat this in the server code to send back a 1:1 match
to the Application?  Maybe that is a starting place.

Thanks for your help Pavel!

Steve

on3vna, 2016-10-05:
@ka6s ,
The SWR warning is indeed because the SWR_F/R is not hooked up.
But you can in the ALEX tab SWR protection uncheck "enable  protection"
Another solution would be connecting the SWR_R to gnd and the SWR_F to 3V .

GL
73 de ON3VNA

pavel-demin, 2016-10-05:
> I have a High SWR warning which MAY be the culprit. I don't have the SWR_F/R hooked up yet!

Do you have something connected to the Analog input 0 and 1 pins (http://wiki.redpitaya.com/index.php?title=Extension_connectors)? Maybe the easiest workaround would be to connect them to GND.

> How would I defeat this in the server code to send back a 1:1 match to the Application?

You could try to comment out the lines 933-950 and 977-994:

https://github.com/pavel-demin/red-pitaya-notes/blob/master/projects/sdr_transceiver_hpsdr/server/sdr-transceiver-hpsdr.c#L933

https://github.com/pavel-demin/red-pitaya-notes/blob/master/projects/sdr_transceiver_hpsdr/server/sdr-transceiver-hpsdr.c#L977

ka6s, 2016-10-05:
> transceiver_hpsdr/server/sdr-transceiver-hpsdr.c#L933
> transceiver_hpsdr/server/sdr-transceiver-hpsdr.c#L977

I disabled the SWR protection and the warning went away - so that is out.
I'll next look at the Mic to see if I DO have the requisite supply
voltage.. but need to go off to work.  Thanks all for the help!

73 de Steve KA6S

ka6s, 2016-10-06:
> projects/sdr_transceiver_hpsdr/server/sdr-transceiver-hpsdr.c#L933
> projects/sdr_transceiver_hpsdr/server/sdr-transceiver-hpsdr.c#L977

Latest information.  I plugged in a new mic and still no success.  I then
plugged in my real HPSDR and saw a nice unmodulated noise pulse like Pavel
described with the same setup, so it isn't the PowerSDR application
configuration.  It'll get mic data from the real HPSDR hardware - so it is
something in my hardware/software on the Red Pitaya I guess.

Steve

ka6s, 2016-10-06:
Pavel,

Could I please trouble you to point out the server code involved in moving
the data from the i2s input to the tx message buffer output?  Looks like
I'm going to have to go through that path to figure out where things are
getting lost.  I took a look at the server code and I'm not sure what the
paths are any longer..

Thanks,

Steve

pavel-demin, 2016-10-06:
> Could I please trouble you to point out the server code involved in moving the data from the i2s input to the tx message buffer output?

The audio samples are read on line 912:

https://github.com/pavel-demin/red-pitaya-notes/blob/master/projects/sdr_transceiver_hpsdr/server/sdr-transceiver-hpsdr.c#L912

They are copied to the tx message buffer on lines 970 and 1014:

https://github.com/pavel-demin/red-pitaya-notes/blob/master/projects/sdr_transceiver_hpsdr/server/sdr-transceiver-hpsdr.c#L970

https://github.com/pavel-demin/red-pitaya-notes/blob/master/projects/sdr_transceiver_hpsdr/server/sdr-transceiver-hpsdr.c#L1014

ka6s, 2016-10-07:
> transceiver_hpsdr/server/sdr-transceiver-hpsdr.c#L912
> transceiver_hpsdr/server/sdr-transceiver-hpsdr.c#L970
> transceiver_hpsdr/server/sdr-transceiver-hpsdr.c#L1014

Pavel,

You can close this one - this was a bad solder joint on the Codec card
connector. No audio getting to the Red Pitaya.  It now works!

73 de Steve KA6S

pavel-demin, 2016-10-08:
It's great that you've solved the problem. I'm closing this issue.

closed 2016-10-08 by pavel-demin

R6YY, 2016-12-28:
Уважаемый Павел! Возможно ли использование в качестве Audio Codec модуль на основе ЧИПа ALIENTEK VS1053 (например такой https://ru.aliexpress.com/item/ALIENTEK-VS1053-Module-MP3-Player-Audio-Decoding-STM32-Microcontroller-Development-Board/32708705751.html?spm=2114.41010708.4.78.8EKUpZ ). Документация на ЧИП здесь https://www.sparkfun.com/datasheets/Components/SMD/vs1053.pdf
----------------
Dear Pavel! Whether the use is possible as the Audio Codec Module chip based ALIENTEK VS1053 (eg a https://ru.aliexpress.com/item/ALIENTEK-VS1053-Module-MP3-Player-Audio-Decoding-STM32-Microcontroller-Development-Board/ 32708705751.html? spm = 2114.41010708.4.78.8EKUpZ). Documentation CHIP VS1053 here https://www.sparkfun.com/datasheets/Components/SMD/vs1053.pdf

pavel-demin, 2016-12-28:
> Возможно ли использование в качестве Audio Codec модуль на основе ЧИПа ALIENTEK VS1053

На мой взгляд, это совсем неподходящий чип.

R6YY, 2016-12-28:
Павел! Большое спасибо, что откликнулись.

Мой вопрос связан с желанием уменьшить затраты на приобретение 
дополнительных модулей.

Поэтому спрошу: нет ли у Вас на примете чего подходящего из 
рекомендованных WM8731.

73, Сергей

28.12.2016 11:54, Pavel Demin пишет:

pavel-demin, 2016-12-28:
> Поэтому спрошу: нет ли у Вас на примете чего подходящего из рекомендованных WM8731.

Есть еще TLV320AIC23B - полный аналог WM8731.

Я использую MikroElektronika Audio Codec Board:
https://shop.mikroe.com/add-on-boards/audio-voice/audio-codec-proto

Но они иногда высылают платы с неправильным кварцевым генератором. Нужен кварцевый генератор на 12.288 МГц.

Есть несколько схем для самостоятельной сборки:
http://www.circuitvalley.com/2015/01/audio-codec-breakout-tvl32aic23b-raspberry-linux.html
http://www.mikrocontroller.net/articles/Audio_Codec_Board_mit_TLV320AIC23b

R6YY, 2016-12-28:
Павел! Большое спасибо за консультацию.

Для начала своих экспериментов попробую USB звуковую карту.

С наступающим Вас Новым годом, всех благ и успеха в деле.

73, Сергей R6YY

28.12.2016 12:49, Pavel Demin пишет:

R6YY, 2017-02-05:
Павел, здравствуйте!

Я прошу уточнить: возможно ли подключение РП в качестве СДР 
приёмопередатчика напрямую к компьютеру? У меня без роутера ничего не 
выходит.
Я загружаю на РП один из Ваших образов (например 
sdr_transceiver_hpsdr-0.94-1451), запускаю на компе ПоверСДР, но 
последний не видит РП.
Прошу помочь советом. Перечитал краснодарский форум и радиосканер, но 
ответа не нашёл.
Заранее признателен, Сергей R6YY

pavel-demin, 2017-02-05:
Здравствуйте Сергей,

Самый простой способ - это установить на компьютере статический IP адрес 192.168.1.101. Вот ссылка на инструкцию:
http://wiki.redpitaya.com/index.php?title=Windows

С уважением,

Павел

R6YY, 2017-02-05:
Уважаемый Павел!

Большое Вам спасибо за консультацию. РП запела. Я в восторге после трёх 
дней неудач.

Позвольте ещё вопрос. Я запустил "SDR приемник совместим с HPSDR".
Что надо сделать в ПоверСДР, чтобы работали два канала приёмника?
Я пытался включить конфигурацию АНАН-100Д, но результата никакого.
Если не затруднит, подскажите пжл.

Сергей R6YY

05.02.2017 15:42, Pavel Demin пишет:

pavel-demin, 2017-02-05:
> Я запустил "SDR приемник совместим с HPSDR". Что надо сделать в ПоверСДР, чтобы работали два канала приёмника?

Лучше запускать "SDR transceiver compatible with HPSDR" и использовать конфигурацию Hermes. Второй канал включается кнопкой RX2.

R6YY, 2017-02-06:
Павел, большое спасибо за консультацию.

Я весьма впечатлён работой РП. Вчера вечером сравнивал РП и Гермес на 
80М на диапазонные настроеннве антенны (без переделок, перемычка входа 
2-5). Я разницы не увидел. Пробовал в качестве скиммер-сервера. Блеск! А 
я уж было хотел распрощаться с РП.

Вам всяческих удач и всего доброго.

Сергей R6YY

06.02.2017 1:04, Pavel Demin пишет:
>
>     Я запустил "SDR приемник совместим с HPSDR". Что надо сделать в
>     ПоверСДР, чтобы работали два канала приёмника?
>
> Лучше запускать "SDR transceiver compatible with HPSDR" и использовать 
> конфигурацию Hermes. Второй канал включается кнопкой RX2.
>

R6YY, 2017-04-20:
Уважаемый Павел,

прошу Вашего совета.

RP проработав несколько часов в качестве CW Skimmer на основе Вашего 
"SDR receiver compatible with HPSDR" прекратила функционировать. При 
запуске HDSDR шумовая полка на минимуме панорамы, а посередь её палка, в 
динамике однотонный звук частотой около 600-700гц.

В поисках неисправности запустил PowerSDR в режиме двух приёмников. 
Оказалось, что второй приёмник работает штатно. А первый также, как 
описано выше.

Не подскажете ли в чём причина и возможно ли победить неисправность.

Очень надеюсь на Ваш совет.

С уважением, Сергей