pavel-demin/red-pitaya-notes #362: curl on rtl8192cu.tgz gets empty file

dragniko, 2016-10-04:
mkdir -p tmp
curl -L https://googledrive.com/host/0B-t5klOOymMNfmJ0bFQzTVNXQ3RtWm5SQ2NGTE1hRUlTd3V2emdSNzN6d0pYamNILW83Wmc/rtl8192cu/rtl8192cu.tgz -o tmp/rtl8192cu.tgz
  % Total    % Received % Xferd  Average Speed   Time    Time Time  Current
                                 Dload  Upload   Total   Spent Left  Speed
^M  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0^M100  1613  100  1613    0     0  10437      0 --:--:-- --:--:-- --:--:-- 10474
mkdir -p tmp/linux-xlnx-xilinx-v2016.2
tar -zxf tmp/linux-xlnx-xilinx-v2016.2.tar.gz --strip-components=1 --directory=tmp/linux-xlnx-xilinx-v2016.2
tar -zxf tmp/rtl8192cu.tgz --directory=tmp/linux-xlnx-xilinx-v2016.2/drivers/net/wireless/realtek

gzip: stdin: not in gzip format
tar: Child returned status 1
tar: Error is not recoverable: exiting now
Makefile:89: recipe for target 'tmp/linux-xlnx-xilinx-v2016.2' failed
make: *** [tmp/linux-xlnx-xilinx-v2016.2] Error 2

pavel-demin, 2016-10-04:
Thanks for spotting this problem. It's related to #353. Googledrive has recently stopped its [hosting service](https://support.google.com/drive/answer/2881970). I've updated all the web pages but forgot to update the scripts.

pavel-demin, 2016-10-04:
I've replaced links to googledrive with links to dropbox in Makefile and scripts with [this commit](https://github.com/pavel-demin/red-pitaya-notes/commit/9b4d014b62536d46e3e43bddf5136fe45a77ed2e).

Could you please give my Makefile another try?

closed 2016-10-04 by pavel-demin