pavel-demin/red-pitaya-notes #369: Unable to decode

allenwait, 2016-10-20:
Hello,
I am having problems decoding WSPR, I have set my callsign and grid and modified to bands to the 8 bands i wish to monitor but to no avail. In the decode-wspr.log I have the following messages:
Thu Oct 20 05:09:01 CEST 2016
Sleeping ...
Thu Oct 20 05:10:00 CEST 2016
Recording ...
/root/write-c2-files: no process found
Error on line 20 in configuration file.
Decoding ...
Cannot open data file 'wspr_*_161020_0310.c2'
Thu Oct 20 05:11:01 CEST 2016
Sleeping ...
Thu Oct 20 05:12:00 CEST 2016
Recording ...
/root/write-c2-files: no process found
Error on line 20 in configuration file.
Decoding ...
Cannot open data file 'wspr_*_161020_0312.c2'

Any help will be appreciated.

Regard,
Allen - VK4FAW

pavel-demin, 2016-10-20:
Hello Allen,

Looks like the problem is in the modified configuration file. After outputting the 'Error on line 20 in configuration file' message, write-c2-files stops and doesn't record any .c2 files.

Could you please check that the 'bands' list in write-c2-files.cfg contains eight uncommented  comma separated entries and that there is no trailing commas?

Best regards,

Pavel

allenwait, 2016-10-20:
Hi Pavel,

Thank you for your prompt reply.  Unfortunately still no luck.  Please find following the current log file and the write-c2-files.cfg set up.  Also, may I ask how I can connect to a wireless network (client mode)?

Thank you again for your help.

Al

Thu Jan  1 01:01:01 CET 1970
Sleeping ...
Thu Oct 20 08:44:26 CEST 2016
Recording ...
/root/write-c2-files: no process found
Error on line 20 in configuration file.
Decoding ...
Cannot open data file 'wspr_*_161020_0644.c2'
Thu Oct 20 08:45:01 CEST 2016
Sleeping ...

// frequency correction, from -100.0 ppm to +100.0 ppm
corr = 0.0;

// comma separated list of bands
// trailing commas are not allowed
bands = (
//  { freq  =  0.137500; chan = 1; },
//  { freq  =  0.475700; chan = 1; },
  { freq  =  1.838100; chan = 1; },
  { freq  =  3.594100; chan = 1; },
//  { freq  =  5.288700; chan = 1; },
  { freq  =  7.040100; chan = 1; },
  { freq  = 10.140200; chan = 1; },
  { freq  = 14.097100; chan = 1; },
  { freq  = 18.106100; chan = 1; },
  { freq  = 21.096100; chan = 1; },
//  { freq  = 24.926100; chan = 1; },
  { freq  = 28.126100; chan = 1; },
//  { freq  = 50.294500; chan = 1; }
);

pavel-demin, 2016-10-20:
`{ freq = 28.126100; chan = 1; },` is the last uncommented entry and the comma after this entry is the source of the problem. Please remove the comma from this entry.

>  Also, may I ask how I can connect to a wireless network (client mode)?

Normally, it should be enough to replace the content of /etc/network/interfaces.d/wlan0 with the following lines:
```
allow-hotplug wlan0
iface wlan0 inet dhcp
  wpa-ssid MYSSID
  wpa-psk MYPSKHASH
  wpa-group CCMP
```

MYSSID and MYPSKHASH should be replaced with the SSID and PSK hash of your Wi-Fi access point.

If you don't have the PSK hash, then it can be obtained from the WPA passphrase with the following command:
```
wpa_passphrase MYSSID MYPASSPHRASE
```

This command can be run from the Red Pitaya command line.

Commands to restart wlan0 are:
```
ifdown wlan0
ifup wlan0
```

More information can be found at https://wiki.debian.org/WiFi/HowToUse

allenwait, 2016-10-20:
Hi Pavel,

Excellent, works great.  My apologies for bothering you for something so simple.

Thank you again and take care.

Al

closed 2016-10-20 by allenwait